ros-drivers/velodyne #45: fail to read calibration

bricerebsamen, 2015-04-02:
When running the tests, it complains about not being able to read the calibration file:

```
[velodyne_pointcloud.rosunit-cloud_nodelet_32e_hz_test/test_hz][passed]

SUMMARY
 * RESULT: SUCCESS
 * TESTS: 1
 * ERRORS: 0
 * FAILURES: 0

rostest log file is in /home/default/.ros/log/rostest-zoox164-28207.log
-- run_tests.py: verify result "/home/default/zoox_ros_ws/build/test_results/velodyne_pointcloud/rostest-tests_cloud_nodelet_32e_hz.xml"
[100%] Built target _run_tests_velodyne_pointcloud_rostest_tests_cloud_nodelet_32e_hz.test
-- run_tests.py: execute commands
  /opt/ros/indigo/share/rostest/cmake/../../../bin/rostest --pkgdir=/home/default/zoox_ros_ws/src/velodyne/velodyne_pointcloud --package=velodyne_pointcloud --results-filename tests_cloud_node_hz.xml /home/default/zoox_ros_ws/src/velodyne/velodyne_pointcl
oud/tests/cloud_node_hz.test
... logging to /home/default/.ros/log/rostest-zoox164-28616.log
[ROSUNIT] Outputting test results to /home/default/zoox_ros_ws/build/test_results/velodyne_pointcloud/rostest-tests_cloud_node_hz.xml
YAML Exception: yaml-cpp: error at line 0, column 0: bad conversion
[ERROR] [1427931670.442163378]: Unable to open calibration file: /home/default/zoox_ros_ws/src/velodyne/velodyne_pointcloud/params/64e_utexas.yaml
testcloud_node_32e_hz_test ... ok
```

Note the `ERROR` towards the end. In gdb I traced it to calibration.cc:70 where it reads `HORIZ_OFFSET_CORRECTION`. The corresponding field does not exist in the calibration file. Was probably introduced in fc20169be77cb7f649961fb97f91e565fe9ba28a

jack-oquin, 2015-04-02:
Yes. I've seen this recently, too. That field did not exist in the old 64E model.

I am not sure what to do. Do you have a suggestion? Should we make the field optional, or change the default calibration to some newer model?

bricerebsamen, 2015-04-02:
I think it depends on how critical this parameter is, i.e. can it be optional and/or have a default value. It also depends whether you want to invalidate all the calibration files that people have generated, more or less painstakingly, over the years.

Seems that before the recent changes, you were using the `vert_offset_correction` for the `horiz_offset_correction` which was bad anyway. Also it seems that 1 could be used as a default value.

So I'd say check whether `horiz_offset_correction` is available and use 1 as a default value otherwise. That will take care of the other peoples old calibration files. Maybe print a warning to notify that the calibration file should be fixed. Then fix the calibration files used in the tests so that they can be used as a reference file.

I don't use this package personally.

jack-oquin, 2015-04-02:
That sounds reasonable to me.

jack-oquin, 2015-04-02:
For the 32E, there appears to be an error in the test cases. They try to use the generic calibration for that device, which does contain the ``horizontal_offset_correction``, set to zero.

Somehow, the correct calibration file is not being selected.

jack-oquin, 2015-05-05:
I think we need to make ``horizontal_offset_correction`` optional. It is not used by all devices.

See also: #41

prclibo, 2015-05-06:
@bricerebsamen why should `1` be the default value for `horizontal_offset_correction `?
For reference, https://github.com/Kitware/VeloView/blob/master/VelodyneHDL/vtkVelodyneHDLReader.cxx#L1136 VeloView is using `0` as default.

jack-oquin, 2015-05-06:
@prclibo: It looks like it should be one because it is used as a multiplier for the sin or cos or the rotation angle.

See here for an example: https://github.com/ros-drivers/velodyne/blob/master/velodyne_pointcloud/src/lib/rawdata.cc#L211

Does that seem wrong to you?

prclibo, 2015-05-07:
@jack-oquin According to the manual:

> The horizontal offset of each laser as viewed from the back of the laser.

Although I am not sure what does this mean, I guess this is like some offset from the laser emitter to the laser center. 1meter sounds like too big as the default value.

jack-oquin, 2015-05-07:
I don't know what it means, either.

bricerebsamen, 2015-05-07:
I don't use their calibration, and I don't use your pointcloud node. But I have a similar parameter in my calibration. The lasers are not exactly centered on the rotation axis. There is a slight offset to the side, either by design or as a result of a manufacturing error. If that's the case, 0 should be the default value.

jack-oquin, 2015-05-07:
Ahh, thanks to you both for clearing that up.

jack-oquin, 2016-03-12:
Fixed in #80

jack-oquin, 2016-03-12:
OK, it's fixed now.

Please test and re-open if there is any problem.

closed 2016-03-12 by jack-oquin