ros-drivers/velodyne #48: Generated calibration has invalid values

jray-ecr, 2015-05-01:
When I do this:

    rosrun velodyne_pointcloud gen_calibration.py example.xml

then run 

    roslaunch velodyne_driver nodelet_manager.launch
    roslaunch velodyne_pointcloud cloud_nodelet.launch  calibration:=/path/to/example.yaml

I get see an error in the nodelet manager output:

    YAML Exception: yaml-cpp: error at line 0, column 0: bad conversion
    [ERROR] [1430497621.262533093]: Unable to open calibration file:  /path/to/example.yaml

I spent some time comparing the output to the output from the 64e_utexas.yaml that is included, and found that the min_intensity and max_intensity values seemed to be causing a problem.  If I convert them from floats to ints (e.g. 40.0 to 40) in the yaml, then the command 

    roslaunch velodyne_pointcloud cloud_nodelet.launch  calibration:=/path/to/example-fixed.yaml

works as expected.

jray-ecr, 2015-05-01:
Unable to figure out how to attach files, so here's the xml for example.xml:

<?xml version="1.0" encoding="UTF-8" standalone="yes" ?>
<!DOCTYPE boost_serialization>
<boost_serialization signature="serialization::archive" version="4">
<DB class_id="0" tracking_level="1" version="0" object_id="_0">
	<distLSB_>0.2</distLSB_>
	<position_ class_id="1" tracking_level="0" version="0">
		<xyz>
			<count>3</count>
			<item>0</item>
			<item>0</item>
			<item>0</item>
		</xyz>
	</position_>
	<orientation_ class_id="2" tracking_level="0" version="0">
		<rpy>
			<count>3</count>
			<item>0</item>
			<item>0</item>
			<item>0</item>
		</rpy>
	</orientation_>
	<colors_ class_id="3" tracking_level="0" version="0">
		<count>64</count>
		<item_version>0</item_version>
		<item class_id="4" tracking_level="0" version="0">
			<rgb>
				<count>3</count>
				<item>0.84018773</item>
				<item>0.39438292</item>
				<item>0.78309923</item>
			</rgb>
		</item>
		<item>
			<rgb>
				<count>3</count>
				<item>0.840177</item>
				<item>0.77233541</item>
				<item>0.39436942</item>
			</rgb>
		</item>
		<item>
			<rgb>
				<count>3</count>
				<item>0.08879225</item>
				<item>0.8096742</item>
				<item>0.71766233</item>
			</rgb>
		</item>
		<item>
			<rgb>
				<count>3</count>
				<item>0.70946825</item>
				<item>0.40906385</item>
				<item>0.82345313</item>
			</rgb>
		</item>
		<item>
			<rgb>
				<count>3</count>
				<item>0.5796597</item>
				<item>0.9642939</item>
				<item>0.064805068</item>
			</rgb>
		</item>
		<item>
			<rgb>
				<count>3</count>
				<item>0.32542917</item>
[... 1616 lines cut ...]
				<distCorrectionX_>159.73015</distCorrectionX_>
				<distCorrectionY_>155.3174</distCorrectionY_>
				<vertOffsetCorrection_>14.76663</vertOffsetCorrection_>
				<horizOffsetCorrection_>2.5999999</horizOffsetCorrection_>
				<focalDistance_>1250</focalDistance_>
				<focalSlope_>1</focalSlope_>
			</px>
		</item>
		<item>
			<px class_id_reference="9" object_id="_64">
				<id_>63</id_>
				<rotCorrection_>0.72077739</rotCorrection_>
				<vertCorrection_>-11.769718</vertCorrection_>
				<distCorrection_>144.03641</distCorrection_>
				<distCorrectionX_>144.85272</distCorrectionX_>
				<distCorrectionY_>147.61101</distCorrectionY_>
				<vertOffsetCorrection_>14.709725</vertOffsetCorrection_>
				<horizOffsetCorrection_>-2.5999999</horizOffsetCorrection_>
				<focalDistance_>950</focalDistance_>
				<focalSlope_>0.60000002</focalSlope_>
			</px>
		</item>
	</points_>
</DB>
</boost_serialization>

jack-oquin, 2015-05-01:
Thanks for the bug report.

What device model do you have?

jray-ecr, 2015-05-02:
it is a Velodyne HDL-64E S2. The XML I posted is the calibration from velodyne.
-- 
Sent by whistling into a 2400 baud modem

prclibo, 2015-07-22:
@jray-ecr did #42 fix this?

jack-oquin, 2016-03-04:
The fix in #42 has been merged. Please re-open if it did not solve this problem.

closed 2016-03-04 by jack-oquin