ros-drivers/velodyne #50: a possible  improvement in the coordinate calculation due to the mathematical model

kunlileo, 2015-06-11:
Hi guys! 

I am a student from KIT. I am now working on my graduation project so I get to know the velodyne scanner and the codes of it. Somehow, I noticed that the codes for the calculation of the xy_distance uses a expression that is different from the one I worked out or the one I read about in other papers. I manage to adjust my expression in the codes and I got a better result of the scanns. 

Here are the several changes I made:
original one : 
// Compute the distance in the xy plane (w/o accounting for rotation)
          float xy_distance = distance * cos_vert_angle;
new one:
          float xy_distance = distance * cos_vert_angle  + vert_offset * sin_vert_angle;

original one : 
         xy_distance = distance_x * cos_vert_angle;
new one:
          xy_distance = distance_x * cos_vert_angle + vert_offset * sin_vert_angle ;

original one : 
         xy_distance = distance_y * cos_vert_angle;
new one:
         xy_distance = distance_y * cos_vert_angle + vert_offset * sin_vert_angle ;

original one : 
        float  z = distance_y * sin_vert_angle + vert_offset;
new one:
        float z = distance_y * sin_vert_angle + vert_offset*cos_vert_angle;

That's all. That would be very nice if you could test the codes with the work you are working on, and see if there would be any improvement and give me some feedback. 

Hope you guys have a nice day.

kunlileo, 2015-06-11:
and of course the I used the for the calculation of x  this code:
 x = xy_distance * sin_rot_angle - horiz_offset * cos_rot_angle;
instead of 
 x = xy_distance * sin_rot_angle + horiz_offset * cos_rot_angle;

jack-oquin, 2015-06-19:
Thanks for the suggestion. A pull request containing these corrections would be helpful.

prclibo, 2015-07-13:
@kunlileo Hi, I noticed your PR #51 . I don't see why you are adding a `vert_offset * sin_vert_angle` term. Nor do I understand what this term stands for. The original code is consistent with the velodyne 64E_S2 [manual](http://www.velodynelidar.com/lidar/products/manual/63-HDL64E%20S2%20Manual_Rev%20D_2011_web.pdf)

Could you possibly provide your reference paper about the new added term. Thank you!

kunlileo, 2015-07-13:
@prclibo  Hi, Thanks for your comment. That was actually due to the mathematical model we used to describe the offsets.I read three papers about the velodyne LIDAR and two of them used the same model we got here. These two papers are "LIDAR Velodyne HDL-64E Calibration using Pattern Planes" and "Velodyne HDL-64E Proximity Laser Scanner Calibration-annotated". In the paper "LIDAR Velodyne HDL-64E Calibration using Pattern Planes" the model was very detailed described but some diagrams are also wrong which can just mislead you.( It  took me a long time to understand the model because of the wrongly described diagrams.) But in these to papers they are using the same terms for the calculation for the x ,y ,z which are derived out of the model. In my mid-term presentation  I also made a draft for the Model, but  I don't have it now with me. If you could just give me your email address I could just send you the scan of my draft which might help you to understand the model.

jack-oquin, 2015-07-13:
@prclibo and @kunlileo: thanks for posting these references. We should probably include them in whatever code we finally end up using.

prclibo, 2015-07-22:
@kunlileo Thanks for the reference. The paper clarifies my confusion about the so-called 'vertical correction' in that it is actually not on a vertical direction. It would be good to including the papers in the code.

jack-oquin, 2016-04-10:
Can you re-run this test and verify whether or not the current ``master`` source fixes your problem?

jack-oquin, 2016-06-09:
Closing since there was no response.

closed 2016-06-09 by jack-oquin