ros-drivers/velodyne #52: Possible wrong type

xwirth, 2015-06-25:
Hello,

there is a possible wrong type in velodyne_pointcloud rawdata.h for VLP16.
Original code is : 
  static const int VLP16_BLOCK_TDURATION = 110.592;
  static const int VLP16_DSR_TOFFSET = 2.304;²
  static const int VLP16_FIRING_TOFFSET = 55.296;

Corrected code would be :   
  static const float VLP16_BLOCK_TDURATION = 110.592f;
  static const float VLP16_DSR_TOFFSET = 2.304f;
  static const float VLP16_FIRING_TOFFSET = 55.296f;

Sincerely

jack-oquin, 2015-07-04:
Your comment looks right to me. Assuming the fractional values are correct, typing them as ``int`` causes undesired truncation.

jack-oquin, 2016-04-11:
This is fixed now in ``master``. Please test to confirm, and re-open if there is still a problem

closed 2016-04-11 by jack-oquin