ros-drivers/velodyne #54: Compile Error

droter, 2015-07-16:
Hi,

I cloned the source and tried to compile but I am getting this error:

catkin_ws/src/velodyne/velodyne_driver/include/velodyne_driver/input.h:36:18: fatal error: pcap.h: No such file or directory
 #include <pcap.h>

I am running Indigo on Ubuntu 14.04.

Thanks,

Matt

jack-oquin, 2015-07-16:
You are missing one of the dependencies, libpcap-dev.

jack-oquin, 2015-07-23:
A generally useful way to install all the dependencies of everything in your catkin workspace that's not already resolved within that workspace:

```
cd ~/catkin_ws
rosdep install --from-paths src --ignore-src --rosdistro indigo -y
```

Please close this issue if that fixes your problem.

droter, 2015-07-23:
Thanks for your help.

Yes, after installing the dependencies I was able to build.

Matt

closed 2015-07-23 by droter

firesnow1234, 2019-01-18:
How about Ubuntu16.04 in this issues？

JWhitleyAStuff, 2019-01-18:
The instructions provided above resolve this issue on Ubuntu 14.04, 16.04, and 18.04.