ros-drivers/velodyne #56: VLP-16 Data Structure

ferrefrederick, 2015-07-23:
Hi everyone, 

I'm currently working with a VLP-16 and using it for various applications. 

What I've noticed is that, and correct me if I'm wrong, the data structure of the VLP-16 is copied from the HDL-32E, meaning that in Last or Strongest Return Mode, there is actually a Block of 32 "hits" per Laser Fire. The second block of 16 "hits" is just a copy of the first "16". 

To be picky, this is doubling the bandwidth required for no reason.. 

Afterwards, if we are in dual return mode, it doubles this initial information, so it sends 2 blocks of 32 (meaning there is 2 blocks of 16 that are useless), again doubling the bandwidth for no reason. 

Can anyone confirm this ? And if so, isn't this an issue ? (probably on Velodyne's side though). 

And a side question as a bonus : For anyone using the return mode, how are you differentiating between same point sent twice and an actual point and its echo ? 

Regards,

Frederick

jack-oquin, 2015-07-23:
Thanks for the report. 

For cross-reference, here are some ROS questions and answers about this topic:

* http://answers.ros.org/question/212668/pointcloud-visualisation-blinking-when-using-velodyne-vlp16-dual-return-mode/
* http://answers.ros.org/question/212653/solvedhow-to-select-dual-return-mode-for-velodyne-vlp16/

A summary of the current status:

* The ROS PointCloud2 message used to publish Velodyne scan data has no way to represent 3D points obtained from multi-scan devices like the VLP-16.
* There is reportedly a way to work around this issue by configuring the device via its web server not to use multi-echo mode.

If you are able to attempt that web server configuration trick, please report back here whether it helps or not.

I still have no access to a VLP-16, so all this is based on others' reports. But, I am hoping that will change in the next month or two.

ferrefrederick, 2015-07-23:
Hi Jack, 

Thanks for the quick answer. 

My problem is more around the sending of information twice, considering half of it is useless. I have access to a VLP-16, So i can confirm the echo mode or Strongest/Last is configurable in the webserver easily. What it does though, is enable or not the echo, and that is functioning as intended. 

Even when not in echo mode, there is are two blocks of 16 'hits' per laser fire which should (ideally) not be the case. To be honest, this is more a discussion to have with Velodyne directly (which I'm going to contact), but I wanted to know if other people who have one have noticed this and what they think about it. 

Regards,

jack-oquin, 2015-07-23:
As far as I can recall, you are the first to mention the wasted bandwidth problem.

Please update this issue with whatever you learn from talking to Velodyne.

To be clear: are you trying to eliminate the wasted space in the raw packets? Or are the converted 3D points getting published twice each? 

I can't think of anything the driver can do about the extra packet contents.

ferrefrederick, 2015-08-07:
Hi all, 

The doc I had from Velodyne was wrong, they sent me an updated version and I get what I had assumed wrong. What the VLP-16 does is use the same structure as the 32, but it sends the Azimuth N in the first 16 and the Azimuth N+1 in the next 16. So it does not actually send the same point twice. And that 32 block is doubled in Dual return mode, which is logical (echo of Azimuth N and echo of Azimuth n+1). 

I'm guessing you do the interpolation of N and N+1 here in the driver, I should really dive in the code of what you do but haven't had the time...

jack-oquin, 2015-08-07:
Thanks for checking. 

Is this still an issue, or should we close it?

jack-oquin, 2015-10-27:
Since there has been no further discussion and there are no proposed driver changes, I am closing this issue.

If there is more to discuss or things we still need to do, please re-open it.

closed 2015-10-27 by jack-oquin

jack-oquin, 2015-11-11:
I've read more about this device, and I think it's probably OK to turn on Dual return mode. The point cloud will have more points, and there is currently no way to distinguish which are strongest echo and which are last echo. But, the data can be represented in the cloud.