ros-drivers/velodyne #57: Using pcap timestamp

pierriko, 2015-08-31:
Hello,

Is there any way to convert a pcap to a rosbag using timestamp form the pcap ?

In `InputPCAP::getPacket`: `pkt->stamp = ros::Time::now();` while `pcap_pkthdr` contains a `timeval`.

* https://github.com/ros-drivers/velodyne/blob/master/velodyne_driver/src/lib/input.cc#L245

Would it be interesting to replace the `ros::Time::now` by `header.ts` ?

jack-oquin, 2015-08-31:
Not without adding a [Phase Locked Loop](https://en.wikipedia.org/wiki/Phase-locked_loop) to handle skew between the device clock and the system clock.

The ROS time is based on system time. All messages from all sources must use that so they make sense together.

If you are willing to develop or donate a good PLL implementation for this driver under an appropriate open source license, that would be a valuable contribution. There have been lengthy discussions on this topic in the past, but no one has contributed a solution.

pierriko, 2015-09-01:
Hello and thank you for your answer,

Correct me if I'm wrong, but timestamps from the pcap come from the kernel time at the moment of the tcpdump, not the device clock.

My use-case here, is that I have others dumps with system timestamps (GPS, IMU, Gyro, ...) and I'd like to keep the sams time frame for all of them.

jack-oquin, 2015-09-03:
Please excuse me, I completely misunderstood your question. I thought you were asking about GPS time, which some Velodyne models provide.

Your idea seems reasonable. I guess no one ever tried it before. Most people probably save the ``/velodyne_packets`` topic in rosbag format along with other topics like ``/tf`` and ``/odom``. That's what I do. That way, the ``rosbag play --clock`` option will publish the appropriate ROS time for running with those data.

The PCAP timestamp may need some conversion to ROS time format. Have you tried making the suggested change? Does it work with your other data?

The PCAP file parameter is mostly used for unit tests simulating input from various device models. We might need to introduce an additional option to control whether to use current time or the PCAP time stamps.

Do you want the driver to actually publish ROS time? It would seem an odd place to do that sort of thing.

jack-oquin, 2015-09-04:
That conversion looks reasonable, but an additional parameter will also be needed, so users can say whether they want to use current time or PCAP time during playback. 

The parameter default will have to work the way it currently does, to avoid surprising people.

JWhitleyAStuff, 2019-02-07:
@jonla1 - See #58. Any way you could work on this PR?

closed 2019-02-07 by JWhitleyAStuff