ros-drivers/velodyne #59: make a new release for Indigo, and maybe Jade

jack-oquin, 2015-09-24:
There are lots of enhancements that should be released.

Docs need updating.

jack-oquin, 2015-11-11:
Version 1.2.0 was released to Indigo in August, 2014. I want to keep that distro stable, yet release some major new features like VLP-16 support, because Indigo is a long-term support distribution. That needs to be done carefully to avoid affecting existing users while still supporting the new hardware.

Other changes can be released to Jade as a new version 1.3.0. Those features would be tested on Indigo, but not released as binaries. That way, people wanting those features on Indigo could get them by building it from source.

jack-oquin, 2015-11-11:
I am using the github milestones feature to mark which issues I intend to release to Indigo, or only to Jade. 

Some of those determinations are merely my personal judgement of the possible risk balanced with the value of providing that change in the long-term release distribution.

Anyone needing some feature marked with a Jade milestone released to Indigo should make a remark to that effect in the affected issue. I can probably be persuaded to change the target with reasonable justification.

jack-oquin, 2016-03-04:
Looking through the open issues, there are several (#21, #31, and #50) saying that the calculation of XYZ points is not correct. We need to carefully review all of them and figure out the best solution, then decide whether it should be released in Indigo or only Jade (probably both).

tonybaltovski, 2016-06-24:
A release that supports the VLP-16 would be great.  The current master works great with it.

kmhallen, 2016-11-08:
Until this is resolved, up-to-date binary packages can be found here:
https://bitbucket.org/DataspeedInc/ros_binaries

JWhitleyAStuff, 2017-11-27:
A new release is here! v1.3.0 was pushed to public last week.

closed 2017-11-27 by JWhitleyAStuff