ros-drivers/velodyne #60: velodyne_pointcloud: dynamic reconfigure limits min_range to 0.9 meters

jack-oquin, 2015-10-14:
Some models may support shorter ranges, so configuration should allow smaller settings.

jack-oquin, 2015-10-14:
Apparently, the VLP-16 *is* limited to 0.9 meters, like most of the other models.

jack-oquin, 2015-10-14:
The Windows ``veloview`` program *does* show returns as short as 0.45 meters.

So, there must be some additional bug in this driver.

jack-oquin, 2015-10-14:
The [VLP-16 manual](http://velodynelidar.com/docs/manuals/VLP-16%20User%20Manual%20and%20Programming%20Guide%2063-9243%20Rev%20A.pdf) on page 3 does mention a "range from 1m to 100m".

piyushk, 2015-10-14:
@jack-oquin Thanks for finding that. I missed it in my original sweep. 

I'll perform additional testing tomorrow to see if I can find a solution to this problem.

jack-oquin, 2015-10-14:
This issue is fixed. The remaining problem with close-range VLP-16 data deserves a separate issue, #61.

closed 2015-10-14 by jack-oquin