ros-drivers/velodyne #61: VLP-16: VeloView returns data at shorter range than this driver

jack-oquin, 2015-10-14:
Using our driver, I am unable to see anything closer than about 0.9m with the VLP-16.

Using the VeloView program on Windows, we were able to see objects with the same device at about half that distance, 45cm.

When using a VLP-16 with indoor robots, that difference is critical. We need to figure out how it is done.

xref: #60 
cc: @piyushk, @PatrickHussey, @bricerebsamen

PatrickHussey, 2015-10-15:
Back when I contributed the VLP-16 support updates I had looked into the min range issue for the same reasons (indoor applications). I can look at my dev branch next week and see what I did to improve it. 

Next week I'll be preparing a new platform with the sensor so I'll have a chance to see what I did to improve it and contribute it.

jack-oquin, 2015-10-15:
Thanks, Patrick. That would be very helpful.

jack-oquin, 2015-10-27:
@PatrickHussey: I have not been able to figure this out on my own yet, any ideas?

PatrickHussey, 2015-10-27:
Just started updating things for the new platform, will get back to you by Friday/Monday.

PatrickHussey, 2015-10-29:
Found the issue, you can see the changes in my local copy. I started merging other changes to get it up to date, and issue a pull request but ran out of time.

See
VLP16.launch
raw_input.cc
VelodyneConfig.cfg

The changes committed in those files allows you to get 0.45m with accurate measurements

PatrickHussey, 2015-10-29:
Sorry changes are committed to the one I forked:

https://github.com/PatrickHussey/velodyne

-Patrick Hussey
www.stanleyinnovation.com<http://www.stanleyinnovation.com>

On Oct 29, 2015, at 5:22 PM, Patrick Hussey <Patrick.Hussey@stanleyinnovation.com<mailto:Patrick.Hussey@stanleyinnovation.com>> wrote:

Found the issue, you can see the changes in my local copy. I started merging other changes to get it up to date, and issue a pull request but ran out of time.

See
VLP16.launch
raw_input.cc
VelodyneConfig.cfg

The changes committed in those files allows you to get 0.45m with accurate measurements

jack-oquin, 2015-10-29:
Thanks, Patrick!

PatrickHussey, 2015-10-29:
No problem so now that I have a little time here is the explanation:

in raw_input.cc points with intensities out of range were allowed to be added to the pointcloud. When an object was closer than 0.45m it would register as a point ~0.45m away. So filtering out the points that exceed min/max intensity did the trick.

Needed to change the dynamic reconfigure since it would override setting limits below 0.9.

Also updated the launch file with the appropriate limits.

jack-oquin, 2015-10-29:
> Needed to change the dynamic reconfigure since it would override setting limits below 0.9.
>
> Also updated the launch file with the appropriate limits.

Those changes are already committed in master.

PatrickHussey, 2015-10-29:
Sorry didn't have time to update to the present release, just made the changes in the local copy I had made. If they are already in the master you can just ignore. Will update my local copy to the master so I can help out going forward.

Thanks!

jack-oquin, 2015-10-30:
For the record, the [VLP-16 Users Manual](http://velodynelidar.com/docs/manuals/VLP-16%20User%20Manual%20and%20Programming%20Guide%2063-9243%20Rev%20A.pdf) states on page 10 that: "Points with distances less than one meter should be ignored."

Since we really need that extra check that Patrick provided to yield shorter ranges for indoor use, I would like to know Velodyne's official recommendation on this.

jack-oquin, 2015-11-04:
@PatrickHussey: I tried to duplicate the logic of your short-range intensity modification, but it's not working with my device. I still see no points closer than about a meter.

Perhaps I just made some silly coding error. I did not use your exact code. 

I pushed that modification to a separate ``short_range_vlp16`` branch. Maybe you can take a look and help me figure out what went wrong. The commit is 2644d7c.

The logic I inferred from reading the code in your fork was to ignore all points for which the intensity is not in range. That's what I tried to implement. 

Now that I think about it, I don't see how that could ever result in more points than we were getting without that extra test. Previously, it was forcing the intensity to be in that range, so now we should see fewer points than before. Right?

PatrickHussey, 2015-11-04:
The min_range changes allowed additional points to come in, the intensity gate was to exclude ones that were reported erroneously with the opened up range.

We tested by physically measuring the distance to the tip of an object from the sensor, and then slowly decreasing the distance to the object until we saw points that get "reflected back"

I won't have much time in the next couple weeks but I'll take a look again if I have a chance

jack-oquin, 2015-11-04:
I already had the min_range change. That parameter was set to 0.4m. I can verify that setting using ``rosparam get``.

I tested it by moving a whiteboard closer until the middle points started disappearing, then measuring the distance with a meter stick. That happens at a little under one meter.

jack-oquin, 2015-11-04:
What firmware version are you using? Mine seems to be 3.0.27.0, which appears to be the most recent available.

PatrickHussey, 2015-11-04:
not sure I would need to check, we got the ones we have came very early on

jack-oquin, 2015-11-04:
I verified it by going to the web config interface (192.1689.1.201) and checking the "info" tab.

jack-oquin, 2015-11-11:
@PatrickHussey: Because Indigo is a long-term support distribution, I want to fix this issue in that distro without affecting existing users of other device models. Since VLP-16 support is new (thanks to you) but not yet released, it looks OK to make fixes affecting only that model.

But, we should not change the semantics of minimum and maximum intensity for other models in a way that would affect existing installations. If we decide the semantics of excluding out-of-range intensities is correct and the former clamping to the specified limits is wrong, then that change can be introduced in Jade, perhaps with a parameter to provide backwards compatibility for anyone who needs it.

PatrickHussey, 2015-11-12:
Sounds reasonable to me. In the version I contributed to the VLP-16 data gets parsed by a separate function from the other velodyne sensors for that very reason. I didn't want to disrupt what was already being used.

I'm not sure what has changed since then as I've been working off that branch from almost a year ago on VLP-16 specifically. But if the driver still has a separate function for generating the VLP-16 point cloud, any changes for the VLP-16 should be contained and not affect the other sensors.

jack-oquin, 2015-11-13:
It does still have a separate ``unpack_vlp16()``, and I appreciate you doing it that way. It will definitely make things easier for Indigo.

For Jade, we can think about whether we want to factor out some of the code that's common to both types of devices.

jack-oquin, 2015-11-18:
Today, this is working for me in ``master`` with a real VLP-16. I can see points out to about 40cm, which is the configured min_range. 

I tried several different commits and branches, trying to figure out why it did not work last week. But, everything since 85228edd84c1f626c020863b77f1dbb4da3f1ea3 seems to work now, including my re-coding of the out-of-range intensity skipping @PatrickHussey suggested, currently in a separate ``short_range_vlp16`` branch.

I regret that I cannot explain what was wrong before or why it works now. Perhaps I did something stupid in my catkin workspace, and things were not getting re-compiled when I thought they were. Certainly, some of the earlier commits (d65ef885ff3f4fbe20d24be9dd31238b009d8642 and 85228edd84c1f626c020863b77f1dbb4da3f1ea3) were necessary.

I am closing this issue now because I can no longer reproduce it. I would ask anyone seeing similar problems to re-open it, because I am unhappy with the lack of a clear root cause for the bug I saw. It was definitely failing for some reason.

jack-oquin, 2015-11-18:
It is interesting to observe that the *intensity* of the close-up points seems *lower* than those at about a one meter or more range. That was observed using a highly reflective white-board. It is the opposite of what I would expect.

closed 2015-11-18 by jack-oquin

koelsinha, 2019-03-13:
Hi,

I am having the same issue with my VLP16. I have changed the min_range parameter in the launch file do not see any change. The following is my launch file:

```
<!-- -*- mode: XML -*- -->
<!-- run velodyne_pointcloud/CloudNodelet in a nodelet manager for a VLP-16 -->

<launch>

  <!-- declare arguments with default values -->
  <arg name="calibration" default="$(find velodyne_pointcloud)/params/VLP16db.yaml"/>
  <arg name="device_ip" default="" />
  <arg name="frame_id" default="velodyne" />
  <arg name="manager" default="$(arg frame_id)_nodelet_manager" />
  <arg name="max_range" default="50.0" />
  <arg name="min_range" default="0.5" />
  <arg name="pcap" default="" />
  <arg name="port" default="2368" />
  <arg name="read_fast" default="false" />
  <arg name="read_once" default="false" />
  <arg name="repeat_delay" default="0.0" />
  <arg name="rpm" default="600.0" />

  <!-- start nodelet manager and driver nodelets -->
  <include file="$(find velodyne_driver)/launch/nodelet_manager.launch">
    <arg name="device_ip" value="$(arg device_ip)"/>
    <arg name="frame_id" value="$(arg frame_id)"/>
    <arg name="manager" value="$(arg manager)" />
    <arg name="model" value="VLP16"/>
    <arg name="pcap" value="$(arg pcap)"/>
    <arg name="port" value="$(arg port)"/>
    <arg name="read_fast" value="$(arg read_fast)"/>
    <arg name="read_once" value="$(arg read_once)"/>
    <arg name="repeat_delay" value="$(arg repeat_delay)"/>
    <arg name="rpm" value="$(arg rpm)"/>
  </include>

  <!-- start cloud nodelet -->
  <include file="$(find velodyne_pointcloud)/launch/cloud_nodelet.launch">
    <arg name="calibration" value="$(arg calibration)"/>
    <arg name="manager" value="$(arg manager)" />
    <arg name="max_range" value="$(arg max_range)"/>
    <arg name="min_range" value="$(arg min_range)"/>
  </include>

</launch>
```

I would really appreciate it if you could take a look and tell me if I am missing something.
Also, do I need to change anything in the cfg folder? The following are my CloudNode.cfg and TransformNode.cfg files:

CloudNode.cfg:

```
#!/usr/bin/env python
PACKAGE = "velodyne_pointcloud"

from math import pi
from dynamic_reconfigure.parameter_generator_catkin import *

gen = ParameterGenerator()

gen.add("min_range", double_t, 0, "min range to publish", 0.9, 0.1, 10.0)
gen.add("max_range", double_t, 0, "max range to publish", 130, 0.1, 200)
gen.add("view_direction", double_t, 0, "angle defining the center of view",
        0.0, -pi, pi)
gen.add("view_width", double_t, 0, "angle defining the view width",
        2*pi, 0.0, 2*pi)

exit(gen.generate(PACKAGE, "cloud_node", "CloudNode"))
```
TransforNode.cfg

```
#!/usr/bin/env python
PACKAGE = "velodyne_pointcloud"

from math import pi
import dynamic_reconfigure.parameter_generator_catkin as pgc

gen = pgc.ParameterGenerator()

gen.add("min_range", 
  pgc.double_t, 
  0, 
  "min range to publish", 
  0.9, 0.1, 10.0)
  
gen.add("max_range", 
  pgc.double_t, 
  0, 
  "max range to publish", 
  130, 0.1, 200)
  
gen.add("view_direction", 
  pgc.double_t, 
  0, 
  "angle defining the center of view",
  0.0, -pi, pi)
  
gen.add("view_width", 
  pgc.double_t, 
  0, 
  "angle defining the view width",
  2*pi, 0.0, 2*pi)
  
gen.add("frame_id", 
  pgc.str_t, 
  0, 
  "new frame of reference for point clouds", 
  "odom")

exit(gen.generate(PACKAGE, "transform_node", "TransformNode"))
```

Thanks in advance!

aniketmpatil, 2024-02-09:
Were there any further updates on this issue? Or does the driver still work best with 0.9m of min_range?