ros-drivers/velodyne #62: VLP 16 - How to convert velodyne_packets to velodyne_points to view in rviz?

dwolfe64, 2015-10-28:
I am using the ROS to convert a .PCAP to a bag file. The bag file contains velodyne_packets topic and from the velodyne pointcloud driver ( https://github.com/ros-drivers/velodyne/tree/master/velodyne_pointcloud )  inside of params the VLP16db.yaml file doesn't seem to work. See problem below.

When I run the cloud node ( rosrun velodyne_pointcloud cloud_node _calibration:=/home/abe/velodyne/src/velodyne-master/velodyne_pointcloud/params/VLP16db.yaml ) it runs until I start playing the rosbag that contains the velodye_packets. it does a core dump when i try. Any suggestions?

jack-oquin, 2015-10-28:
Thanks for the report. 

Is it possible to make a small extract from your bag file that we can use for debugging?

jack-oquin, 2015-10-29:
@dwolfe64: I am very anxious to get a copy of your bag file so I can debug this on my own system.

This could be related to #63. If so, a bag that will reliably crash the driver would be a huge help.

dwolfe64, 2015-10-29:
@jack-oquin sorry for the delay. Here is a dropbox downloaded link to a small bagfile. https://www.dropbox.com/s/rd0ho9dlgml9jk6/RCP2.bag?dl=0 
Thanks for your help!

jack-oquin, 2015-10-29:
Thanks, @dwolfe64! That should help a lot.

jack-oquin, 2015-10-30:
I can reproduce the crash. It happens because of an assert failure in line 298 of rawdata.cc:
```
      assert(0xEEFF == raw->blocks[block].header);
```
That suggests the presence of a mangled packet. 

@PatrickHussey: An assert seems too harsh for that situation. Perhaps we should just skip that block and continue, logging a warning. What do you think?

jack-oquin, 2015-10-30:
Substituted the following in place of the ``assert()`` above:
```
      if (0xEEFF != raw->blocks[block].header) {
        ROS_WARN_STREAM("invalid VLP-16 header: block " << block <<
                        ", value: " << raw->blocks[block].header);
        continue;                       // skip to the next block
      }
```
Running the test again, I get many more warnings like these, but the node does not crash and I can see the point cloud in rviz:
```
[ WARN] [1446219345.315365857]: invalid VLP-16 header: block 0, value: 0
[ WARN] [1446219345.315443900]: invalid VLP-16 header: block 1, value: 0
[ WARN] [1446219345.315481810]: invalid VLP-16 header: block 2, value: 7287
[ WARN] [1446219345.315531828]: invalid VLP-16 header: block 3, value: 0
[ WARN] [1446219345.315569404]: invalid VLP-16 header: block 4, value: 0
[ WARN] [1446219345.315613002]: invalid VLP-16 header: block 5, value: 0
[ WARN] [1446219345.316359431]: invalid VLP-16 header: block 0, value: 0
[ WARN] [1446219345.316442793]: invalid VLP-16 header: block 1, value: 0
[ WARN] [1446219345.316480970]: invalid VLP-16 header: block 2, value: 7287
[ WARN] [1446219345.316529839]: invalid VLP-16 header: block 3, value: 0
[ WARN] [1446219345.316569162]: invalid VLP-16 header: block 4, value: 0
[ WARN] [1446219345.316613015]: invalid VLP-16 header: block 5, value: 0
[ WARN] [1446219345.317639602]: invalid VLP-16 header: block 0, value: 0
[ WARN] [1446219345.317719670]: invalid VLP-16 header: block 1, value: 0
[ WARN] [1446219345.317757178]: invalid VLP-16 header: block 2, value: 7287
[ WARN] [1446219345.317809569]: invalid VLP-16 header: block 3, value: 0
[ WARN] [1446219345.317847420]: invalid VLP-16 header: block 4, value: 0
[ WARN] [1446219345.317887025]: invalid VLP-16 header: block 5, value: 0
[ WARN] [1446219345.319438506]: invalid VLP-16 header: block 0, value: 0
[ WARN] [1446219345.319494475]: invalid VLP-16 header: block 1, value: 0
[ WARN] [1446219345.319532518]: invalid VLP-16 header: block 2, value: 7287
[ WARN] [1446219345.319573683]: invalid VLP-16 header: block 3, value: 0
[ WARN] [1446219345.319610856]: invalid VLP-16 header: block 4, value: 0
[ WARN] [1446219345.319651439]: invalid VLP-16 header: block 5, value: 0
[ WARN] [1446219345.320318625]: invalid VLP-16 header: block 0, value: 0
[ WARN] [1446219345.320370130]: invalid VLP-16 header: block 1, value: 0
[ WARN] [1446219345.320407831]: invalid VLP-16 header: block 2, value: 7288
[ WARN] [1446219345.320448461]: invalid VLP-16 header: block 3, value: 0
[ WARN] [1446219345.320486977]: invalid VLP-16 header: block 4, value: 0
[ WARN] [1446219345.320527700]: invalid VLP-16 header: block 5, value: 0
[ WARN] [1446219345.321340550]: invalid VLP-16 header: block 0, value: 0
[ WARN] [1446219345.321396071]: invalid VLP-16 header: block 1, value: 0
[ WARN] [1446219345.321433966]: invalid VLP-16 header: block 2, value: 7288
[ WARN] [1446219345.321470824]: invalid VLP-16 header: block 3, value: 0
[ WARN] [1446219345.321507464]: invalid VLP-16 header: block 4, value: 0
[ WARN] [1446219345.321543975]: invalid VLP-16 header: block 5, value: 0
[ WARN] [1446219345.323406091]: invalid VLP-16 header: block 0, value: 0
[ WARN] [1446219345.323489276]: invalid VLP-16 header: block 1, value: 0
[ WARN] [1446219345.323510948]: invalid VLP-16 header: block 2, value: 7288
[ WARN] [1446219345.323532911]: invalid VLP-16 header: block 3, value: 0
[ WARN] [1446219345.323555638]: invalid VLP-16 header: block 4, value: 0
[ WARN] [1446219345.323579070]: invalid VLP-16 header: block 5, value: 0
[ WARN] [1446219345.324093560]: invalid VLP-16 header: block 0, value: 0
[ WARN] [1446219345.324155480]: invalid VLP-16 header: block 1, value: 0
[ WARN] [1446219345.324176592]: invalid VLP-16 header: block 2, value: 7288
[ WARN] [1446219345.324197712]: invalid VLP-16 header: block 3, value: 0
[ WARN] [1446219345.324220281]: invalid VLP-16 header: block 4, value: 0
[ WARN] [1446219345.324243875]: invalid VLP-16 header: block 5, value: 0
[ WARN] [1446219345.324647438]: invalid VLP-16 header: block 0, value: 0
[ WARN] [1446219345.324708000]: invalid VLP-16 header: block 1, value: 0
[ WARN] [1446219345.324731488]: invalid VLP-16 header: block 2, value: 7288
[ WARN] [1446219345.324755428]: invalid VLP-16 header: block 3, value: 0
[ WARN] [1446219345.324778339]: invalid VLP-16 header: block 4, value: 0
[ WARN] [1446219345.324801892]: invalid VLP-16 header: block 5, value: 0
[ WARN] [1446219345.325769238]: invalid VLP-16 header: block 0, value: 0
[ WARN] [1446219345.325833116]: invalid VLP-16 header: block 1, value: 0
[ WARN] [1446219345.325856654]: invalid VLP-16 header: block 2, value: 7288
[ WARN] [1446219345.325880610]: invalid VLP-16 header: block 3, value: 0
[ WARN] [1446219345.325904577]: invalid VLP-16 header: block 4, value: 0
[ WARN] [1446219345.325928067]: invalid VLP-16 header: block 5, value: 0
```

jack-oquin, 2015-10-30:
The previous patch is clearly not a final fix, and I would like to understand better what is going on.

The errors I can see all come in the first six consecutive blocks of a packet. Why not the last six? Why all six? Did the device actually send these packets?

@dwolfe64: can you say more about how these data were collected? Do you have a corresponding PCAP by any chance?  What firmware level are you running?

Are there other types of packets the device could be transmitting? GPS or IMU?

dwolfe64, 2015-10-30:
@jack-oquin here is the link to the pcap file (https://www.dropbox.com/s/pp2eff482pxroiz/Test%202%202015-10-23-11-26-56_Velodyne-VLP-16-Data.pcap?dl=0). I recorded this file as our DJI S1000 was landing (it is only a few seconds long of data). I have the VLP16 hooked up to our UAV platform and I was using Ubiquiti wifi radios to stream our data from the sensor to our base station which was recording the data via Windows version of Veloview 3.0.6

jack-oquin, 2015-10-31:
The PCAP does contain port 8308 position packets interspersed with the data packets, but the driver should only be reading port 2368. If they were somehow getting published as /velodyne_packets messages, that might explain the error we are seeing.

Was the bag file produced by running velodyne_driver on that PCAP?

dwolfe64, 2015-10-31:
Yes velodyne_driver was used to convert pcap to bag

dwolfe64, 2015-10-31:
I followed the steps you outlined here: http://answers.ros.org/question/213080/convert-raw-velodyne-vlp16-pcap-to-bagfile/

PatrickHussey, 2015-10-31:
You can filter the pcap file for packets only received on port 2368 with wireshark and it should solve your problem. I've run into the same thing and solved it this way.

jack-oquin, 2015-10-31:
The PCAP reader in input.cc should be filtering those other packets, just as the actual driver does. Apparently, that is not working. The original HDL-64E for which this driver was written did not produce those packets. I guess we never noticed. 

I'll take a look at it. I can use the PCAP @dwolfe64 provided to test my fix.

jack-oquin, 2015-11-01:
The InputPCAP class definitely does *not* check the UDP port, only the IP address. That needs to be fixed as part of this issue.

jack-oquin, 2015-11-01:
Before fixing InputPCAP, I want to find a good resolution to the ``assert()`` problem.

My current thinking is for ``unpack_vlp16()``  to skip the remainder of any packet containing a data block without UPPER_BANK in the header. The rationale is that it is either an irrelevant packet (like the position packet in this issue), or the data from the device have somehow gotten corrupted. In either case, the points in that packet are not reliable and should not be included in the point cloud. We certainly do not want the driver to terminate in those situations.

That looks easy to do, and should probably suffice. 

@PatrickHussey, @dwolfe64, @nxydes, @konduri, @jlblanco Comments?

konduri, 2015-11-01:
@jack-oquin  I'm for the idea of skipping the reminder of the packet. I am however curious to see if the unclean data case is a recoverable issue or if it persists. I can put this patch and check if the ros warns stop in the logs over time. can you or @dwolfe64  do anything similar and get that confirmed? If the unclean data issue persists, we might have to fix the root cause.

jack-oquin, 2015-11-01:
I created a separate ``vlp16_fixes`` branch for working on these issues. I recommend using that as a base for further testing.

It currently has an improved version of the assert patch, see 70a8fae, which only warns of bad packets at most once a minute, so as not to fill up your log with warnings. Hopefully, that will be adequate. Let me know if you think of a better approach. 

We should add a bad packet count to the diagnostics topic. That is probably the best way to distinguish intermittent from persistent problems.

jack-oquin, 2015-11-11:
Commit 70a8fae is now merged into the ``master`` branch.

jack-oquin, 2015-11-13:
I opened a separate issue to fix the InputPCAP bug discovered here: #66.

dwolfe64, 2015-11-16:
@jack-oquin thank you so much for looking into this. I'm excited to give it a try now.

jack-oquin, 2015-11-18:
I am glad we were able to find and identify this problem. It helped a lot that you provide a PCAP to demonstrate the bug.

I merged the logic for skipping bad packets into master last week. So, I think this issue is solved even without the resolution of the separate problem #66 we also discovered here.

If you find that master works for your tests, let's close this issue and use #66 for the other bug.

dwolfe64, 2015-11-19:
@jack-oquin I just tried it and it works! Thanks again

closed 2015-11-19 by jack-oquin