ros-drivers/velodyne #63: Velodyne Drivers crash

nxydes, 2015-10-29:
Hello all,

We are experiencing some frustrating issues with these drivers where the entire nodelet stops functioning. It typically happens after we have run the robot for many hours (12-36) and then begins to happen at a relatively frequent rate (every hour to every couple hours). It seemingly happens randomly and we can not induce it. 

We are running our robots with ubuntu 14.04, ros indigo and the VLP16 model. We also are using the ros package pointcloud_to_laserscan to compress the 3d pointcloud into a laserscan for localization. This is run as a nodelet so it should be using the same process and publishes /velodyne_scan and /velodyne_scan2 .

Most of the time the symptoms are that none of the relevant topics are being published (velodyne_nodelet_manager/bond,  /velodyne_points,  /velodyne_packet  /velodyne_scan#) and the node is not pingable. Usually the node still exists but upon running "rosnode info /velodyne_nodelet_manager" at the end of the description it will state in some way an error communicating with the process.

Sometimes the node itself dies off completely and is not listed in rosnode list.

In trying to debug and find the root cause we initiated some debugging. These are the settings we used:
log4j.logger.ros.velodyne_nodelet_manager=DEBUG
log4j.logger.ros.cloud_nodelet=DEBUG 
log4j.logger.ros.driver_nodelet=DEBUG 
log4j.logger.ros.nodelet=DEBUG 
log4j.logger.ros.velodyne_driver=DEBUG 
log4j.logger.ros.velodyne_pointcloud=DEBUG 

This writes data really fast into the log file and grows by over a hundred MB per hour. However, unfortunately this came up with nothing. This is written to the file over and over for every pointcloud:

^[[32m[DEBUG] [1445929728.337558536]: Received packet, time: 1445929728.299481392^[[0m
^[[32m[DEBUG] [1445929728.337645137]: Received packet, time: 1445929728.300796509^[[0m
^[[32m[DEBUG] [1445929728.337735126]: Received packet, time: 1445929728.302118540^[[0m
^[[32m[DEBUG] [1445929728.337870391]: Received packet, time: 1445929728.303478241^[[0m
^[[32m[DEBUG] [1445929728.337972481]: Received packet, time: 1445929728.304841995^[[0m
^[[32m[DEBUG] [1445929728.338067212]: Received packet, time: 1445929728.306172848^[[0m
^[[32m[DEBUG] [1445929728.338160992]: Received packet, time: 1445929728.307489872^[[0m
^[[32m[DEBUG] [1445929728.338254090]: Received packet, time: 1445929728.308785200^[[0m
^[[32m[DEBUG] [1445929728.338347160]: Received packet, time: 1445929728.310112000^[[0m
^[[32m[DEBUG] [1445929728.338450413]: Received packet, time: 1445929728.311497688^[[0m
^[[32m[DEBUG] [1445929728.338568938]: Received packet, time: 1445929728.312825441^[[0m
^[[32m[DEBUG] [1445929728.338664660]: Received packet, time: 1445929728.314089775^[[0m
^[[32m[DEBUG] [1445929728.338756399]: Received packet, time: 1445929728.315409422^[[0m
^[[32m[DEBUG] [1445929728.338847289]: Received packet, time: 1445929728.316730022^[[0m
^[[32m[DEBUG] [1445929728.338940818]: Received packet, time: 1445929728.318071604^[[0m
^[[32m[DEBUG] [1445929728.339032473]: Received packet, time: 1445929728.319428921^[[0m
^[[32m[DEBUG] [1445929728.339123703]: Received packet, time: 1445929728.320755959^[[0m
^[[32m[DEBUG] [1445929728.339214184]: Received packet, time: 1445929728.322075367^[[0m
^[[32m[DEBUG] [1445929728.339307896]: Publishing 26503 Velodyne points, time: 1445929728322075^[[0m
^[[32m[DEBUG] [1445929728.427584178]: Publishing a full Velodyne scan.^[[0m

When the node dies  the log file is abruptly stopped without any indication as to why. In one case I observed it stopped in the middle of the received packet, time: blocks and at one point I observed it stop between publishing the velodyne points and publishing the velodyne scan

Looking at the master.log log file we found slightly more interesting outputs which I will paste at the end here, I would like to attach documents but it says I need write permission to this repository to do so. 

Any help in figuring out what is causing our nodes to crash would be greatly appreciated!

One instance:
[rosmaster.master][INFO] 2015-10-18 19:09:14,454: +PARAM [/amcl/initial_pose_y] by /amcl
[rosmaster.master][INFO] 2015-10-18 19:09:14,454: +PARAM [/amcl/initial_pose_a] by /amcl
[rosmaster.master][INFO] 2015-10-18 19:09:14,455: +PARAM [/amcl/initial_cov_xx] by /amcl
[rosmaster.master][INFO] 2015-10-18 19:09:14,456: +PARAM [/amcl/initial_cov_yy] by /amcl
[rosmaster.master][INFO] 2015-10-18 19:09:14,457: +PARAM [/amcl/initial_cov_aa] by /amcl
[rosmaster.master][INFO] 2015-10-18 19:09:16,544: +PARAM [/amcl/initial_pose_x] by /amcl
[rosmaster.master][INFO] 2015-10-18 19:09:16,546: +PARAM [/amcl/initial_pose_y] by /amcl
[rosmaster.master][INFO] 2015-10-18 19:09:16,547: +PARAM [/amcl/initial_pose_a] by /amcl
[rosmaster.master][INFO] 2015-10-18 19:09:16,548: +PARAM [/amcl/initial_cov_xx] by /amcl
[rosmaster.master][INFO] 2015-10-18 19:09:16,549: +PARAM [/amcl/initial_cov_yy] by /amcl
[rosmaster.master][INFO] 2015-10-18 19:09:16,550: +PARAM [/amcl/initial_cov_aa] by /amcl
[rosmaster.master][INFO] 2015-10-18 19:09:18,644: +PARAM [/amcl/initial_pose_x] by /amcl
[rosmaster.master][INFO] 2015-10-18 19:09:18,646: +PARAM [/amcl/initial_pose_y] by /amcl
[rosmaster.master][INFO] 2015-10-18 19:09:18,647: +PARAM [/amcl/initial_pose_a] by /amcl
[rosmaster.master][INFO] 2015-10-18 19:09:18,647: +PARAM [/amcl/initial_cov_xx] by /amcl
[rosmaster.master][INFO] 2015-10-18 19:09:18,648: +PARAM [/amcl/initial_cov_yy] by /amcl
[rosmaster.master][INFO] 2015-10-18 19:09:18,649: +PARAM [/amcl/initial_cov_aa] by /amcl
[rosmaster.master][INFO] 2015-10-18 19:09:20,744: +PARAM [/amcl/initial_pose_x] by /amcl
[rosmaster.master][INFO] 2015-10-18 19:09:20,746: +PARAM [/amcl/initial_pose_y] by /amcl
[rosmaster.master][INFO] 2015-10-18 19:09:20,748: +PARAM [/amcl/initial_pose_a] by /amcl
[rosmaster.master][INFO] 2015-10-18 19:09:20,749: +PARAM [/amcl/initial_cov_xx] by /amcl
[rosmaster.master][INFO] 2015-10-18 19:09:20,750: +PARAM [/amcl/initial_cov_yy] by /amcl
[... 538 lines cut ...]
    xmlrpcapi(api).publisherUpdate('/master', topic, pub_uris)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1224, in __call__
    return self.__send(self.__name, args)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1578, in __request
    verbose=self.__verbose
  File "/usr/lib/python2.7/xmlrpclib.py", line 1264, in request
    return self.single_request(host, handler, request_body, verbose)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1297, in single_request
    return self.parse_response(response)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1473, in parse_response
    return u.close()
  File "/usr/lib/python2.7/xmlrpclib.py", line 793, in close
    raise Fault(**self._stack[0])
Fault: <Fault -1: 'publisherUpdate: unknown method name'>

[rosmaster.master][INFO] 2015-10-29 10:38:12,386: publisherUpdate[/rosout] -> http://min12:41819/
[rosmaster.master][INFO] 2015-10-29 10:38:12,387: publisherUpdate[/rosout] -> http://min12:44797/
[rosmaster.master][INFO] 2015-10-29 10:38:12,395: publisherUpdate[/velodyne_nodelet_manager/bond] -> http://min12:41819/
[rosmaster.master][INFO] 2015-10-29 10:38:12,396: publisherUpdate[/velodyne_nodelet_manager/bond] -> http://min12:41819/
[rosmaster.master][INFO] 2015-10-29 10:38:12,397: publisherUpdate[/rosout] -> http://min12:41819/
[rosmaster.master][INFO] 2015-10-29 10:38:12,399: publisherUpdate[/rosout] -> http://min12:44797/
[rosmaster.master][INFO] 2015-10-29 10:38:12,406: publisherUpdate[/velodyne_nodelet_manager/bond] -> http://min12:41819/
[rosmaster.master][INFO] 2015-10-29 10:38:12,485: -PUB [/rosout] /pointcloud_to_laserscan http://min12:49823/
[rosmaster.master][INFO] 2015-10-29 10:38:12,486: -PUB [/velodyne_nodelet_manager/bond] /pointcloud_to_laserscan http://min12:49823/
[rosmaster.master][INFO] 2015-10-29 10:38:12,486: publisherUpdate[/rosout] -> http://min12:44797/

jack-oquin, 2015-10-29:
Thanks for the very informative report.

Note that VLP-16 was only recently added, has not yet been extensivley tested or included in an official driver release. 

So, it is quite possible that there may be some kind of model-specific bug, perhaps triggered by some combination of data in a VLP-16 packet, see #62.

jack-oquin, 2015-10-29:
I now have access to a VLP-16 belonging to the University of Texas at Austin. Since I heard about this problem, I have tried running the driver for about six hours while displaying the point cloud several times, so far without a crash. I am usually only there one day a week, so it's hard for me to run the test much longer.

I hope that the bag file from #62 may help me debug this issue more easily. If not, we may need to add additional debug logging to the driver to help us find it.

jack-oquin, 2015-10-29:
@nxydes: I interpret the rosmaster Python traceback as probably some kind of crash triggered when you query the status of a dead nodelet manager. Does that seem right to you?

If so, it is probably a separate problem from the driver crash itself, and we should report it as a [ros/ros_comm issue](https://github.com/ros/ros_comm/issues) for rosmaster.

jack-oquin, 2015-10-29:
I think we need a gdb thread backtrace of the velodyne manager process crashing.

Doing that in ROS is somewhat tricky. If you need detailed instructions, please ask.

jack-oquin, 2015-10-29:
Was this launched via the ``velodyne_pointcloud/launch/VLP16_points.launch`` script, so some other way?

If other, please post the commands needed to load and run the nodelet.

nxydes, 2015-10-29:
jack-oquin

We also came to the conclusion that the python traceback was merely indicating a "dead" node or in someway was no longer able to communicate.

As far as gdb thread backtrace we would appreciate the detailed instructions, a colleague of mine has some experience with it but we would like to know how you would do it.

I'm trying to get some bag data from one of the crashes, I can strip our ip from the bags and send you the bag files with just the velodyne data. That way we can see if it is something inherent with the data.

Here is the launch file we are using:

<?xml version="1.0"?>

<launch>
  <env name="ROSCONSOLE_CONFIG_FILE" value="$(find velodyne_pointcloud)/launch/velodyne_rosconsole.conf"/>

  
  <arg name="pcap" default="" />
  <arg name="calibration" default="$(find velodyne_pointcloud)/params/VLP16db.yaml"/>
  <arg name="min_range" default="0.4" />
  <arg name="max_range" default="130.0" />

  
  <include file="$(find velodyne_driver)/launch/nodelet_manager.launch">
    <arg name="model" value="VLP16"/>
  </include>

  
  <include file="$(find velodyne_pointcloud)/launch/cloud_nodelet.launch">
    <arg name="calibration" value="$(arg calibration)"/>
  </include>

  <arg name="camera" default="velodyne" />

  
  <node pkg="nodelet" type="nodelet" name="pointcloud_to_laserscan" args="load pointcloud_to_laserscan/pointcloud_to_laserscan_nodelet velodyne_nodelet_manager">

      <remap from="cloud_in" to="velodyne_points"/>
      <remap from="scan" to="velodyne_scan"/>
      <rosparam>
          #target_frame: camera_link # Leave disabled to output scan in pointcloud frame
          transform_tolerance: 0.01
          min_height: -1.34
          max_height: 0.5

          angle_min: -3.14 # -M_PI/2
          angle_max: 3.14 # M_PI/2
          angle_increment: 0.01 # M_PI/360.0
          scan_time: 0.3333
          range_min: 0.90
          range_max: 130.0
          use_inf: true

          # Concurrency level, affects number of pointclouds queued for processing, thread number governed by nodelet manager
          # 0 : Detect number of cores
          # 1 : Single threaded
          # 2->inf : Parallelism level
          concurrency_level: 1
      </rosparam>

  </node>

  <node pkg="nodelet" type="nodelet" name="pointcloud_to_laserscan2" args="load pointcloud_to_laserscan/pointcloud_to_laserscan_nodelet velodyne_nodelet_manager">

      <remap from="cloud_in" to="velodyne_points"/>
      <remap from="scan" to="velodyne_scan2"/>
      <rosparam>
          #target_frame: camera_link # Leave disabled to output scan in pointcloud frame
          transform_tolerance: 0.01
          min_height: 0.5
          min_height: 0.5
          max_height: 10.0

          angle_min: -3.14 # -M_PI/2
          angle_max: 3.14 # M_PI/2
          angle_increment: 0.01 # M_PI/360.0
          scan_time: 0.3333
          range_min: 0.90
          range_max: 130.0
          use_inf: true

          # Concurrency level, affects number of pointclouds queued for processing, thread number governed by nodelet manager
          # 0 : Detect number of cores
          # 1 : Single threaded
          # 2->inf : Parallelism level
          concurrency_level: 1
      </rosparam>

  </node>

</launch>
                                                                                                                                                91,1          Bot

jlblancoc, 2015-10-29:
Hi @nxydes , 

Yes, you will need to provide a gdb backtrace for further debugging...

It seems there is a prefix param which will launch the node under `gdb` automatically, see: http://wiki.ros.org/roslaunch/Tutorials/Roslaunch%20Nodes%20in%20Valgrind%20or%20GDB

Once the program crashes, you will see the command line prompt of `gdb`. Once there, execute the command `bt` (for a backtrace of the call stack) and copy the result here so the guys can better identify the origin of the problem ;-)

jack-oquin, 2015-10-29:
Running the ``pointcloud_to_laserscan_nodelet`` in the same nodelet manager address space with the velodyne driver is an excellent idea for system efficiency.

Unfortunately, it is hard to tell from the outside which nodelet in that process actually crashed.

As an experiment, would you mind running the laserscan nodelets in a separate address space? That will convince us whether the bug is in the Velodyne driver or in the conversion nodelet.

BTW, if I am reading it correctly, the second through fourth paragraphs of your launch script could be replaced with something like this, adding the other three args, if they seem worthwhile.:
```
 <!-- start Velodyne driver for VLP-16 -->
  <include file="$(find velodyne_pointcloud)/launch/VLP16_points.launch">
    <arg name="pcap" value="$(arg pcap)"/>
  </include>
```

jack-oquin, 2015-10-29:
@jlblancoc: thanks for posting. That is a good link for getting gdb backtraces. 

In this case, the core dump instructions described there should be sufficient.

jlblancoc, 2015-10-29:
You are welcome! We are receiving a VLP-16 in our University soon, so it's me who must thank you all for the effort of publishing and polishing these drivers! :+1:

nxydes, 2015-10-29:
@jack-oquin 

"As an experiment, would you mind running the laserscan nodelets in a separate address space? That will convince us whether the bug is in the Velodyne driver or in the conversion nodelet."

Not sure what you mean by running it in separate address space. We started out launching it in a separate launch file and only recently condensed it into a single launch file. We still had identical issues at that time. 

However, pointcloud_to_laserscan also has a regular node that I can launch. I can give that a shot today and see if it makes a difference. If you can instruct me on how to do separate addresses then I can try that as well, the only resources I found online simply said it could be done not how.

My colleague is looking into gdb core dump and we will get back to you with that as well! Thanks!

jack-oquin, 2015-10-29:
A ROS nodelet manager loads and runs all its nodelets using a pool of threads in a single Linux process with its own address space. If any of them crashes (segfault, bus error, etc.), the whole process terminates.

So, running ``pointcloud_to_laserscan`` as separate nodes instead of nodelets will be sufficient to establish that the crash is indeed in the Velodyne driver. Alternatively, you could load them into a different nodelet manager (other than the one we call `` velodyne_nodelet_manager``), but that is not important for this experiment.

konduri, 2015-10-30:
Hi guys, im having trouble running with the launch-prefix in the roslaunch, so might need a 101. i have to ssh into the machine and launch  the rosnodes. I've gotten three results with three launch-prefixes. All of them bad.

### First result
gdb -ex run --args  , throws a bunch of  ioctl for device errors
```
-  <node pkg="nodelet" type="nodelet" name="velodyne_nodelet_manager"
+  <node pkg="nodelet" type="nodelet" launch-prefix="gdb -ex run --args" name="velodyne_nodelet_manager"
   
-  <node pkg="nodelet" type="nodelet" name="pointcloud_to_laserscan" args="load pointcloud_to_laserscan/pointcloud_to_laserscan_nodelet velodyne_nodelet_manager">
+  <node pkg="nodelet" type="nodelet"  name="pointcloud_to_laserscan"  launch-prefix="gdb -ex run --args" args="load pointcloud_to_laserscan/pointcloud_to_laserscan_nodelet velodyne_no

-  <node pkg="nodelet" type="nodelet" name="cloud_nodelet"
+  <node pkg="nodelet" type="nodelet" name="cloud_nodelet" launch-prefix="gdb -ex run --args"
```
process[cloud_nodelet-4]: started with pid [8222]
  2 process[pointcloud_to_laserscan-5]: started with pid [8231]
  3 [tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
  4 [tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
  5 [tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
  6 process[pointcloud_to_laserscan2-6]: started with pid [8242]
  7 process[remap_outter-7]: started with pid [8262]
  8 process[remap_inner-8]: started with pid [8288]
  9 [tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
 10 [tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
 11 [tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]

### Second result
running with "xterm -e gdb --args" fails because im trying to launch remotely. However, if i plug in the computer to a display(this was on a test bench so i could do that) and launch it, xterms shells with gdb popped up and i could run them with "run" on each

### Third result
running with ""screen -d -m gdb --args""  is still thowing errors 

screen -d -m gdb --args /opt/ros/indigo/lib/nodelet/nodelet load pointcloud_to_laserscan/pointcloud_to_laserscan_nodelet velodyne_nodelet_manager cloud_in:=velodyne_points scan:=velodyne_scan __name:=pointcloud_to_laserscan __log:=/home/min2/.ros/log/55ffaebe-7ea5-11e5-9743-78d004206830/pointcloud_to_laserscan-5.log

Please make sure that all the executables in this command exist and have
executable permission. This is often caused by a bad launch-prefix.
The traceback for the exception was written to the log file
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]

jack-oquin, 2015-10-30:
I doubt you need a launch prefix to start with. 

Just set ``ulimit -c unlimited`` in the shell before launching, so you get a core dump. Then, run gdb on the core file after it crashes.

konduri, 2015-10-30:
sounds good. Will try it out and post results.

jack-oquin, 2015-10-30:
For future reference: if you need to run gdb on a nodelet via the launch prefix method, you need to apply it to the stanza that launches the original nodelet manager process. In this case, it is buried inside [velodyne_driver/launch/nodelet_manager.launch](https://github.com/ros-drivers/velodyne/blob/master/velodyne_driver/launch/nodelet_manager.launch#L10), which initially creates the nodelet manager, then loads the driver.

nxydes, 2015-10-31:
@jack-oquin 

We observed that the `error: [Errno 111] Connection refused` only happens if we kill the velodyne nodelet manager process while another communicating process (driver or pointcloud) is still alive.

When this happens it appears to cause that first Connection Refused error which we believe indicates the problem exists with the nodelet manager itself and not with the pointcloud to laserscan or other nodes. 

However, we are still investigating all possible cases. We haven't had a crash since this morning and so my colleague konduri is still waiting to try to use the gdb.

Thank you for your help and patience

jack-oquin, 2015-10-31:
As an experiment, you may want to try applying this temporary patch used to identify an assert failure in issue #62. It would be interesting to know if that stops your system from crashing. If it does, you will probably see lots warning messages. 
```
$ git diff
diff --git a/velodyne_pointcloud/src/lib/rawdata.cc b/velodyne_pointcloud/src/lib/rawda
index ec6fc6a..dfecc9c 100755
--- a/velodyne_pointcloud/src/lib/rawdata.cc
+++ b/velodyne_pointcloud/src/lib/rawdata.cc
@@ -295,7 +295,11 @@ namespace velodyne_rawdata
     const raw_packet_t *raw = (const raw_packet_t *) &pkt.data[0];
 
     for (int block = 0; block < BLOCKS_PER_PACKET; block++) {
-      assert(0xEEFF == raw->blocks[block].header);
+      if (0xEEFF != raw->blocks[block].header) {
+        ROS_WARN_STREAM("invalid VLP-16 header: block " << block <<
+                        ", value: " << raw->blocks[block].header);
+        continue;                       // skip to the next block
+      }
       azimuth = (float)(raw->blocks[block].rotation);
       if (block < (BLOCKS_PER_PACKET-1)){
         azimuth_diff = (float)((36000 + raw->blocks[block+1].rotation - raw->blocks[bl
```

nxydes, 2015-10-31:
@jack-oquin 

So we apparently did successfully obtain core dumps. Please find them at this link here which has access for 30 days:

https://www.dropbox.com/sh/5hu6bk0cqsouq9p/AAApR9Ltm6TDLonqjWLWOA-pa?dl=0

We are working through how to run the gdb analysis on this but figured you would like access in parallel. Within the shared folder I have included the launch file and configuration we used to obtain the core dump. Thank you!

I will try your temporary patch, perhaps over the weekend if not Monday.

jack-oquin, 2015-10-31:
When I run this, I can't see any of the symbols, because things are in different locations on my system. 
```
gdb /opt/ros/indigo/lib/nodelet/nodelet ~/Downloads/velodyne/core.23640
```
We really need to run gdb on the same system that generated the core file, preferably with ``thread all apply bt``, although a simple ``bt`` is probably adequate.

The fact that it terminated with SIGABRT is consistent with the possibility of the assert in input.cc getting triggered for some reason. The temporary patch above will at least eliminate that as a possibility.
```
warning: .dynamic section for "/usr/lib/x86_64-linux-gnu/libstdc++.so.6" is not at the expected address (wrong library or version mismatch?)

warning: .dynamic section for "/usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0" is not at the expected address (wrong library or version mismatch?)

warning: Could not load shared library symbols for 4 libraries, e.g. /home/min12/k5_adm_3/devel/lib//libdriver_nodelet.so.
Use the "info sharedlibrary" command to see the complete listing.
Do you need "set solib-search-path" or "set sysroot"?
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Core was generated by `/opt/ros/indigo/lib/nodelet/nodelet manager __name:=velodyne_nodelet_manager __'.
Program terminated with signal SIGABRT, Aborted.
#0  0x00007f7bb30bbf79 in __GI___open_catalog (cat_name=0x7328203c20746e63 <error: Cannot access memory at address 0x7328203c20746e63>, nlspath=<optimized out>, 
    env_var=0x5f2820666f657a69 <error: Cannot access memory at address 0x5f2820666f657a69>, catalog=0x44415f434c5f6570) at open_catalog.c:151
151	open_catalog.c: No such file or directory.
Traceback (most recent call last):
  File "/usr/share/gdb/auto-load/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.19-gdb.py", line 63, in <module>
    from libstdcxx.v6.printers import register_libstdcxx_printers
ImportError: No module named 'libstdcxx'
(gdb) bt
#0  0x00007f7bb30bbf79 in __GI___open_catalog (cat_name=0x7328203c20746e63 <error: Cannot access memory at address 0x7328203c20746e63>, nlspath=<optimized out>, 
    env_var=0x5f2820666f657a69 <error: Cannot access memory at address 0x5f2820666f657a69>, catalog=0x44415f434c5f6570) at open_catalog.c:151
#1  0x4649544e4544495f in ?? ()
#2  0x294e4f4954414349 in ?? ()
#3  0x6f657a6973202f20 in ?? ()
#4  0x765f6c6e5f282066 in ?? ()
#5  0x7079745f65756c61 in ?? ()
#6  0x4544495f434c5f65 in ?? ()
#7  0x544143494649544e in ?? ()
#8  0x29295d305b4e4f49 in ?? ()
#9  0x0000000000000000 in ?? ()
```

konduri, 2015-11-01:
I've run ```gdb /opt/ros/indigo/lib/nodelet/nodelet core.23640``` and the following is the result of ```bt```

```
#0  0x00007f7bb30bbf79 in __GI_raise (sig=sig@entry=6)
    at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
#1  0x00007f7bb30bf388 in __GI_abort () at abort.c:89
#2  0x00007f7bb30b4e36 in __assert_fail_base (
    fmt=0x7f7bb3206718 "%s%s%s:%u: %s%sAssertion `%s' failed.\n%n", 
    assertion=assertion@entry=0x7f7b8eb09bd0 "0xEEFF == raw->blocks[block].header", 
    file=file@entry=0x7f7b8eb09aa8 "/home/min12/k5_adm_3/src/k5_sensor_stack/navigation_sensors/velodyne/velodyne_pointcloud/src/lib/rawdata.cc", 
    line=line@entry=298, 
    function=function@entry=0x7f7b8eb09f60 <velodyne_rawdata::RawData::unpack_vlp16(velodyne_msgs::VelodynePacket_<std::allocator<void> > const&, pcl::PointCloud<velodyne_pointcloud::PointXYZIR>&)::__PRETTY_FUNCTION__> "void velodyne_rawdata::RawData::unpack_vlp16(const VelodynePacket&, velodyne_rawdata::VPointCloud&)") at assert.c:92
#3  0x00007f7bb30b4ee2 in __GI___assert_fail (
    assertion=0x7f7b8eb09bd0 "0xEEFF == raw->blocks[block].header", 
    file=0x7f7b8eb09aa8 "/home/min12/k5_adm_3/src/k5_sensor_stack/navigation_sensors/velodyne/velodyne_pointcloud/src/lib/rawdata.cc", line=298, 
    function=0x7f7b8eb09f60 <velodyne_rawdata::RawData::unpack_vlp16(velodyne_msgs::VelodynePacket_<std::allocator<void> > const&, pcl::PointCloud<velodyne_pointcloud::PointXYZIR>&)::__PRETTY_FUNCTION__> "void velodyne_rawdata::RawData::unpack_vlp16(const VelodynePacket&, velodyne_rawdata::VPointCloud&)")
---Type <return> to continue, or q <return> to quit---
    at assert.c:101
#4  0x00007f7b8eafedfe in velodyne_rawdata::RawData::unpack_vlp16(velodyne_msgs::VelodynePacket_<std::allocator<void> > const&, pcl::PointCloud<velodyne_pointcloud::PointXYZIR>&) ()
   from /home/min12/k5_adm_3/devel/lib/libvelodyne_rawdata.so
#5  0x00007f7b8eafe228 in velodyne_rawdata::RawData::unpack(velodyne_msgs::VelodynePacket_<std::allocator<void> > const&, pcl::PointCloud<velodyne_pointcloud::PointXYZIR>&) () from /home/min12/k5_adm_3/devel/lib/libvelodyne_rawdata.so
#6  0x00007f7b8eda5967 in velodyne_pointcloud::Convert::processScan(boost::shared_ptr<velodyne_msgs::VelodyneScan_<std::allocator<void> > const> const&) ()
   from /home/min12/k5_adm_3/devel/lib//libcloud_nodelet.so
#7  0x00007f7b8edc30b0 in boost::_mfi::mf1<void, velodyne_pointcloud::Convert, boost::shared_ptr<velodyne_msgs::VelodyneScan_<std::allocator<void> > const> const&>::operator()(velodyne_pointcloud::Convert*, boost::shared_ptr<velodyne_msgs::VelodyneScan_<std::allocator<void> > const> const&) const ()
   from /home/min12/k5_adm_3/devel/lib//libcloud_nodelet.so
#8  0x00007f7b8edc1a82 in void boost::_bi::list2<boost::_bi::value<velodyne_pointcloud::Convert*>, boost::arg<1> >::operator()<boost::_mfi::mf1<void, velodyne_pointcloud::Convert, boost::shared_ptr<velodyne_msgs::VelodyneScan_<std::allocator<void> > const> const&>, boost::_bi::list1<boost::shared_ptr<velodyne_msgs::VelodyneScan_<std::allocator<void> > const> const&> >(boost::_bi::type<void>, boost::_mfi::mf1<void, velodyne_pointcloud::Convert, boost::shared_ptr<velodyne_msgs::VelodyneScan_<std::allocator<void> > const> const&>&, boost::_bi::list1<boost::---Type <return> to continue, or q <return> to quit---
shared_ptr<velodyne_msgs::VelodyneScan_<std::allocator<void> > const> const&>&, int) () from /home/min12/k5_adm_3/devel/lib//libcloud_nodelet.so
#9  0x00007f7b8edc0418 in void boost::_bi::bind_t<void, boost::_mfi::mf1<void, velodyne_pointcloud::Convert, boost::shared_ptr<velodyne_msgs::VelodyneScan_<std::allocator<void> > const> const&>, boost::_bi::list2<boost::_bi::value<velodyne_pointcloud::Convert*>, boost::arg<1> > >::operator()<boost::shared_ptr<velodyne_msgs::VelodyneScan_<std::allocator<void> > const> >(boost::shared_ptr<velodyne_msgs::VelodyneScan_<std::allocator<void> > const> const&) ()
   from /home/min12/k5_adm_3/devel/lib//libcloud_nodelet.so
#10 0x00007f7b8edbe506 in boost::detail::function::void_function_obj_invoker1<boost::_bi::bind_t<void, boost::_mfi::mf1<void, velodyne_pointcloud::Convert, boost::shared_ptr<velodyne_msgs::VelodyneScan_<std::allocator<void> > const> const&>, boost::_bi::list2<boost::_bi::value<velodyne_pointcloud::Convert*>, boost::arg<1> > >, void, boost::shared_ptr<velodyne_msgs::VelodyneScan_<std::allocator<void> > const> const&>::invoke(boost::detail::function::function_buffer&, boost::shared_ptr<velodyne_msgs::VelodyneScan_<std::allocator<void> > const> const&) ()
   from /home/min12/k5_adm_3/devel/lib//libcloud_nodelet.so
#11 0x00007f7b8edc1d15 in boost::function1<void, boost::shared_ptr<velodyne_msgs::VelodyneScan_<std::allocator<void> > const> const&>::operator()(boost::shared_ptr<velodyne_msgs::VelodyneScan_<std::allocator<void> > const> const&) const ()
   from /home/min12/k5_adm_3/devel/lib//libcloud_nodelet.so
#12 0x00007f7b8edc05ca in boost::detail::function::void_function_obj_invoker1<boost::function<void (boost::shared_ptr<velodyne_msgs::VelodyneScan_<std::allocato---Type <return> to continue, or q <return> to quit---
r<void> > const> const&)>, void, boost::shared_ptr<velodyne_msgs::VelodyneScan_<std::allocator<void> > const> >::invoke(boost::detail::function::function_buffer&, boost::shared_ptr<velodyne_msgs::VelodyneScan_<std::allocator<void> > const>) () from /home/min12/k5_adm_3/devel/lib//libcloud_nodelet.so
[... cut ...]
Is the information above sufficient to figure out what the issue is?

I will be making the edit that you asked and run the driver again to see if that changes things. I could alternatively catkin_make with ```catkin_make -DCMAKE_BUILD_TYPE=Debug``` option which should set the ```-g``` flag while compile and see the result of running the gdb on the core dump file.

I am new to gdb so do let me know if this helps or should i just try to run with the code change that you suggested and run gdb over the core dumps if it crashes.

Thanks!

jack-oquin, 2015-11-01:
Yes. The assertion in frame 3 is the same one that caused the crash in #62. The temporary patch above should remove that crash, which comes from non-data packets somehow getting mixed in with the port 2368 data. One way that happens appears to be a bug in the PCAP reader, but there may be other causes, too.

We will definitely remove that assert(), but I am not yet sure what the best recovery is when the driver sees a mis-formated data block. Right now, I am leaning towards skipping the entire packet. It's either the wrong type, or it has been mangled. We certainly do not want the driver to crash in that situation.

konduri, 2015-11-02:
@jack-oquin  For info, will be putting in the patch and posting the results here.

jack-oquin, 2015-11-02:
I created a separate ``vlp16_fixes`` branch for working on these issues. I recommend using that as a base for further testing, instead of the patch above.

It only warns of bad packets at most once a minute, so as not to fill up your log with warnings.

nxydes, 2015-11-03:
Jack, I wanted to get some wireshark logs of the data during a crash before implementing the fix. As it is I pinpointed that a crash happened somewhere at the end of these two pcap files which I uploaded to the same dropbox folder (Currently uploading, should be about 15 mins):

https://www.dropbox.com/sh/5hu6bk0cqsouq9p/AAApR9Ltm6TDLonqjWLWOA-pa?dl=0

I know the crash happens either at the end of the first file or the beginning of the second. 

Both files have a number of packets with odd file sizes:

[image]

[image]

I can definitely see how one of those packets could have a bad header which throw that assert error. I wanted to replay the wireshark data with tcpreplay and see if the ros drivers would crash but the rosdrivers don't seem to recognize the packets. Is there a specific way to replay the data?

Now, I have loaded the assert replacement in the vlp16_fixes branch onto the machine with the troublesome VLP16 and will see if it crashes any more.

Thanks for your help!
Nick

nxydes, 2015-11-03:
By comparing our rosbag data with the tshark data, I am 90% sure I found the packet that crashed the drivers:

[image]

103932	115.718506000	192.168.1.201	255.255.255.255	UDP	1152	Source port: opentable  Destination port: opentable [BAD UDP LENGTH 1214 > IP PAYLOAD LENGTH]

This is in the file velodyne_packets_00157_20151102162631

jack-oquin, 2015-11-03:
Thanks, @nxydes, that's a big help. I'll take a closer look tomorrow and see if the driver needs some more defensive code for truncated packets.

jack-oquin, 2015-11-03:
While ``InputSocket::getPacket()`` is carefully checks for a correct packet size from the device, there is no similar defensive code in ``InputPCAP::getPacket()``.

The PCAP reader was originally just a testing aid, so it is not check the input as carefully as it should.

jack-oquin, 2015-11-03:
@nxydes: I can see the truncated frame 103932 you noticed. It contains valid-looking port 2368 data, except that the length in the UDP header (1214) is different from the payload size (1118), leaving approximately one data block missing at the end.

The socket driver *should* [skip packets like that](https://github.com/ros-drivers/velodyne/blob/master/velodyne_driver/src/lib/input.cc#L164), but the PCAP reader class does not.

So, I would expect problems when processing these packets from the PCAP, but not when running with the live device -- unless there is some subtle problem not evident from reading the ``input.cc`` implementation.

nxydes, 2015-11-03:
Jack, it seems likely I got the timing off. It is hard to compare two sets of data with different start times at that precision. I was hoping I was withing 2 seconds or so and I was hoping that packet had a header issue.

Regardless, our testing continues. I enabled debug with catkin_make -DCMAKE_BUILD_TYPE=Debug to hopefully gain more knowledge about what is happening. We re-enabled your temporary fix this morning and will see how it goes.

jack-oquin, 2015-11-03:
Sounds good, Nick. 

While I expect the driver to skip truncated packets, there are no explicit test cases for that situation, and there could still be undetected bugs. If there is another core dump, we should learn something from it.

nxydes, 2015-11-03:
So far we have not had a crash on the troublesome unit after the fix, however, it really will take many hours to confirm.

I do have a question about your implementation though. When you do the check of the header at line 300:

`      if (UPPER_BANK != raw->blocks[block].header) {`

What exactly is this checking? Are you checking for a ff:ee: which is at the start of each velodyne data block: http://velodynelidar.com/docs/manuals/VLP-16%20User%20Manual%20and%20Programming%20Guide%2063-9243%20Rev%20A.pdf

or are you checking the udp header? I noticed in the wireshark logs the ff:ee will not be in the expected location on the truncated or larger than expected packets.

jack-oquin, 2015-11-03:
[Upper bank is 0xeeff](https://github.com/ros-drivers/velodyne/blob/master/velodyne_pointcloud/include/velodyne_pointcloud/rawdata.h#L58), which is the little-endian two-byte representation of the header field.

Ideally, that should use some kind of byte-reversal interface like ``ntohs()``, as documented in #64, but the driver is not currently portable to big-endian platforms, and I don't plan to work on that until someone has a big-endian platform use case for testing it, although it would not hurt to fix this particular instance, since we are aware of it.

The 0xffee tag immediately follows the UDP header and repeats every 100 bytes. It would likely be missing at the end of any truncated packet, but I'd rather rely on an explicit check of the packet length, as is done when reading directly from the live socket interface.

nxydes, 2015-11-04:
Thanks for the explanation Jack. I ask because I see in all of the truncated or incorrectly sized UDP packets, there are either extra readings or less readings than 100 between 0xffee tags. This causes future 0xffee tags to be incorrectly located. 

Some good news on our front, we had 2 machines acting up last night, put the patch on them both and haven't had another failure since. We now have it on 3 active machines and will see how they perform tonight.

Once again, thanks for your help in debugging this!

nxydes, 2015-11-04:
Jack we did have 1 drivers crash last night. However, it did not have the same signature as the previous many crashes leading me to believe it was not caused by the same error. Unfortunately, it also didn't produce a core dump file.

This crash was weird because the nodelet manager was still showing up in rosnode list, but was unpingable and none of the related processes existed in a ps aux. After relaunching it, the pointcloud was being published but the driver nodelet would die after 15 seconds and put this in its log:

^[[0m[ INFO] [1446618201.351578154]: Loading nodelet /driver_nodelet of type velodyne_driver/DriverNodelet to manager velodyne_nodelet_manager with the following remappings:^[[0m
^[[32m[DEBUG] [1446618201.353555737]: Waiting for service velodyne_nodelet_manager/load_nodelet to be available...^[[0m
^[[0m[ INFO] [1446618214.469915987]: Bond broken, exiting^[[0m

Anyways, restarting all of ROS helped this bug. If it crops up again we will look into it more but for now looks like the fix for the headers seems to be working

jack-oquin, 2015-11-04:
Thanks for testing, @nxydes. 

If you found yet another, different bug, please open a new issue for it. I really want to get this driver working reliably.

nxydes, 2015-11-05:
@jack-oquin 

I am happy to report that it seems to be working for us. This is from the troublesome unit:

```
^[[0m[ INFO] [1446619151.905274348]: laser_ring[11] = 13, angle = 0.191986^[[0m
^[[0m[ INFO] [1446619151.905300050]: laser_ring[13] = 14, angle = 0.226893^[[0m
^[[0m[ INFO] [1446619151.905321819]: laser_ring[15] = 15, angle = 0.261799^[[0m
^[[0m[ INFO] [1446619151.905604896]: Data will be processed as a VLP-16...num_lasers: 16^[[0m
^[[0m[ INFO] [1446619151.919762077]: Reconfigure Request^[[0m
^[[0m[ INFO] [1446619152.132514380]: Got a subscriber to scan, starting subscriber to pointcloud^[[0m
^[[0m[ INFO] [1446619152.557733571]: Got a subscriber to scan, starting subscriber to pointcloud^[[0m
^[[33m[ WARN] [1446621735.464572275]: skipping invalid VLP-16 packet: block 3 header value is 3505^[[0m
^[[33m[ WARN] [1446624136.910236566]: skipping invalid VLP-16 packet: block 1 header value is 15242^[[0m
^[[33m[ WARN] [1446625689.886501681]: skipping invalid VLP-16 packet: block 4 header value is 3005^[[0m
^[[33m[ WARN] [1446628480.769522331]: skipping invalid VLP-16 packet: block 10 header value is 5385^[[0m
^[[33m[ WARN] [1446629580.491524862]: skipping invalid VLP-16 packet: block 1 header value is 3371^[[0m
^[[33m[ WARN] [1446637238.088027400]: skipping invalid VLP-16 packet: block 1 header value is 0^[[0m
^[[33m[ WARN] [1446638158.831741241]: skipping invalid VLP-16 packet: block 11 header value is 32508^[[0m
^[[33m[ WARN] [1446641142.184375492]: skipping invalid VLP-16 packet: block 5 header value is 2340^[[0m
^[[33m[ WARN] [1446641397.793182524]: skipping invalid VLP-16 packet: block 8 header value is 2420^[[0m
^[[33m[ WARN] [1446641495.304353085]: skipping invalid VLP-16 packet: block 7 header value is 27768^[[0m
^[[33m[ WARN] [1446641710.764724014]: skipping invalid VLP-16 packet: block 2 header value is 1141^[[0m
^[[33m[ WARN] [1446643128.166876978]: skipping invalid VLP-16 packet: block 11 header value is 4927^[[0m
^[[33m[ WARN] [1446692604.330010024]: skipping invalid VLP-16 packet: block 9 header value is 4427^[[0m
^[[33m[ WARN] [1446702262.238585292]: skipping invalid VLP-16 packet: block 1 header value is 2355^[[0m
^[[33m[ WARN] [1446703298.836485412]: skipping invalid VLP-16 packet: block 2 header value is 7719^[[0m
^[[33m[ WARN] [1446703981.373283889]: skipping invalid VLP-16 packet: block 4 header value is 3580^[[0m
^[[33m[ WARN] [1446708273.546500004]: skipping invalid VLP-16 packet: block 5 header value is 3027^[[0m
^[[33m[ WARN] [1446709444.451234438]: skipping invalid VLP-16 packet: block 2 header value is 4738^[[0m
^[[33m[ WARN] [1446711391.843466018]: skipping invalid VLP-16 packet: block 3 header value is 3884^[[0m
^[[33m[ WARN] [144671505
```

"Was ulimit -c unlimited set in your shell, but still no core dump?"

Not sure, we think we may have launched the node improperly. If that issue comes up again we will open a ticket.

nxydes, 2015-11-09:
@jack-oquin  After the last week of testing we consider this fix to be stable and are merging it into our master branches. Thank you again for your prompt assistance! I will be sure to open new tickets if I can track down any other issues.

closed 2015-11-09 by nxydes

konduri, 2015-11-09:
Yeah, Thanks a lot guys!!