ros-drivers/velodyne #64: make driver protable to big-endian machines

jack-oquin, 2015-11-01:
There are several places, mostly in unpacking raw data, where little-endian data access is assumed. 

Eliminating those dependencies does not look difficult, but I only plan to do that work if someone with a big-endian platform actually requests it. Otherwise, it would not get adequate testing, so why bother?

jlblancoc, 2015-11-03:
We plan to test a VLP16 under an ARM platform (Debian armhf). I **think** most ARM platforms are already little endian nowadays, so the extra work should not be justified. 
In any case, what are the chances of using a Velodyne sensor in a big endian system? How many people use a MIPS?

BTW: we could provide feedback on armhf if that is helpful.

jack-oquin, 2015-11-03:
Feedback on ARM would be great. 

It should be relatively simple to make it portable to big-endian, as long as we have a reasonable test platform for it.

JWhitleyWork, 2022-12-04:
Seeing as several places I have worked have used this driver without issue on ARMv8 machines, I don't think this requires any changes. Closing.

closed 2022-12-04 by JWhitleyWork