ros-drivers/velodyne #69: velodyne_pointcloud fails to compile with ROS indigo - VelodyneConfigConfig.h not found

DavidB-CMU, 2016-01-15:
I'm trying to compile the latest `velodyne` stack from source so I can try the VLP-16. 
The package `velodyne_pointcloud` fails to compile because the `VelodyneConfigConfig.h` file is not generated and then can't be found: 

```
No such file or directory
[/home/USERNAME/catkin_workspaces/velodyne_ws/devel/include/velodyne_pointcloud
                                                                   /VelodyneConfigConfig.h] 
```

Thanks in advance.

```bash
<snip> 

-- Boost version: 1.54.0                                                                                                                       
-- Found the following Boost libraries:                                                                                                        
--   signals                                                                                                                                   
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.26")                                                                                 
-- checking for module 'yaml-cpp'                                                                                                              
--   found yaml-cpp, version 0.5.1                                                                                                             
-- Configuring done                                                                                                                            
-- Generating done                                                                                                                             
-- Build files have been written to: /home/USERNAME/catkin_workspaces/velodyne_ws/build/velodyne_pointcloud                                    
[velodyne_pointcloud] <== '/home/USERNAME/catkin_workspaces/velodyne_ws/build/velodyne_pointcloud/build_env.sh /usr/bin/cmake /home/USERNAME/catkin_workspaces/velodyne_ws/src/velodyne/velodyne_pointcloud -DCATKIN_DEVEL_PREFIX=/home/USERNAME/catkin_workspaces/velodyne_ws/devel -DCMAKE_INSTALL_PREFIX=/home/USERNAME/catkin_workspaces/velodyne_ws/install -DCMAKE_BUILD_TYPE=RelWithDebInfo' finished with return code '0' 
[velodyne_pointcloud] ==> '/home/USERNAME/catkin_workspaces/velodyne_ws/build/velodyne_pointcloud/build_env.sh /usr/bin/make --jobserver-fds=3,6 -j' in '/home/USERNAME/catkin_workspaces/velodyne_ws/build/velodyne_pointcloud' 
Scanning dependencies of target velodyne_pointcloud_gencfg                                                                                     
Scanning dependencies of target velodyne_rawdata                                                                                               
Scanning dependencies of target ringcolors_node                                                                                                
Scanning dependencies of target ringcolors_nodelet    
                                                                                         
[  6%] Generating dynamic reconfigure files from cfg/VelodyneConfig.cfg: /home/USERNAME/catkin_workspaces/velodyne_ws/devel/include/velodyne_pointcloud/VelodyneConfigConfig.h /home/USERNAME/catkin_workspaces/velodyne_ws/devel/lib/python2.7/dist-packages/velodyne_pointcloud/cfg/VelodyneConfigConfig.py 

/usr/bin/env: python                                                                                                                           
: No such file or directory                                                                                                                    
make[2]: *** [/home/USERNAME/catkin_workspaces/velodyne_ws/devel/include/velodyne_pointcloud/VelodyneConfigConfig.h] Error 127     
            
make[1]: *** [CMakeFiles/velodyne_pointcloud_gencfg.dir/all] Error 2                                                                           
make[1]: *** Waiting for unfinished jobs....                                                                                                   
[ 20%] [ 20%] Building CXX object src/lib/CMakeFiles/velodyne_rawdata.dir/rawdata.cc.o                                                         
Building CXX object src/lib/CMakeFiles/velodyne_rawdata.dir/calibration.cc.o                                                                   

<snip>
```

DavidB-CMU, 2016-01-15:
Solved, same solution as here: 
http://answers.ros.org/question/195512/generate_dynamic_reconfigure_options-stopped-generating-h-files/

There was a problem with line-ending characters in the `velodyne_pointcloud/cfg/VelodyneConfig.cfg` file. 
The solution: 
> $ sudo apt-get install dos2unix
> $ dos2unix VelodyneConfig.cfg 

and then catkin_make was successful.

trainman419, 2016-01-15:
This is a valid bug, and should ld be fixed in the source.

jack-oquin, 2016-01-15:
I agree. Thanks for the report.

jack-oquin, 2016-03-18:
I don't see any DOS line-ending characters in my clone.

Any idea how they got there?

DavidB-CMU, 2016-03-18:
Hi Jack, I'm thinking now that it was probably an issue with the computer I was using, or when transferring some files.  
We can probably close this issue.  
Thanks.

jack-oquin, 2016-03-19:
OK with me. If someone can reproduce the issue with a fresh clone, please re-open this.

closed 2016-03-19 by jack-oquin