ros-drivers/velodyne #70: Does this ROS packages(velodyne) works for VLP-16?

won13y, 2016-01-20:
I bought VLP-16 recently.

I am about to test it with velodyne driver as this github provides.

I have a question: Does this ROS packages(velodyne) works for VLP-16??

jlblancoc, 2016-01-20:
Yes it does. Try the launch file named vlp16.

-----
Escrito desde Zapatófono, disculpe la brevedad.

jack-oquin, 2016-01-20:
Note that you still need to build it from source. 

Binary releases for Indigo and Jade are still pending. Some time soon, I hope.

hongliuuuu, 2016-02-10:
@jlblancoc Hi, I lanched the file vlp16, but there is no data pubulished on the topic /velodyne_points, do you know why? Through wireShark, I'm sure the network setting is right, thanks.

jlblancoc, 2016-02-11:
I don't know how to help, Hongliu... if you are directly watching the output in RViz, make sure to switch the global tf reference to `velodyne`, otherwise, the 3D view will remain blank even if the sensor and the node are both working ok.

jack-oquin, 2016-02-11:
@hongliuuuu: your question can better be answered on answers.ros.org. Be sure to add the ``velodyne`` tag, so the right people see it.

hongliuuuu, 2016-02-11:
Thanks a lot, problem solved! Again, thank you very much!

Best wishes,

Léo

yumeng12, 2016-03-01:
I met a similar problem.When I enter "rosrun velodyne_pointcloud gen_calibration.py32db.xml",,the screen display 
'converting "32db.xml" to "32db.yaml"
gen_calibration.py: unable to read 32db.xml'.
I know that "32db.xml" is not correct file,but I can't find correct file.Then I find a similar file in VLP-16 datas.However,the problem still exists .I lanched the file vlp16, but there is no data pubulished on any topics.

jack-oquin, 2016-03-01:
@yumeng12: this issue is about the VLP-16.

If you have a question about the HD-32E, please ask on answers.ros.org with tag ``velodyne``. I will try to answer it.

closed 2016-03-04 by jack-oquin

hcv1027, 2017-08-02:
HI @hongliuuuu , I experienced the same problem like you, can you share me your solution?