ros-drivers/velodyne #71: Do the VLP-16 axes in Rviz match the axes in the User Manual?

DavidB-CMU, 2016-01-29:
Do the cartesian axes of the PointCloud from the VLP-16 (as seen in Rviz) match the axes shown in Velodyne's User Manual? 
For me it looks to be different. 

http://velodynelidar.com/docs/manuals/VLP-16%20User%20Manual%20and%20Programming%20Guide%2063-9243%20Rev%20A.pdf

Here are the x,y,z axes specified in the User Manual: 
[image]

This is those same axes in relation to the real sensor: 
[image]

With the sensor in this same position, when I look at the PointCloud in Rviz, the axes of the /velodyne frame appear to be different by 1 rotation of 90 degrees. 
You can see the x-axis and y-axis are in a different position: 
[image]

jack-oquin, 2016-01-29:
You are correct: this driver's coordinates do *not* match the diagram in the the VLP-16 manual.

The driver translates points into X, Y, Z coordinates the same way it does for other Velodyne models. It follows the standard ROS coordinate conventions from [REP-0103](http://www.ros.org/reps/rep-0103.html#axis-orientation) for axis orientation, so the X axis points to the "front" of the device.

Is that what you are seeing? Would you prefer it to work differently?

DavidB-CMU, 2016-01-29:
Ok yes so I'm seeing the correct axes as per the ROS convention for cameras.

I don't mind, software is easy to modify, but it could be useful to add that to the documentation. 
Thanks Jack. I can close this ticket.

closed 2016-01-29 by DavidB-CMU

jack-oquin, 2016-01-29:
If you prefer to use the manufacturer's coordinate convention, it's easy to define a TF frame of reference that is 90 degrees to the right of the ``velodyne`` frame of reference.

Kaiser1401, 2017-03-22:
I have a related question to this.

Is this 'Coordinate-System-Change'  only donw for the Velodyne 16? The axes from the 32 seems to match with the ones in the manual. I couldn't find a different handling of 16 and 32 though in the code yet.

jack-oquin, 2017-03-28:
All models should work the same.

snehagn, 2017-10-18:
I just wanted to make sure I am understanding this right: the ROS driver publishes the data in a FLU (x - front, y - left, z -up) frame? So, if I am facing the LiDAR down toward the earth, that means the information from the ROS driver is giving me: x - front, y - right, z - down? Thanks!