ros-drivers/velodyne #72: Velodyne PCAP player "ghosting" problem

plepej, 2016-02-05:
Dear,

this is my first pose so please excuse me any irregularities. I am using velodyne_pointcloud 32e_points.launch with pcap argument to replay data captured by VLP-32E. I am using pierriko code to receive real time stamp, anyway the problem of "ghosting" is the same in any branch. I will try to show this problem in pictures:

[image]
[image]
[image]
[image]

This pcap has been captured on a moving vehicle, if you look at the pillar on the pictures you see in first two frames there is one pillar and in third and fourth a second pillar appears, as I notices this appears multiple times over the whole file. The frames are sequential. If the same PCAP file is played with the VeloView there is no "ghosting" problem. There is only one pillar through all frames.

Did anyone notice this problem, is there a solution to this? 

Thank You!

jack-oquin, 2016-04-11:
Do you think this could be related to #15? 

That issue has a long discussion, but the basic point is the need to transform successive moving data frames into a non-moving frame of reference, typically ``/odom`` (or perhaps ``/map``).  The [TransformNodelet](http://wiki.ros.org/velodyne_pointcloud#TransformNodelet) is provided for that. It depends on other parts of your system publishing odometry and the appropriate transforms from ``/odom`` to ``/base_link`` and to ``/velodyne``.

jack-oquin, 2016-06-09:
If there is no response regarding this issue, it should be closed.

closed 2016-06-12 by plepej