ros-drivers/velodyne #79: Error in min/max intensity values in gen_calibration.py converting from xml to yaml

selitvin, 2016-03-07:
For the case where first non zero minIntensity_ value appears in the middle of the list, it will be errornously placed as the minIntensity of the first laser in the array. For example:
```
<minIntensity_>
   <count>64</count>
   <item_version>0</item_version>
   <item>0</item>
   <item>0</item>
   <item>10</item> 
```
The yaml-file will have 10 as the attribute of laser 0 and not 2.

This is happening because the script calls addLaserCalibration(index, ...). If the calibration dictionary is still empty, the value is appended and not inserted by laser index.

jack-oquin, 2016-03-12:
Thanks for the report.

Do you have a test case that demonstrates the bug? Have you experimented with any fixes?

selitvin, 2016-03-12:
I was writing a similar code and then comparing my calibration data in yaml
vs the my data in xml and saw the difference. I don't have a code example I
could share at the moment.

jack-oquin, 2016-03-12:
Test data with detailed instructions for demonstrating the bug, would be enormously helpful.

selitvin, 2016-03-13:
Attached is a tar ball.
Notice, in the xml, minIntensity is 0 for multiple lasers at the beginning
of the list. In yaml, however, the first laser has the minIntensity set to
30. The value of 30 belongs to a laser with index of 30+.

(I ran the xml->yaml conversion:

python gen_calibration.py calib.xml)

Regards,

- Yevgeni

jack-oquin, 2016-03-13:
The attachment did not make it through the github message system.

Could you send it directly to me: ``jack.oquin at gmail.com``?

colinjing, 2017-02-07:
change the (if value != 0/255:) =>（if value != 256:）
indent   index += 1 
Maybe it works~

minIntensities = db.find('DB/minIntensity_')
if minIntensities != None:
    index = 0
    for el in minIntensities:
        if el.tag == 'item':
            if enabled_lasers[index]:
                value = int(el.text)
                if value != 256:
                    addLaserCalibration(index, 'min_intensity', value)
                index += 1

# add maximum laser intensities
maxIntensities = db.find('DB/maxIntensity_')
if maxIntensities != None:
    index = 0
    for el in maxIntensities:
        if el.tag == 'item':
            if enabled_lasers[index]:
                value = int(el.text)
                if value != 256:
                    addLaserCalibration(index, 'max_intensity', value)
                index += 1