ros-drivers/velodyne #83: Velodyne point could node doesn't work

srirambadri, 2016-03-15:
$ rosrun velodyne_pointcloud cloud_node _calibration:=32db.yaml
[ INFO] [1458049427.031892132]: correction angles: 32db.yaml
YAML Exception: yaml-cpp: error at line 0, column 0: bad conversion
[ERROR] [1458049427.036928149]: Unable to open calibration file: 32db.yaml
[ INFO] [1458049427.043593822]: Reconfigure Request

jack-oquin, 2016-03-18:
Where is your calibration file? Is there a ``32db.yaml`` in the directory from which you are running the cloud node?

jack-oquin, 2016-03-21:
This looks similar to another recent bug, see #84.

Please report whether the latest master fixes your problem.

srirambadri, 2016-03-22:
Hi Jack,

Thanks a lot. Ya the error is fixed now.

jack-oquin, 2016-03-22:
Good. Thanks for verifying it.

closed 2016-03-22 by jack-oquin