ros-drivers/velodyne #84: Unable to open calibration file  VLP16db.yaml

onlytailei, 2016-03-16:
Unable to open calibration file: /home/tyler/catkin_ws/src/velodyne/velodyne_pointcloud/params/VLP16db.yaml

That's why?

srirambadri, 2016-03-16:
I had the same problem with the HDL32db.yaml file

onlytailei, 2016-03-16:
By the way, I use ubuntu 14.04 and ros indigo

onlytailei, 2016-03-16:
@srirambadri 
Hi, I tried to install this pkg in Ubuntu 12.04 with Ros Hydro. Fortunately, it worked. So I guess it should be the problem of yaml-cpp version.

srirambadri, 2016-03-16:
@onlytailei 
hi, 
thanks.

jack-oquin, 2016-03-18:
What is the question? Is this a bug report?

If so, what did you do, and what went wrong?

jack-oquin, 2016-03-19:
OK, this is a bug, and I can reproduce it. It was apparently caused by some recent updates to the YAML file reader.

As a workaround, try checking out commit 79c8d82, which precedes those changes.

jack-oquin, 2016-03-19:
Let me know if that version works for you.

jack-oquin, 2016-03-19:
xref: #82, #80

jack-oquin, 2016-03-20:
@onlytailei and @srirambadri: Please check out the latest master containing fbe603a. I think it should fix your problem.

@wjwwood: Please code review this change. It seems to work again for the 32E, VLP-16 and the older 64E (with integer intensities), but fails to read the 64E_s2.1 calibration (containing float intensities).

andrepfr, 2016-03-21:
@jack-oquin: Thanks for the fix. Worked for me with the VLP-16.

onlytailei, 2016-03-21:
Thanks! I will check it soon.

jack-oquin, 2016-03-21:
@wjwwood: The gtest added by c3e5178 demonstrates the failure of my latest attempt to handle your test case and also the example HDL-64 S2.1 YAML configuration. Both contain float intensities, so my latest fix does not solve your original problem.

I am still looking for a solution that works for both. That would be simpler if we only had to support one yaml-cpp API.

wjwwood, 2016-03-21:
Hmm, sorry this change has caused so many issues. @jack-oquin are there unit tests that fail in each configuration? I can pretty easily switch between yaml-cpp 0.3 and yaml-cpp 0.5.

I could also try to change the yaml-cpp rosdep rule for OS X, now that indigo is the oldest supported platform. But the question is which is more work, fixing this problem or changing the rule.

jack-oquin, 2016-03-21:
@wjwwood: No worries. Besides, I think I have fixed it now. 

YAML has a scalar type, but does not directly distinguish ``float`` from ``int``. So, my solution (9a5d583) is to always read intensities into a ``float``, then convert to the desired ``int``. 

That works for me with the 0.5 API. I think it should work with 0.3 as well. Please try it on OSX to verify.

My unit test coverage is adequate for this now, it includes both ``float`` and ``int`` intensity values. Just run: 
```
make run_tests_velodyne_pointcloud_gtest
```

wjwwood, 2016-03-21:
Works for me on master. Thanks @jack-oquin.

jack-oquin, 2016-03-21:
@wjwwood: Your change led to several significant improvements for both continuous integration and unit test coverage. And, you fixed a long-standing bug that I had not looked at at all.

I appreciate your contribution.

closed 2016-03-21 by jack-oquin