ros-drivers/velodyne #85: Connect two HDL 32e together

srirambadri, 2016-03-16:
Hey,

I have two HDL 32e velodyne laser scanners and I connected them using a switch to my computer.

I ran the velodyne package and I'm getting the pointcloud.

The problem is both the pointclouds of the different HDL's are getting combined.

Is there a possibility to seperate the pointclouds?

jack-oquin, 2016-03-18:
As already answered on answers.ros.org, you can use the ``device_ip`` parameter.

Did that solve your problem?

srirambadri, 2016-03-22:
thanks, I use this to separate the pointclouds

jack-oquin, 2016-03-22:
Can you suggest where the documentation could be improved to describe that parameter better?

For example, it seems to be missing from the [velodyne_driver wiki page](http://wiki.ros.org/velodyne_driver?distro=indigo#Parameters) parameter list.

srirambadri, 2016-03-22:
As suggested in answers.ros.org, I added two parameters to the nodules_manager.launch file namely , device_ip and port.

I then set the parameters in the main launch file i.e., 32 e_points.launch.

jack-oquin, 2016-03-25:
I updated the [parameter documentation](http://wiki.ros.org/velodyne_driver#Parameters). Please review, and let me know whether it seems clear.

closed 2016-03-25 by jack-oquin

srirambadri, 2016-03-25:
Perfect. thanks