ros-drivers/velodyne #90: add new parameter arguments to various launch files

jack-oquin, 2016-04-29:
At least these parameters are missing from some of the launch scripts:

 * port
 * device_ip
 * rpm

There may be others.

Affected scripts include:

 * ``velodyne_driver/launch/nodelet_manager.launch``
 * ``velodyne_pointcloud/launch/VLP16_points.launch``
 * ``velodyne_pointcloud/launch/32e_points.launch``

There are probably others.

ZepherusFF, 2016-05-08:
Hi Jack, Thanks for the answer/opening this up. 

I think I'm gonna need more details x) 

Should I setup my launch file like this : 
<launch>

  
  <arg name="pcap" default="" />
  <arg name="calibration" default="$(find velodyne_pointcloud)/params/VLP16db.yaml"/>
  <arg name="min_range" default="0.4" />
  <arg name="max_range" default="130.0" />

  
  <include file="$(find velodyne_driver)/launch/nodelet_manager.launch">
    <arg name="model" value="VLP16"/>
    <arg name="pcap" value="$(arg pcap)"/>
**   <arg name="return_type" value="Dual" />
   <arg name="rpm" value="1200" />
   <arg name="port" value="myport_number" />
   <arg name="IP" value="my_ip_number" />
**
  </include>

  
  <include file="$(find velodyne_pointcloud)/launch/cloud_nodelet.launch">
    <arg name="calibration" value="$(arg calibration)"/>
    <arg name="min_range" value="$(arg min_range)"/>
    <arg name="max_range" value="$(arg max_range)"/>
  </include>

</launch>

and my nodelet manager like this : 

<launch>

  
  <node pkg="nodelet" type="nodelet" name="velodyne_nodelet_manager"
        args="manager" />
  <arg name="model" default="64E" />
  <arg name="pcap" default="" />
  <arg name="read_once" default="false" />
  <arg name="read_fast" default="false" />
  <arg name="repeat_delay" default="0.0" />
**  <arg name="return_type" default="Dual" />
  <arg name="rpm" default="1200" />
  <arg name="port" default="2378" />
  <arg name="IP" default="192.168.1.201"/>**
  <arg name="frame_id" default="velodyne" />
  <node pkg="nodelet" type="nodelet" name="driver_nodelet"
        args="load velodyne_driver/DriverNodelet velodyne_nodelet_manager" >
    <param name="model" value="$(arg model)"/>
    <param name="pcap" value="$(arg pcap)"/>
    <param name="read_once" value="$(arg read_once)"/>
    <param name="read_fast" value="$(arg read_fast)"/>
    <param name="repeat_delay" value="$(arg repeat_delay)"/>
    <param name="rpm" value="$(arg rpm)"/>
    <param name="frame_id" value="$(arg frame_id)"/>
  </node>    

</launch>

jack-oquin, 2016-05-09:
That does not look exactly right. 

The args have to be declared first and them passed to the appropriate nodes as params, or include files as arguments.

see: http://wiki.ros.org/roslaunch/XML/arg

jack-oquin, 2016-05-09:
You can go ahead and submit a pull request and we can discuss the details there.

ZepherusFF, 2016-05-23:
Hi jack, sorry for the late reply, I've been swamped... 

Do we want them to be passed as arguments ? In that case I need to define them before the <--! start nodelet --> part if I understand correctly.

jack-oquin, 2016-05-23:
No problem. I've been swamped, too.

Yes, I think we want to add at least ``port``, ``device_id`` and ``rpm`` as launch file arguments, passing them through to nodes and embedded launch scripts as needed.

jack-oquin, 2016-08-26:
This is fixed now, thanks to PR #104 from @altrouge.

closed 2016-08-26 by jack-oquin