ros-drivers/velodyne #93: Kinetic release needed

jack-oquin, 2016-06-15:
Since Ubuntu Xenial is not supported by either Indigo or Jade, we need a Kinetic release.

See: http://answers.ros.org/question/236814/velodyne-package-for-ubuntu-1604-lts/

jack-oquin, 2016-06-16:
For anyone needing to run on Xenial and Kinetic: the current source ``master`` compiles and passes its unit tests on that system.

televator, 2016-08-26:
Will the fact that the master branch works for kinetic eventually lead to a new release?

jack-oquin, 2016-08-26:
Not immediately, but it certainly helps.

ksatyaki, 2016-10-12:
Are there any troubles releasing it for kinetic? I could help if that's the case.

jack-oquin, 2016-10-13:
No, I just want to release the current sources to Indigo, first. Still working on it.

Thanks for the offer to help.

fnaser, 2016-10-27:
Ubuntu 16.04 and ROS kinetic installed. With the latest master I get this error message:

`CMake Error at /home/fnaser/smartfm/catkin_ws/devel/share/velodyne_driver/cmake/velodyne_driverConfig.cmake:106 (message):
  Project 'velodyne_driver' specifies
  '/home/fnaser/smartfm/catkin_ws/src/velodyne-master/velodyne_driver/include'
  as an include dir, which is not found.  It does neither exist as an
  absolute directory nor in
  '/home/fnaser/smartfm/catkin_ws/src/velodyne-master/velodyne_driver//home/fnaser/smartfm/catkin_ws/src/velodyne-master/velodyne_driver/include'.
  Ask the maintainer 'Jack O'Quin <jack.oquin@gmail.com>, Brice Rebsamen
  <brice.rebsamen@gmail.com>' to fix it.
`

jack-oquin, 2016-10-28:
@fnaser: those files exist in my sources, and they do get installed. Any idea what's going on with your workspace? 

Would you check out and build a clean version, just to verify?

kmhallen, 2016-11-08:
Until an official release is ready, binary packages can be found here:
https://bitbucket.org/DataspeedInc/ros_binaries

pjreed, 2017-02-13:
:+1:

It would be nice to get a new release for Indigo as well, as it's been a long time since the last release and there have been a lot of changes.

jack-oquin, 2017-03-28:
Agreed. I've not had any time to work on it for quite a while.

awav, 2017-04-16:
@jack-oquin I sucessfully installed velodyne packages for Ubuntu 16.04.02 from sources, 
The only issue I met was `em` & `empy` namespaces collision, resolved by uninstalling `em` and reinstalling `empy`.

cglwn, 2017-05-26:
Is there anything I can do to help with getting this resolved? Our group's mostly on kinetic and would like to be able to install this through the ROS repos.

tstoyanov, 2017-06-11:
Hi @jack-oquin, any updates regarding the kinetic release? I can offer to help preparing the release as well. Our fork at https://github.com/OrebroUniversity/velodyne seems to work fine on kinetic. We can push also push a release from there, until the official release is ready. It would be good to have something out, so we can release our packages for kinetic as well.

trevstanhope, 2017-06-19:
@jack-oquin Any chance of adding a quick How-To in the README on building/installing the master from source for Xenial prior to getting an official release?

twailurus, 2017-07-28:
@trevstanhope @tstoyanov @cglwn 

If you want to build the master from source, use the following steps:

Tested on 
- ROS: Kinetic
- OS: 64-bit Ubuntu 16.04 (ARM)
- L4T 28.1
- Platform: NVIDIA Jetson TX2

1. 
sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys FF6D3CDA
sudo sh -c 'echo "deb [ arch=amd64 ] http://packages.dataspeedinc.com/ros/ubuntu $(lsb_release -sc) main" > /etc/apt/sources.list.d/ros-dataspeed-public.list'
sudo apt-get update
2.
sudo sh -c 'echo "yaml http://packages.dataspeedinc.com/ros/ros-public-'$ROS_DISTRO'.yaml '$ROS_DISTRO'" > /etc/ros/rosdep/sources.list.d/30-dataspeed-public-'$ROS_DISTRO'.list'
rosdep update
3.
cd ~/catkin_ws/src
git clone https://github.com/ros-drivers/velodyne.git
(sudo apt-get install libpcap0.8-dev)
cd ~/catkin_ws
catkin_make

tstoyanov, 2017-07-28:
@twailurus Thanks!

The issue for me at least is not so much in compiling and using the code on kinetic. This we have covered in our own fork for a while now. The problem is that without released binary packages through the ROS build farm we cannot release our upstream stack perception_oru. 

Anyway, it is not an urgent issue, as we will probably untangle the dependencies and release without the velodyne interfaces.

JWhitleyAStuff, 2017-09-23:
@jack-oquin - I'd like to bump this issue and offer my support as a maintainer. My company actively uses, recommends, and supports this driver as well as using all of the sensors that it supports. I'd be happy to test and build new releases.

JWhitleyAStuff, 2017-11-27:
A new release is here! v1.3.0 was pushed to public last week.

closed 2017-11-27 by JWhitleyAStuff