ros-drivers/velodyne #95: Published PointCloud is not organized

Bobrip, 2016-06-23:
rawdata.unpack() sets the height of the pointcloud to 1 writing all points into one row. 

It would be useful for a few applications to organize the data at that level to set height = calibration.num_lasers and assign the points to the corresponding row and column.

This doesn't add any computation time or storage space but allows for easy access for neighborhood relations in the cloud.

Bobrip, 2016-06-23:
I just saw there is a pull-request referencing this specific feature: #65 

In it you say 

> Not all devices can support it.

Could you explain which devices wouldn't and why?

jack-oquin, 2016-06-23:
The early 64E models, for example, have more packets for some lasers than for others. 

It's all useless if the device is in motion, anyway.

spuetz, 2016-10-12:
Hi, is there a project going on with respect to the organization of the point clouds? Otherwise, I would build a version on top of the current master version, which is suitable for all velodyne scanners.

spuetz, 2016-10-12:
@fudger Are you still interested in building one general point cloud organization version for all scanners?

fudger, 2016-10-12:
Yes, I am still interested. 

But I think it would be a good idea to make the efforts to organize the point clouds part of the general revision of the driver architecture, which was already discussed in https://github.com/ros-drivers/velodyne/pull/75#issuecomment-248692827. 

Due to the complexity of the driver, I am convinced that you can only find an optimal architecture by considering all requirements, whether it is organizing the point clouds, reading dual returns, returning spherical coordinates, etc. So I would not recommend implementing these features one by one.

The time to start the revision should be defined by @jack-oquin.

willdzeng, 2017-10-09:
Hi guys,
No follow-up on this? @acschaefer @jack-oquin

JWhitleyAStuff, 2018-01-26:
It looks like @spuetz has started work on this with 0ba83bf. This issue is very stale and since work has already begun on it, I'm closing the issue.

closed 2018-01-26 by JWhitleyAStuff

spuetz, 2018-01-26:
Please see https://github.com/ros-drivers/velodyne/pull/65