ros-drivers/velodyne #99: Matlab sample code to read calibration data in 63-HDL64ES2h HDL-64E S2 CD Users Manual.pdf is full of mispelling

elinorapricot, 2016-08-03:
There is a Matlab sample code in 63-HDL64ES2h HDL-64E S2 CD Users Manual.pdf on pages 20-22, there are a lots of sintax error regarding of misspelling commands and variable names in that, which would not be there, if the person, who wrote that code compiled that at least one time...
I am a C++ developer, who had never learnt Matlab, before this issue, I wanted to translate your matlab code to C++ with Matlab Coder, but it didn't succeed, because of errors. 
There is no description about how to use correction information which coming in every UDP packet as the last two byte. 
So I had to went through your code, learning matlab commands and debugging that code, which took a lot of time, given the fact that I had never code in Matlab, and it would took only 1 minute, if the developer of that check what he has done, before he copy that to an User Manual.

Please write a description for using the correction information or publish a correct code at least. Do not give sloppy work out of your hands, because it is reduce trust and respect towards your company.
[63-HDL64ES2h HDL-64E S2 CD Users Manual.pdf](https://github.com/ros-drivers/velodyne/files/399749/63-HDL64ES2h.HDL-64E.S2.CD.Users.Manual.pdf)

trainman419, 2016-08-03:
This driver is not maintained or supported by Velodyne; we cannot help you with issues in the manual.

I suggest you contact Velodyne support directly if you have issues with their code.

closed 2016-08-03 by elinorapricot

trainman419, 2016-08-03:
We do have a gen_calibration.py script which can parse the XML calibration file and produce our internal format; perhaps it will be useful as a reference: https://github.com/ros-drivers/velodyne/blob/master/velodyne_pointcloud/scripts/gen_calibration.py

elinorapricot, 2016-08-03:
Thank you, for your answer. :)