ros-drivers/velodyne #100: Pointcloud2 data reading with the driver

mxrchn, 2016-08-04:
hi, jack, I use the driver to read point cloud data with the command: $ rostopic echo / velodyne_points.
But I can't understand the "data" means.   So how to read x, y, z coordinates from these numbers? Thanks!
The attached file is the picture of these numbers. @jack-oquin
[image]

mxrchn, 2016-08-04:
It's velodyne VLP-16.

jack-oquin, 2016-08-04:
That topic publishes ``sensor_msgs/PointCloud2`` messages. It's not useful to view them with ``rostopic echo``. You can view the data with rviz:
```
rosrun rviz rviz -f velodyne
```

* In the "displays" panel, click "Add", then select "Point Cloud2", then press "OK".
* In the "Topic" field of the new "Point Cloud2" tab, enter "/velodyne_points".

jack-oquin, 2016-08-04:
Questions like this are better sent to answers.ros.org.

closed 2016-08-04 by jack-oquin