ros-drivers/velodyne #102: Where is the origin of coordinates

mxrchn, 2016-08-10:
When I read the rawdata.cc file in the driver, it's hard for me to understand where the origin of coordinates is. If the origin of coordinates is unclear, the calibration and final PointCloud2 published may be inaccurate. Below is the several lines of code in rawdata.cc. So can you tell me where  the origin of coordinates is, the geometric centre of subface or inside the VLP-16?
Thank you in advance! @jack-oquin
[image]

jlblancoc, 2016-08-10:
The reference frame for the read point clouds, it's **inside** the Velodyne
cylinder, approximately at the middle point of the cylinder.

jack-oquin, 2016-08-10:
Yes, that's where you should define the ``velodyne``  frame of reference, relative to your vehicle base.

mxrchn, 2016-08-11:
Is the origin of coordinates in the same height of the optical centre? [37.7]/1.485?
[image]

jlblancoc, 2016-08-11:
I think that's correct, that is the **nominal** origin of coordinates. There is a set of calibration parameters to introduce some offsets wrt that point, though.

chukcha2, 2016-08-11:
VLP-16 manual, on page 48 has more details on this but I don't completely understand it yet. Note this line "APPLY VERTICAL CORRECTION FOR GREATEST ACCURACY".

[image]

B-Leslie, 2016-09-29:
@chukcha2 , I believe each beam has its own detector, and the detectors are arranged in a vertical array. The beams are therefore focused on to different points inside the scanner, and so a small vertical offset corresponding to these different focal points (as specified on the RHS of the diagram on p. 48) should be applied to scan points based on the beam ID.

jack-oquin, 2016-11-03:
Is this question answered satisfactorily? If so, please close the issue.

siyangy, 2016-12-22:
A related question: how does the point get transformed from "velodyne" frame to "odom" frame? Didn't find a StampedTransform provided anywhere in this code repo.

jack-oquin, 2016-12-23:
It's done here: https://github.com/ros-drivers/velodyne/blob/master/velodyne_pointcloud/src/conversions/transform.cc#L104

closed 2016-12-23 by jack-oquin