ros-drivers/velodyne #106: Different results when playing back from rosbag

chiyuan-goh, 2016-08-22:
I'm having some difficulties when recording and playing back `velodyne_packets` from rosbag.

Supposedly, I used ` roslaunch velodyne_pointcloud 64e_points.launch calibration:=foo.yaml` to start the driver. When visualizing the point cloud in rviz, this is shown to work as expected. However, when playing back `/velodyne_packets` from the rosbag and running `rosrun velodyne_pointcloud cloud_node _calibration:=foo.yaml` , the pointcloud does not seem to be correct. There seems to be some phantom points around the sensor (see the red points in the screenshot). 

I'm attaching both the screenshot and the yaml file I used. Any idea what could be wrong?
[newvelo.txt](https://github.com/ros-drivers/velodyne/files/429446/newvelo.txt)

[image]

jmlago91, 2016-08-22:
When you use `roslaunch velodyne_pointcloud 64e_points.launch calibration:=foo.yaml` you are configuring the driver with the parameters in the .launch file.

However when you execute `rosrun velodyne_pointcloud cloud_node _calibration:=foo.yaml` the node is executing cleanly without a full configuration. Use a .launch file or provide parameters in the terminal (like you are doing with `_calibration:=foo.yaml`.

The Velodyne sensor that I'm working with, not provide reliable measurements below 1.62m.
That red ghost points that tou can see in rviz is consequence of publishing all points without filtering the point below a reliable limit.

Adding the `_min_range:=1` parameter (try different values to search the correct value for your sensor, or look in the configuration file that Velodyne provides with the unit) you can control the bottom limit.

jack-oquin, 2016-09-21:
Did @jmlago91 answer your question?

If so, please close it.

closed 2017-04-19 by chiyuan-goh