ros-drivers/velodyne #108: How to use two VLP16( velodyne) with just one computer at a same time

JeongJae0815, 2016-10-07:
Hey,

I have two VPL16 velodyne laser scanners and I connected using a switch to my computer.

I fixed first VPL 16's IP to 192.168.1.201 and second VPL's IP to 192.168.1.202 using Web GUI

and then I ran the velodyne package; roslaunch velodyne_pointcloud VLP16_points2_2.launch

here is the code of VLP16_points2_2.launch

<launch>

<group ns="ns1">

<arg name="pcap" default=""/>

<arg name="calibration" default="$(find velodyne_pointcloud)/params/VLP16db.yaml"/>

<arg name="min_range" default="0.4"/>

<arg name="max_range" default="130.0"/>

<arg name="device_ip" default="192.168.1.201" />
<include file="$(find velodyne_driver)/launch/nodelet_manager.launch">

<arg name="model" value="VLP16"/>

<arg name="device_ip" value="$(arg device_ip)" />

<arg name="pcap" value="$(arg pcap)"/>
</include>

<include file="$(find velodyne_pointcloud)/launch/cloud_nodelet.launch">

<arg name="calibration" value="$(arg calibration)"/>

<arg name="min_range" value="$(arg min_range)"/>

<arg name="max_range" value="$(arg max_range)"/>
</include>

</group>

<group ns="ns2">

<arg name="pcap" default=""/>

<arg name="calibration" default="$(find velodyne_pointcloud)/params/VLP16db.yaml"/>

<arg name="min_range" default="0.4"/>

<arg name="max_range" default="130.0"/>

<arg name="device_ip" default="192.168.1.202"/>

<include file="$(find velodyne_driver)/launch/nodelet_manager.launch">

<arg name="model" value="VLP16"/>

<arg name="device_ip" value="$(arg device_ip)" />

<arg name="pcap" value="$(arg pcap)"/>
</include>

<include file="$(find velodyne_pointcloud)/launch/cloud_nodelet.launch">

<arg name="calibration" value="$(arg calibration)"/>

<arg name="min_range" value="$(arg min_range)"/>

<arg name="max_range" value="$(arg max_range)"/>
</include>

</group>

</launch>

However, I just can get only one topic which has combined data of two sensors Another topic is just empty Is there anything that I gotta do to get both of data?

Thank you in a advance I really appreciate your kind

jack-oquin, 2016-10-08:
I believe there is a bug in ``velodyne_driver/launch/nodelet_manager.launch``: it unconditionally creates the nodelet manager process with the fixed name ``velodyne_nodelet_manager``.

What it should do is create ``$(arg frame_id)_nodelet_manager``. Then they'll have different names if they have different TF frame IDs (which they must, anyway).

JeongJae0815, 2016-10-09:
I really appreciate your kind.
I saw the codes that you fixed :)
I modified my source as you saying
please refer to [VLP16_points_multi.launch](https://github.com/JeongJae0815/Multi_Velodyne/tree/master/velodyne/velodyne_pointcloud/launch)
Is there nothing what I gotta do to use two velodynes?
Actually I'm in ROSCON, so I couldn't check that the code works.
Thank you in advance

JeongJae0815, 2016-10-09:
By the way, should I change <arg name="model" default="64E /> to <arg name="model" default="VLP16" />  in velodyne_driver/launch/nodelet_manager.launch
Actually I did, but I just wanna know whether it is right or not.

jack-oquin, 2016-10-09:
Yes, you need to use the correct model parameter. You can use the ``VLP16_points.launch`` script, which sets it for you..

[This new test script](https://github.com/ros-drivers/velodyne/blob/ed6e94be662e4591a41de68d5bb8d78df80e9038/velodyne_pointcloud/tests/two_nodelet_managers.test) is an example of what you can do (ignore the ``<test>`` stanzas). Although that test passes, it does not prove that the correct data get sent to the correct topic. I think it probably works, though. It uses two different device models to get more test coverage, but two or more VLP-16 devices would work similarly.

Have fun at ROSCon! Wish I could be there.

jack-oquin, 2016-10-09:
Please post a comment here when you get a chance to test this fix, whatever the results.

JeongJae0815, 2016-10-09:
Of course, I'll let you know the reaults. Maybe 8 hours later
Let's keep in touch!

JeongJae0815, 2016-10-10:
I can't build a code that you uploaded. there is error. 
I'm debugging..

JeongJae0815, 2016-10-10:
okay, Let me tell you something.
First, I excuted "rostest velodyne_pointcloud two_nodelet_manager.test", but the result is just 'failure'
so, I maede two new lauch files which are [VLP16_points1.launch](https://github.com/JeongJae0815/New_Velodyne/tree/master/velodyne/velodyne_pointcloud/launch), [VLP16_points2.launch](https://github.com/JeongJae0815/New_Velodyne/tree/master/velodyne/velodyne_pointcloud/launch)

jack-oquin, 2016-10-10:
> First, I excuted "rostest velodyne_pointcloud two_nodelet_manager.test", but the result is just 'failure'

The test name as an "s" before the dot. This works for me:
```
$ rostest velodyne_pointcloud two_nodelet_managers.test
... logging to /home/joq/.ros/log/rostest-iron-19022.log
[ROSUNIT] Outputting test results to /home/joq/.ros/test_results/velodyne_pointcloud/rostest-tests_two_nodelet_managers.xml
testmulti_nodelet_32e_test ... ok
testmulti_nodelet_vlp16_test ... ok

[ROSTEST]-----------------------------------------------------------------------

[velodyne_pointcloud.rosunit-multi_nodelet_32e_test/test_hz][passed]
[velodyne_pointcloud.rosunit-multi_nodelet_vlp16_test/test_hz][passed]

SUMMARY
 * RESULT: SUCCESS
 * TESTS: 2
 * ERRORS: 0
 * FAILURES: 0

rostest log file is in /home/joq/.ros/log/rostest-iron-19022.log
```

I can also launch it with ``roslaunch`` and observe the two output topics.

However, those unit tests depend on having previously run ``make tests`` to download the required PCAP files. Maybe that's your problem.

jack-oquin, 2016-10-10:
> Second, I made two new lauch files which are VLP16_points1.launch, VLP16_points2.launch
when I excute each of them separately, It gives me correct data each of them. I mean I can get not combined data with another
However, when I excute both of files at a same, the result just gives me only one data which started executing before another VLP16, and I can see "bind: Address already in use"

I wonder if we need to specify a different port number for the second device?

I lack access to the hardware needed for trying that myself.

trainman419, 2016-10-10:
When I've worked with multiple VLP-16s in the past, I've needed to specify a separate port number for each sensor. If both sensors use the the same port, a process listening on that port will receive all packets from both sensors.

jack-oquin, 2016-10-10:
Thanks @trainman419, that's what I suspected!

jack-oquin, 2016-10-10:
This is complicated enough to need a tutorial explaining how to do it.

JeongJae0815, 2016-10-19:
I changed my code as you said,so I could see that it works. 
I changed port of each velodynes using web, and then changed port number on VLP16.launch file
I Really Thank @jack-oquin for your kind!

jack-oquin, 2016-10-19:
Glad it's working now. I am sure others will benefit from your efforts.

jack-oquin, 2016-11-03:
The fix for this issue fails when the ``frame_id`` contains a slash, as may happen when using ``tf_prefix``.

alexwhittemore, 2017-09-27:
This has been super helpful, thank you all for taking the time! (Aside: my biggest problem was that both LIDARs were using the same UDP port, which is an issue despite that both are on different physical interfaces and different subnets.)

But now I've got a related but not identical issue: after following in your footsteps, I've got two VLP-16s publishing pointclouds on the topic /velodyne_pointcloud in two separate frames. But I believe for my application (cartographer_ros) that I need them publishing in two separate topics. It's unclear to me that there's anywhere not hardcoded in source to configure this per nodelet manager? Has anyone else run into and satisfied this requirement?

raskolnikov-reborn, 2018-03-03:
remap the topics velodyne_points and velodyne_packets in the cloud_nodelet.launch and nodelet_manager.launch and to $(arg frame_id)_points and $(arg frame_id)_packets. Make sure that you pass along the frame id argument to the respective launch files from velodyne_pointcloud.launch

alexwhittemore, 2018-03-03:
Thanks for coming back with that! I should have updated this with a sample config after I finally sorted it all out. Unfortunately I don't have it on hand anymore :(

zhang261007, 2018-05-15:
thacks a lot !!!

closed 2018-05-15 by JWhitleyAStuff

NTUTian, 2018-07-24:
Appreciate all for taking the time!
But I still have some question about change port number is mean change data port or the other?
Can it be changed to any number?
Sorry,my English isn't very good.

NTUTian, 2018-07-24:
I use @JeongJae0815 code and change the port number but when I open it the status still display error.
What did I miss?
Sorry,my English isn't very good.
Thank you in a advance

JWhitleyAStuff, 2018-07-24:
@NTUTian - I am sorry but the maintainers of this repository can not support another repository (like the one from @JeongJae0815). However, the official Velodyne repository should support many Velodynes (my company has tested up to 4 simultaneously). To do this, you must log in to the web interface on each Velodyne and change the IP address and port to something different. If you are using them all on the same switch, they must all be in the same subnet as each other and as the computer which will be connecting to them. I also recommend changing the broadcast IP on each Velodyne to the IP address of the computer that will be connecting to them. So, in summary, for each Velodyne:

- Change the Velodyne's IP address to something other than the default, but all in the same subnet.
- Change the Velodyne's port to something other than the default, but different for each Velodyne.
- Change the Velodyne's Broadcast IP to the IP of the PC.
- Create separate launch files on the PC with IP addresses and ports that match the Velodyne's IP addresses and ports.

NTUTian, 2018-07-25:
Thanks for coming back with that! 
I will try it.I will inform you if have any news

NTUTian, 2018-07-25:
I changed IP to 192.168.1.201 ,the other is 192.168.1.201 and the data port changed 2364 and 2372 in web .I also changed IP and port in VLP16_points_multi.launch.When I run the source the status is OK but show nothing.
By the way,the Broadcast IP  is mean Host IP?
Thanks for all taking time to answer me.

JeongJae0815, 2018-07-25:
you must change ur IP differently like one is 192.16.1.201 the other is 192.168.1.202 and also data port, telemetry port differently.

NTUTian, 2018-07-25:
Sorry, I made a typo. I use IP 192.168.1.201 and 192.168.1.202 but still show nothing

JeongJae0815, 2018-07-25:
Did you change telementry port? you must change this telementry port.
Can you check the VLP16_points_multi.launch file?
You must reflect the change on this file too.

NTUTian, 2018-07-25:
I already changed Data port and telementry port like this.
And I also change in web 
`<group ns="ns1">
  
  <arg name="calibration" default="$(find velodyne_pointcloud)/params/VLP16db.yaml"/>
  <arg name="device_ip" default="192.168.1.201" />
  <arg name="frame_id" default="velodyne_first" />
  <arg name="manager" default="$(arg frame_id)_nodelet_manager" />
  <arg name="max_range" default="130.0" />
  <arg name="min_range" default="0.4" />
  <arg name="pcap" default="" />
  <arg name="port" default="2368" />
  <arg name="read_fast" default="false" />
  <arg name="read_once" default="false" />
  <arg name="repeat_delay" default="0.0" />
  <arg name="rpm" default="600.0" />

  
  <include file="$(find velodyne_driver)/launch/nodelet_manager.launch">
    <arg name="device_ip" value="$(arg device_ip)"/>
    <arg name="frame_id" value="$(arg frame_id)"/>
    <arg name="manager" value="$(arg manager)" />
    <arg name="model" value="VLP16"/>
    <arg name="pcap" value="$(arg pcap)"/>
    <arg name="port" value="8300"/>
    <arg name="read_fast" value="$(arg read_fast)"/>
    <arg name="read_once" value="$(arg read_once)"/>
    <arg name="repeat_delay" value="$(arg repeat_delay)"/>
    <arg name="rpm" value="$(arg rpm)"/>
  </include>

  
  <include file="$(find velodyne_pointcloud)/launch/cloud_nodelet.launch">
    <arg name="calibration" value="$(arg calibration)"/>
    <arg name="manager" value="$(arg manager)" />
    <arg name="max_range" value="$(arg max_range)"/>
    <arg name="min_range" value="$(arg min_range)"/>
  </include>
</group>

<group ns="ns2">
  
  <arg name="calibration" default="$(find velodyne_pointcloud)/params/VLP16db.yaml"/>
  <arg name="device_ip" default="192.168.1.202" />
  <arg name="frame_id" default="velodyne_second" />
  <arg name="manager" default="$(arg frame_id)_nodelet_manager" />
  <arg name="max_range" default="130.0" />
  <arg name="min_range" default="0.4" />
  <arg name="pcap" default="" />
  <arg name="port" default="2368" />
  <arg name="read_fast" default="false" />
  <arg name="read_once" default="false" />
  <arg name="repeat_delay" default="0.0" />
  <arg name="rpm" default="600.0" />

  
  <include file="$(find velodyne_driver)/launch/nodelet_manager.launch">
    <arg name="device_ip" value="$(arg device_ip)"/>
    <arg name="frame_id" value="$(arg frame_id)"/>
    <arg name="manager" value="$(arg manager)" />
    <arg name="model" value="VLP16"/>
    <arg name="pcap" value="$(arg pcap)"/>
    <arg name="port" value="8304"/>
    <arg name="read_fast" value="$(arg read_fast)"/>
    <arg name="read_once" value="$(arg read_once)"/>
    <arg name="repeat_delay" value="$(arg repeat_delay)"/>
    <arg name="rpm" value="$(arg rpm)"/>
  </include>

  
  <include file="$(find velodyne_pointcloud)/launch/cloud_nodelet.launch">
    <arg name="calibration" value="$(arg calibration)"/>
    <arg name="manager" value="$(arg manager)" />
    <arg name="max_range" value="$(arg max_range)"/>
    <arg name="min_range" value="$(arg min_range)"/>
  </include>
</group>

</launch>`

NTUTian, 2018-07-25:
I changed data port and  telementry port like this code.
I also changed in web
```<group ns="ns1">
  <!-- declare arguments with default values -->
  <arg name="calibration" default="$(find velodyne_pointcloud)/params/VLP16db.yaml"/>
  <arg name="device_ip" default="192.168.1.201" />
  <arg name="frame_id" default="velodyne_first" />
  <arg name="manager" default="$(arg frame_id)_nodelet_manager" />
  <arg name="max_range" default="130.0" />
  <arg name="min_range" default="0.4" />
  <arg name="pcap" default="" />
  <arg name="port" default="2372" />
  <arg name="read_fast" default="false" />
  <arg name="read_once" default="false" />
  <arg name="repeat_delay" default="0.0" />
  <arg name="rpm" default="600.0" />

  <!-- start nodelet manager and driver nodelets -->
  <include file="$(find velodyne_driver)/launch/nodelet_manager.launch">
    <arg name="device_ip" value="$(arg device_ip)"/>
    <arg name="frame_id" value="$(arg frame_id)"/>
    <arg name="manager" value="$(arg manager)" />
    <arg name="model" value="VLP16"/>
    <arg name="pcap" value="$(arg pcap)"/>
    <arg name="port" value="8300"/>
    <arg name="read_fast" value="$(arg read_fast)"/>
    <arg name="read_once" value="$(arg read_once)"/>
    <arg name="repeat_delay" value="$(arg repeat_delay)"/>
    <arg name="rpm" value="$(arg rpm)"/>
  </include>

  <!-- start cloud nodelet -->
  <include file="$(find velodyne_pointcloud)/launch/cloud_nodelet.launch">
    <arg name="calibration" value="$(arg calibration)"/>
    <arg name="manager" value="$(arg manager)" />
    <arg name="max_range" value="$(arg max_range)"/>
    <arg name="min_range" value="$(arg min_range)"/>
  </include>
</group>

<!-- For Second velodyne -->

<group ns="ns2">
  <!-- declare arguments with default values -->
  <arg name="calibration" default="$(find velodyne_pointcloud)/params/VLP16db.yaml"/>
  <arg name="device_ip" default="192.168.1.202" />
  <arg name="frame_id" default="velodyne_second" />
  <arg name="manager" default="$(arg frame_id)_nodelet_manager" />
  <arg name="max_range" default="130.0" />
  <arg name="min_range" default="0.4" />
  <arg name="pcap" default="" />
  <arg name="port" default="2364" />
  <arg name="read_fast" default="false" />
  <arg name="read_once" default="false" />
  <arg name="repeat_delay" default="0.0" />
  <arg name="rpm" default="600.0" />

  <!-- start nodelet manager and driver nodelets -->
  <include file="$(find velodyne_driver)/launch/nodelet_manager.launch">
    <arg name="device_ip" value="$(arg device_ip)"/>
    <arg name="frame_id" value="$(arg frame_id)"/>
    <arg name="manager" value="$(arg manager)" />
    <arg name="model" value="VLP16"/>
    <arg name="pcap" value="$(arg pcap)"/>
    <arg name="port" value="8304"/>
    <arg name="read_fast" value="$(arg read_fast)"/>
    <arg name="read_once" value="$(arg read_once)"/>
    <arg name="repeat_delay" value="$(arg repeat_delay)"/>
    <arg name="rpm" value="$(arg rpm)"/>
  </include>

  <!-- start cloud nodelet -->
  <include file="$(find velodyne_pointcloud)/launch/cloud_nodelet.launch">
    <arg name="calibration" value="$(arg calibration)"/>
    <arg name="manager" value="$(arg manager)" />
    <arg name="max_range" value="$(arg max_range)"/>
    <arg name="min_range" value="$(arg min_range)"/>
  </include>
</group>

</launch>```

NTUTian, 2018-07-25:
Sorry guy,I lost my address IP so I can not go to interface how can I do?

NTUTian, 2018-07-25:
I can not enter the website I think I lost my address ip .What can I do to find my ip or how to  reset it?

alexwhittemore, 2018-07-25:
It's a pain - it's mentioned on page 24 of the web server manual for the VLP-16: http://velodynelidar.com/docs/manuals/63-9266%20REV%20A%20WEBSERVER%20USER%20GUIDE,HDL-32E%20&%20VLP-16.pdf

If you don't know the IP of the sensor, you have to plug it in and start WireShark on the interface. The sensor broadcasts messages whether you want to or not, so the packet capture will show a flood of sensor data on the interface. The IP it's coming from is the one you need to log in to.

NTUTian, 2018-07-26:
Thank a lot I get my IP now,

hroachewilson, 2019-05-06:
Following recommendation from @JWhitleyAStuff, I'm including the velodyne_pointcloud launch files in my launch file but am still not able to set up ros topics for two VLP16 running concurrently. One of my ros topics is always empty, corresponding to the last velodyne plugged in. Similarly, I can only log on to one sensor at a time. The error (warning) I'm getting is Velodyne poll() timeout, which suggests a network issue. My settings are:

Sensor 1:
`Network Address  192.168.1.201
Network Mask      24
Network Gateway 192.168.1.1
Host Address      192.168.1.100
Data Port            2368
Telemetry Port    8308`
Ethernet connection 1 (enp6s0):
`Address             192.168.1.100
Mask                 24
Gateway            0.0.0.0`

Sensor 2:
`Network Address  192.168.1.202
Network Mask      24
Network Gateway 192.168.1.2
Host Address      192.168.1.200
Data Port            2268
Telemetry Port    8208`
Ethernet connection 2 (enp7s0):
`Address             192.168.1.200
Mask                 24
Gateway            0.0.0.0`

Routes have been added from ethernet interfaces to the host addresses on the sensors and my launch file is below. Note that the commands I used for adding routes, as indicated in the [VLP16 setup guide](http://wiki.ros.org/velodyne/Tutorials/Getting%20Started%20with%20the%20Velodyne%20VLP16), were:

`sudo ifconfig enp6s0 192.168.3.100`
`sudo route add 192.168.105.5.`
`sudo ifconfig enp6s0 192.168.3.200`
`sudo route add 192.168.104.14.`

for my particular MAC addresses. Side note, I am not sure why the guide says to assign a second address to my interfaces (i.e. 3.100 & 3.200).

`<?xml version="1.0"?>
<launch>

  
  <arg name="velodyne_calib" default="$(find velodyne_pointcloud)/params/VLP16db.yaml"/>

<group ns="vlp_1">

    
    <include file="$(find velodyne_pointcloud)/launch/VLP16_points.launch">
      <arg name="calibration" value="$(arg velodyne_calib)"/>
      <arg name="device_ip" value="192.168.1.201"/>
      <arg name="frame_id" value="vlp201"/>
      <arg name="port" value="2368"/>
    </include>
</group>

<group ns="vlp_2">
    
    <include file="$(find velodyne_pointcloud)/launch/VLP16_points.launch">
      <arg name="calibration" value="$(arg velodyne_calib)"/>
      <arg name="device_ip" value="192.168.1.202"/>
      <arg name="frame_id" value="vlp202"/>
      <arg name="port" value="2268"/>
    </include>
</group>

</launch>`

hroachewilson, 2019-05-07:
Update. Problem is solved. Correct configuration below.

**Sensor 1:**
Network Address  192.168.1.201
Network Mask      24
Network Gateway 0.0.0.0
Host Address      192.168.1.100
Data Port            2368
Telemetry Port    8308

**Ethernet connection 1 (enp6s0):**
Address             192.168.1.100
Mask                 24
Gateway            0.0.0.0

**Sensor 2:**
Network Address  192.168.2.201
Network Mask      24
Network Gateway 0.0.0.0
Host Address      192.168.2.100
Data Port            2268
Telemetry Port    8208

**Ethernet connection 2 (enp7s0):**
Address             192.168.2.100
Mask                 24
Gateway            0.0.0.0

```<?xml version="1.0"?>
<launch>

  <!-- calibration file path -->
  <arg name="velodyne_calib" default="$(find velodyne_pointcloud)/params/VLP16db.yaml"/>

<group ns="vlp_1">

    <!-- VLP-16 front -->
    <include file="$(find velodyne_pointcloud)/launch/VLP16_points.launch">
      <arg name="calibration" value="$(arg velodyne_calib)"/>
      <arg name="device_ip" value="192.168.1.201"/>
      <arg name="frame_id" value="vlp201"/>
      <arg name="port" value="2368"/>
    </include>
</group>

<group ns="vlp_2">
    <!-- VLP-16 rear -->
    <include file="$(find velodyne_pointcloud)/launch/VLP16_points.launch">
      <arg name="calibration" value="$(arg velodyne_calib)"/>
      <arg name="device_ip" value="192.168.2.201"/>
      <arg name="frame_id" value="vlp202"/>
      <arg name="port" value="2268"/>
    </include>
</group>

</launch>

dkhanna511, 2021-11-29:
@hroachewilson How did you connect the Lidars to the different ethernet ports (enp6s0 and enp7s0) as stated above? Dis you use two systems? I am currently using an ethernet switch hub and am unable to access various ethernet ports. 

Any help is much appreciated.

himhan34, 2022-10-24:
hello. first thank you for the nice code sharing. i am also doing the same thing as you do with  two vlp 16 lidar

[image]

after following the thing that you guys done, there was no error, but i cant see mixed point clouds, 
is there any ways to see them?

[image]

my code is exactly like you.. 
<launch>

  
  <arg name="velodyne_calib" default="$(find velodyne_pointcloud)/params/VLP16db.yaml"/>

<group ns="vlp_1">

    
    <include file="$(find velodyne_pointcloud)/launch/VLP16_points.launch">
      <arg name="calibration" value="$(arg velodyne_calib)"/>
      <arg name="device_ip" value="192.168.1.201"/>
      <arg name="frame_id" value="vlp201"/>
      <arg name="port" value="2368"/>
    </include>
</group>

<group ns="vlp_2">
    
    <include file="$(find velodyne_pointcloud)/launch/VLP16_points.launch">
      <arg name="calibration" value="$(arg velodyne_calib)"/>
      <arg name="device_ip" value="192.168.2.201"/>
      <arg name="frame_id" value="vlp202"/>
      <arg name="port" value="2268"/>
    </include>
</group>

</launch>

can you help me?

JWhitleyWork, 2022-10-24:
@himhan34 Please do not bump dead threads like this. Create a new question and mention this issue if you want to reference it.

himhan34, 2022-10-24:
okey.  i undetstand.
thanks for helping me