ros-drivers/velodyne #109: Is the 'dual two point calibration' (correctly) used?

victl, 2016-12-07:
Hi, I've checked the source code 'rawdata.cc' and found the 'dual two point calibration' method documented in the Velodyne HDL-64E S2 Users Manual is embraced by a `if` clause. I have several questions here:

1. the python script that turns the `db.xml` file into YAML file does not generate the entry `two_pt_correction_available`, which means it'll be false whenever we query its value.
2. the relevant source code is:`if (corrections.two_pt_correction_available) { // ...`, but the users manual wanted this calibration to be applied only when the distance is nearer than 25.04m, should this be added to the `if` condition?
3. after the `if` clause, the code is `float distance_x = distance + distance_corr_x;`, but the variable `distance` had already been modified with `distance += corrections.dist_correction;` in a previous line. We should notice that the `distance_corr_x` (if calculated by enabling two point calibration)  shall be very close to `dist_correction`(by checking a `db.xml` file), so we might be adding distance correction for two times. This might not be the original intention of the Velodyne company?

There are rare articles/papers talking about this topic, all papers I read that discuss LiDAR calibration does not take `dist_correction_x/y` into account. Can anybody point me to an article/paper that discuss more in-depth about these two parameters set by Velodyne?
I'm trying to calibrate a HDL-64E S2 BTW.

Thanks in advance!

jack-oquin, 2016-12-08:
Thanks for your bug report. I maintain this package, but I did not write the 64E S2 code, nor do I have access to that device for testing. Its code was contributed many years ago by another user.

It sounds like you know more about that model than I do, so I would accept a pull request that makes the driver work as you think it should.

JWhitleyAStuff, 2018-12-05:
Closing with a note to add support in the future with #198.

closed 2018-12-05 by JWhitleyAStuff