ros-drivers/velodyne #112: Failed the catkin_make

andrewraharjo, 2017-01-01:
I tried to build from the source but I got this error 

Built target dynamic_reconfigure_gencfg
[  3%] Built target rosgraph_msgs_generate_messages_lisp
[  3%] Built target _velodyne_msgs_generate_messages_check_deps_VelodynePacket
[  3%] Built target _velodyne_msgs_generate_messages_check_deps_VelodyneScan
make: *** [all] Error 2
Invoking "make -j12 -l12" failed

Does anybody have encountered this build issue ? Any pointers ? I'm running 14.04 Indigo

jack-oquin, 2017-01-04:
Which version of ROS?

andrewraharjo, 2017-01-04:
Indigo

jack-oquin, 2017-01-06:
Sorry. I missed that part. I have not seen this problem. It does not fail when running Travis CI with Trusty and Indigo.

* https://travis-ci.org/ros-drivers/velodyne

Must be something wrong with your workspace or your ROS installation.

andrewraharjo, 2017-01-06:
Jack,

I'll try to rebuild the catkin_ws and see how it goes. Thanks for pointing out. I'll keep you posted

andrewraharjo, 2017-01-07:
@jack-oquin apparently there was an issue with Anaconda. Here's the step I did to resolve this:
1. Delete catkin_ws / start fresh
2. Create new workspace
3. export PYTHONPATH=$PYTHONPATH:/usr/lib/python2.7/dist-packages 
4. source ~./bashrc
5. check the import rospkg (this is the reason giving me error)
6. clone the velodyne driver under catkin_ws
7. run the catkin_make

I guess you can close this now. Thanks for the insight

jack-oquin, 2017-01-09:
No problem, glad it's working now.

closed 2017-01-09 by jack-oquin

JWhitleyAStuff, 2018-11-30:
@ajay1606 - This looks like an issue specific to Autoware - the lidar_kf_track is not part of the Velodyne driver. Please open an issue at https://github.com/CPFL/autoware/issues.