ros-drivers/velodyne #113: Coordinate System

rtpavlovsk21, 2017-01-11:
In your readme can you make a warning for swapping coordinate systems to the ROS standard? For example if the system is mounted any other way than vertical the [ROS conventions](http://www.ros.org/reps/rep-0103.html) are not consistent and differ from the [VLP16 manual](http://velodynelidar.com/docs/manuals/VLP-16%20User%20Manual%20and%20Programming%20Guide%2063-9243%20Rev%20A.pdf).

jack-oquin, 2017-01-12:
Velodyne coordinates differ from various ROS standards no matter how the device is mounted.

The documentation does mention this, but not in any README. Which one do you think should explain it?

chukcha2, 2017-01-12:
In case this is useful to put into some README or manual:

Frame of reference for the velodyne, as set up by the velodyne_driver, axes shown below:
X → Forward
Y → Left
Z → Up

[image]

However, the original velodyne coordinate frame as described by the VLP-16 manual is shown below:

[image]

jack-oquin, 2017-01-21:
Would some definition of the "device frame of reference" be useful [here](http://wiki.ros.org/velodyne_pointcloud)?

If you think so, feel free to add it, perhaps in a new section following the overview.

matheusns, 2017-01-25:
I have a VLP16 and I'll have to use it in a vertical way. How can I change the coordinates to fix it?

jack-oquin, 2017-01-26:
You just need to define the [ROS TF tree](http://wiki.ros.org/tf2) correctly. The device frame is called ``velodyne``, or whatever you specified for the ``frame_id`` parameter.

matheusns, 2017-01-30:
@jack-oquin  Could you give an example for us, please? Because I've not gotten the expected result. I want to change the axes and use the Velodyne in a vertical way.

jlblancoc, 2017-01-30:
Hi,

You will probably get a better and faster answer by asking in
http://answers.ros.org because your doubt is quite general and is not
directly related to this package.

Best,

jack-oquin, 2017-01-30:
I agree with @jlblancoc that this is a general ROS coordinate transforms question, not directly related to the Velodyne at all.

Please ask your question on http://answers.ros.org with tags for ``tf``, ``tf2`` and ``velodyne``.

closed 2017-01-30 by jack-oquin