ros-drivers/velodyne #116: Missing is observed in the acquired point cloud data

neo-ayuge, 2017-02-12:
### My environment
* ROS Distro: Indigo
* OS Version: Ubuntu 14.04
* Source build
* version: e263b80
* using device: HDL64E S2(freq. 10Hz)

### Steps to reproduce
* First, I get pcap data via VeloView.
* Second, I execute the following commands.
```bash
$ rosrun velodyne_driver velodyne_node _model:=64E_S2 _pcap:=<my pcap file> _rpm:=<setting frequency>
$ roslaunch velodyne_pointcloud 64e_s2_points.launch
```
64e_s2_points.launch is a simple imitation of "32e_points.launch".
https://github.com/Nishida-Lab/velodyne/blob/master/velodyne_pointcloud/launch/64e_s2_points.launch

* Then, I check point cloud data by rviz.

### Expected behaviour
I want point cloud data like the following.

* Velodyne S2 pointcloud visualize using VeloView
[image]

The figure is screenshot when VeloView is used to display pointcloud data.

### Actual behaviour

* Velodyne S2 pointcloud visualize using rviz
[image]

Please check a link of YouTube.
It's a movie of rviz.
https://youtu.be/2HoVRBuQjd4

And I checked frequency of /velodyne_packets.
I show the results of "rostopic hz" command.

```bash
$ rostopic hz /velodyne_packets
subscribed to [/velodyne_packets]
average rate: 9.855
    min: 0.100s max: 0.105s std dev: 0.00169s window: 10
average rate: 9.847
    min: 0.100s max: 0.105s std dev: 0.00158s window: 20
average rate: 9.845
    min: 0.099s max: 0.105s std dev: 0.00147s window: 30
average rate: 9.857
    min: 0.099s max: 0.105s std dev: 0.00135s window: 40
average rate: 9.858
    min: 0.098s max: 0.105s std dev: 0.00145s window: 49
average rate: 9.857
    min: 0.098s max: 0.105s std dev: 0.00143s window: 59
average rate: 9.869
    min: 0.098s max: 0.105s std dev: 0.00136s window: 69
average rate: 9.864
    min: 0.098s max: 0.105s std dev: 0.00138s window: 79
average rate: 9.859
    min: 0.098s max: 0.110s std dev: 0.00162s window: 89
average rate: 9.853
    min: 0.098s max: 0.110s std dev: 0.00159s window: 99
average rate: 9.839
    min: 0.098s max: 0.110s std dev: 0.00183s window: 108
```

Why does data loss occur?
Please tell me the solution.

Thank you.

jmlago91, 2017-02-12:
I think the problem is linked with this: https://github.com/ros-drivers/velodyne/pull/97/files

Since at this moment the driver is using a fixed data rate, and Velodyne doesn't produce at exact data rate (I could observe in different S3 units that rpm present small variations with the temperature), rviz are not showing exactly one frame (a little section corresponding a few packets due to a small deviation in the data rate).
You can experiment with the packets per scan variable to confirm this problem changing the "npackets" (packet rate/frequency) in the velodyne_driver launch file. Try values like 580-590.

One solution (that implies changing the approaching) is to read the azimuth value of all packets, from 0 to 35999, and making a frame with the points from that packets. That is how VeloView works.

neo-ayuge, 2017-02-15:
@jmlago91 I'm grateful for your response.
After adjusting "npackets", it worked fine.
I'll also try the approach using azimuth value next time.

Thank you.

chukcha2, 2017-02-24:
Does this problem exist for all models then (include vlp16)?

jmlago91, 2017-02-26:
Since the driver uses the same approach for all models, the same behaviour is expected.

chukcha2, 2017-03-01:
In this case, @neo-ayuge , why is this issue closed? Wouldn't it be a good enhancement?

jack-oquin, 2017-03-28:
There is an experimental fix for this problem, see #35.

That fix is currently in the ``cut_at_specified_angle_feature`` branch, which is not currently up-to-date with the ``master`` branch. There is still work to do before we can release it in a way that is backwards compatible with existing uses.

Feel free to try it, and report back here if it solves the problem you are seeing.

chukcha2, 2017-04-24:
Am I understanding this correctly?: no data is lost; the problem is only with visualizing in rviz. Is this correct?

JWhitleyAStuff, 2018-12-06:
The `cut_at_specified_angle` branch was merged into `master` some time ago. I'm not exactly sure how this feature would solve the stated problem, but @jack-oquin seems to think it will. My understanding is that the driver attempts to produce data at exactly 10Hz (100ms interval) while the sensor does not always produce exactly 1 scan worth of data in 100ms. This leads me to the conclusion that the only way we could directly emulate what VeloView is producing is by re-architecting the driver to produce messages ad-hoc as complete scans are received. This is unlikely to happen in the near future without a PR and lots of testing.

If there is more discussion to be had, please feel free to re-open this ticket.

closed 2018-12-06 by JWhitleyAStuff