ros-drivers/velodyne #118: Pcap replay error

merriaux, 2017-03-05:
Hi everyone,
I try to replay a pcap file from a VLP16 record (ubuntu 16.04 and ros kinetic):
roslaunch velodyne_pointcloud VLP16_points.launch pcap:="2014-11-10-11-32-17_Velodyne-VLP_10Hz_Monterey Highway.pcap"

And I obtain this errors above.
I have done something wrong ? someone have a idea ? 
thank for help
Pierre

[FATAL] [1488740219.560444888]: Error[FATAL] [1488740219.560444888]: Error opening Velodyne socket dump file.
[FATAL] [1488740219.673309663]: Failed to load nodelet '/velodyne_nodelet_manager_cloud` of type `velodyne_pointcloud/CloudNodelet` to manager `velodyne_nodelet_manager'
[FATAL] [1488740219.673360752]: Failed to load nodelet '/velodyne_nodelet_manager_driver` of type `velodyne_driver/DriverNodelet` to manager `velodyne_nodelet_manager'
[velodyne_nodelet_manager-1] process has died [pid 21984, exit code -11, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=velodyne_nodelet_manager __log:=/home/merriaux/.ros/log/80bba3f8-01d3-11e7-80c9-4851b7c236cf/velodyne_nodelet_manager-1.log].
log file: /home/merriaux/.ros/log/80bba3f8-01d3-11e7-80c9-4851b7c236cf/velodyne_nodelet_manager-1*.log
[velodyne_nodelet_manager_driver-2] process has died [pid 21985, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load velodyne_driver/DriverNodelet velodyne_nodelet_manager __name:=velodyne_nodelet_manager_driver __log:=/home/merriaux/.ros/log/80bba3f8-01d3-11e7-80c9-4851b7c236cf/velodyne_nodelet_manager_driver-2.log].
log file: /home/merriaux/.ros/log/80bba3f8-01d3-11e7-80c9-4851b7c236cf/velodyne_nodelet_manager_driver-2*.log
[velodyne_nodelet_manager_cloud-3] process has died [pid 21986, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load velodyne_pointcloud/CloudNodelet velodyne_nodelet_manager __name:=velodyne_nodelet_manager_cloud __log:=/home/merriaux/.ros/log/80bba3f8-01d3-11e7-80c9-4851b7c236cf/velodyne_nodelet_manager_cloud-3.log].
log file: /home/merriaux/.ros/log/80bba3f8-01d3-11e7-80c9-4851b7c236cf/velodyne_nodelet_manager_cloud-3*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

jack-oquin, 2017-03-28:
With a ROS launch script, it's necessary to provide the full path name to the PCAP file. The node does not usually run in the current directory of the ``roslaunch`` command.

Questions like this are better asked on questions.ros.org.

kmotillo, 2017-04-30:
I had the same problem and I solved it with your information, thanks **jack-oquin**

aditya369007, 2017-06-15:
@kmotillo Im facing the same issue as that you have posted. Can you guide me on how to solve it?
I am using the $(pwd) for the full path name that roslaunch requires, but still facing issues on replaying pcap files.
Does the pcap file need to be stored in a specific place for this run ?

Much thanks in advance

orz-000, 2018-09-25:
I have same problem and I provided full path of pcap file, but I couldn't run this program.  How can I solve it??

command:
roslaunch velodyne_pointcloud VLP16_points.launch pcap:=/home/koby/Download/vlp16_sample.pcap 

error massage:
... logging to /root/.ros/log/b891589e-c0c8-11e8-9936-0242ac110002/roslaunch-14e5fc6365ab-10521.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://14e5fc6365ab:39525/

SUMMARY

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.13
 * /velodyne_nodelet_manager_cloud/calibration: /opt/ros/kinetic/...
 * /velodyne_nodelet_manager_cloud/max_range: 130.0
 * /velodyne_nodelet_manager_cloud/min_range: 0.4
 * /velodyne_nodelet_manager_driver/device_ip: 
 * /velodyne_nodelet_manager_driver/frame_id: velodyne
 * /velodyne_nodelet_manager_driver/model: VLP16
 * /velodyne_nodelet_manager_driver/pcap: /home/koby/Downlo...
 * /velodyne_nodelet_manager_driver/port: 2368
 * /velodyne_nodelet_manager_driver/read_fast: False
 * /velodyne_nodelet_manager_driver/read_once: False
 * /velodyne_nodelet_manager_driver/repeat_delay: 0.0
 * /velodyne_nodelet_manager_driver/rpm: 600.0
 * /velodyne_nodelet_manager_laserscan/resolution: 0.007
 * /velodyne_nodelet_manager_laserscan/ring: -1

NODES
  /
    velodyne_nodelet_manager (nodelet/nodelet)
    velodyne_nodelet_manager_cloud (nodelet/nodelet)
    velodyne_nodelet_manager_driver (nodelet/nodelet)
    velodyne_nodelet_manager_laserscan (nodelet/nodelet)

ROS_MASTER_URI=http://localhost:11311

process[velodyne_nodelet_manager-1]: started with pid [10538]
process[velodyne_nodelet_manager_driver-2]: started with pid [10539]
[FATAL] [1537884587.764453333]: Failed to load nodelet '/velodyne_nodelet_manager_driver` of type `velodyne_driver/DriverNodelet` to manager `velodyne_nodelet_manager'
[velodyne_nodelet_manager-1] process has died [pid 10538, exit code -11, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=velodyne_nodelet_manager __log:=/root/.ros/log/b891589e-c0c8-11e8-9936-0242ac110002/velodyne_nodelet_manager-1.log].
log file: /root/.ros/log/b891589e-c0c8-11e8-9936-0242ac110002/velodyne_nodelet_manager-1*.log
process[velodyne_nodelet_manager_cloud-3]: started with pid [10563]
[velodyne_nodelet_manager_driver-2] process has died [pid 10539, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load velodyne_driver/DriverNodelet velodyne_nodelet_manager __name:=velodyne_nodelet_manager_driver __log:=/root/.ros/log/b891589e-c0c8-11e8-9936-0242ac110002/velodyne_nodelet_manager_driver-2.log].
log file: /root/.ros/log/b891589e-c0c8-11e8-9936-0242ac110002/velodyne_nodelet_manager_driver-2*.log
process[velodyne_nodelet_manager_laserscan-4]: started with pid [10578]

JWhitleyAStuff, 2018-09-25:
@orz-000 - It's difficult to tell what caused the error without the log. Did you build the driver from source or install it from the ROS repository? Have you modified the launch file? Can you please paste the log that it mentions in the output here?

orz-000, 2018-09-28:
@JWhitleyAStuff -Thank you for indicating and I fixed my docker file. I built and ran VLP16_points.launch, but I couldn't run all process and same error messages appeared.
What should I do? How can I solve it?

[FATAL] [1538150971.656599982]: Failed to load nodelet '/velodyne_nodelet_manager_driver` of type `velodyne_driver/DriverNodelet` to manager `velodyne_nodelet_manager'
[velodyne_nodelet_manager-1] process has died [pid 92, exit code -11, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=velodyne_nodelet_manager __log:=/root/.ros/log/cf64bb98-c338-11e8-9e14-0242ac110002/velodyne_nodelet_manager-1.log].
log file: /root/.ros/log/cf64bb98-c338-11e8-9e14-0242ac110002/velodyne_nodelet_manager-1*.log

[velodyne_nodelet_manager_driver-2] process has died [pid 93, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load velodyne_driver/DriverNodelet velodyne_nodelet_manager __name:=velodyne_nodelet_manager_driver __log:=/root/.ros/log/cf64bb98-c338-11e8-9e14-0242ac110002/velodyne_nodelet_manager_driver-2.log].
log file: /root/.ros/log/cf64bb98-c338-11e8-9e14-0242ac110002/velodyne_nodelet_manager_driver-2*.log

file: /root/.ros/log/fd4f7b44-c2b9-11e8-9f68-0242ac110002/velodyne_nodelet_manager_driver-2*.log

messages: [ INFO] [1538096563.363554239]: Loading nodelet /velodyne_nodelet_manager_driver of type velodyne_driver/DriverNodelet to manager velodyne_nodelet_manager with the following remappings:

JWhitleyAStuff, 2018-09-28:
@orz-000 - I don't have an immediate solution for you. Please let me know your environment outside of Docker (Kerenel version, distribution, etc) and I'll do my best to replicate it on my machine. I'll put this ticket in "Needs Testing" for now. Can't promise a quick response on this one.

closed 2018-09-28 by JWhitleyAStuff

sherlei, 2022-06-02:
> With a ROS launch script, it's necessary to provide the full path name to the PCAP file. The node does not usually run in the current directory of the `roslaunch` command.
> 
> Questions like this are better asked on questions.ros.org.

MoonSheepJx, 2022-11-28:
hello, I have the same problem when I use 
`roslaunch velodyne_pointcloud VLP16_points.launch`
Thank for any helps!

JWhitleyWork, 2022-12-04:
@MoonSheepJx Please do not bump closed issues. Create a new one and link to the closed one if necessary. Thanks.