ros-drivers/velodyne #119: VLP_points.launch.xml producing unsued args for the include of error

gowmo, 2017-03-21:
I am trying to use VLP16 as 3D sensor.  While trying to execute the VLP_points.lanch.xml  , it is generating the below error ..

unused args [device_ip, manager, port] for include of [/opt/ros/indigo/share/velodyne_driver/launch/nodelet_manager.launch]

I tried to change and modfiy the args in the xml, but to no avail.. Everything looks okay, but I am not able to figure out why those three args alone unused if  the rest of the args are okay.

jack-oquin, 2017-03-28:
What version of the driver are you running? Is this a binary install, or are you building it from source?

gowmo, 2017-03-28:
It's a source installation. I am trying to use the VLP16 lidar with the Kobuki turtlebot. I am able to view the 3d point cloud from velodyne when I do a seperate launch without invloving turtlebot. But, when I try to launch the 3dsensor from the turtlebot , I received the above error. 

From the error, I found that the arguments are not being used in the nodelet_manager.launch and I have commented those above mentioned unused args and continued with the 3dsensor.launch of the turtlebot where I receive the error "unknown Velodyne LIDAR model VLP16" .. 

I am confused why the velodyne lidar is not being recognized. Please help.

Thanks in advance.

jack-oquin, 2017-03-28:
Some of the velodyne launch files have been updated to support additional arguments. For some reason, the Kobuki launch file is not compatible with that.

Can you please post a link to the github source of the Kobuki launch for the velodyne?

gowmo, 2017-03-28:
I am not sure of the github source of Kobuki Velodyne launch. All I am trying to do is to add a new 3D sensor to the Kobuki turtlebot based on the below link   [http://wiki.ros.org/turtlebot/Tutorials/indigo/Adding%20New%203D%20Sensor]
as  Kobuki is by default not configured to be run with VLP .
  
As per the steps mentioned in that link, we need to add the necessary files for the new 3D sensor to use with the kobuki. 
I used the launch file from your driver and urdf file from [https://bitbucket.org/DataspeedInc/velodyne_simulator/src/b2b4496e8150c4e4082a0b8227922a721f4dc91d/velodyne_description/urdf/?at=master](url)   and I am trying to create the remaining wherever based based on the kinect files(editing to VLP).... where I am not successful.

Thanks in advance.

jack-oquin, 2017-03-29:
I can't figure out where you are invoking ``VLP16_points.launch``. Is it a command line ``roslaunch``, or some other launch script.

If it's another launch script, I need to see the relevant parts somehow.

gowmo, 2017-03-29:
We are invoking the VLP16_points.launch from 3dsensor.launch through roslaunch.The launch file is availabe in the below link [](https://github.com/turtlebot/turtlebot/blob/indigo/turtlebot_bringup/launch/3dsensor.launch) 

That 3dsensor.launch above invokes  a 3d sensor launch. we have places the VLP16_points.laucnh as VLP16.launch and are trying to provide the argument to 3d_sensor as VLP16. Please find below snippet  of the 3dsensor launch file

<include file="$(find turtlebot_bringup)/launch/includes/3dsensor/$(arg 3d_sensor).launch.xml">

Thank You

jack-oquin, 2017-03-29:
Still no help. I don't see any ``VLP16.launch.xml`` in this directory:

 https://github.com/turtlebot/turtlebot/tree/kinetic/turtlebot_bringup/launch/includes/3dsensor

I need to see what's in it.

gowmo, 2017-04-02:
Yes, there is no VLP16.launch.xml in that directory. I have changed your VLP16_points.xml as VLP16.launch.xml so that the 3dsensor.launch trigger the velodyne. 

We have to make necessary changes to the turtlebot to add the new 3D sensor as per the http://wiki.ros.org/turtlebot/Tutorials/indigo/Adding%20New%203D%20Sensor . Based on the procedure as per the link, I tried to replicate the velodyne files looking at the kinect and asus xtion pro files, which went as an unsuccesful attempt. 

Thanks in advance.

jack-oquin, 2017-04-02:
I don't see how to help without knowing what you are doing.

Can you paste in your modified ``VLP16.launch.xml``?

JWhitleyAStuff, 2018-06-11:
This ticket hasn't been updated in over a year with no response from the submitter. Closing.

closed 2018-06-11 by JWhitleyAStuff