ros-drivers/velodyne #120: Fix catkin_lint errors

LukasKalbertodt, 2017-05-17:
Running [`catkin_lint`](https://github.com/fkie/catkin_lint) results in these errors:

```
velodyne_driver: error: missing run_depend on 'roslaunch'
velodyne_driver: error: missing run_depend on 'rostest'
velodyne_driver: error: unconfigured build_depend on 'roslaunch'
velodyne_driver: error: unconfigured build_depend on 'rostest'
velodyne_driver: error: unconfigured build_depend on 'pluginlib'
velodyne_driver: CMakeLists.txt(49): error: duplicate find_package(roslaunch)
velodyne_driver: CMakeLists.txt(50): error: duplicate find_package(rostest)
velodyne_pointcloud: error: unconfigured build_depend on 'roslaunch'
velodyne_pointcloud: error: unconfigured build_depend on 'rostest'
velodyne_pointcloud: error: unconfigured build_depend on 'tf2_ros'
velodyne_pointcloud: error: unconfigured build_depend on 'pluginlib'
velodyne_pointcloud: tests/CMakeLists.txt(11): error: missing test_depend on 'rosunit'
```

... and these warnings:
```
velodyne_pointcloud: warning: executable file 'src/lib/rawdata.cc' is not installed
velodyne_pointcloud: CMakeLists.txt(32): warning: use of link_directories() is strongly discouraged
```

Fixing those would be awesome! :rainbow:

LukasKalbertodt, 2017-06-23:
@jack-oquin @totosolat Any update on this? Would you even accept a PR that fixes those errors? 

We currently can't use `catkin_lint` in our project the way we want, because this package causes errors in our workspace. Resolving this issue would be very much appreciated!

jack-oquin, 2018-01-06:
Fixing these problems should probably be done as part of a migration to ``package.xml`` format 2.

JWhitleyAStuff, 2018-05-10:
This was recently done with PR #160.

closed 2018-05-10 by JWhitleyAStuff