ros-drivers/velodyne #121: Make PointXYZIR compatible with PCL template classes

cglwn, 2017-05-26:
I tried converting some existing code I had to use the `velodyne_pointcloud::PointXYZIR` type provided in this repository, but it wouldn't build due to uses of `pcl::KdTreeFLANN<velodyne_pointcloud::PointXYZIR`, `pcl::PCLBase<velodyne_pointcloud::PointXYZIR>` and `pcl::Passthrough<velodyne_pointcloud::PointXYZIR`. From the PCL documentation on how to [add a new type](http://pointclouds.org/documentation/tutorials/adding_custom_ptype.php#how-to-add-a-new-pointt-type), it seems like we need to explicitly `include` the header and implementation of the class templates to get it building.

wdxairforce, 2018-10-06:
@cglwn Hi, i am encountering the same problem you meet, may i ask if you have find any solution to deal with it?

cglwn, 2018-10-06:
@wdxairforce The change I did in #122 worked for me, but required rebuilding. Jack suggested something else that might work but I didn't get the change to try. 

> Would it work to put PCL_NO_PRECOMPILE and those other includes before the include of velodyne_pointcloud/point_types.h?

I'm pretty far removed from working in this area so I'm afraid I can't help further.

closed 2022-12-04 by JWhitleyWork