ros-drivers/velodyne #124: VLP-16 packet size is not matched

hcv1027, 2017-08-03:
My working environment (Ubuntu 16.04 + ROS Kinetic) can't catch the data from VLP-16 correctly.
It seems that VLP-16 dose not send data packet with correct size.
By the document and code, VLP-16 should send each data packet with 1206 bytes, but it only sends 512 bytes. (Both code log and protocol packet analysis tool _wireshark_ show it is 512 bytes)
The data packet seems not only has wrong size, and its contents are all zero.
Below is the screen shot.
Can someone help me to debug this problem?
[image]
[image]
[image]

jlblancoc, 2017-08-03:
Hi @hcv1027 

Those 512 B packets are *not* scan data (UDP port 2368), but "telemetry" packets (UDP port 8308). They are all zero because, I guess, the LiDAR is not connected to a GPS source, right? Those packets are populated with an NMEA frame, when available from a GPS. 

Your problem is, apparently, that the LiDAR is not emitting range data. Try clicking on "laser: off" -> save; (wait 1 minute), "laser: on" -> save, and see if something changes.

hcv1027, 2017-08-04:
Hi @jlblancoc 
If the save you said is "Save configuration" button, then I just tried and it did not work.
It still only sends "telemetry" packets.
Any other suggestion?

h3ct0r, 2017-08-31:
Have you tried to set the destination IP address to your ROS master IP?

JWhitleyAStuff, 2017-11-06:
@jlblancoc - I would recommend you contact Velodyne on this issue. This is something related to either the configuration (though that looks correct) or the proper functioning of your device. I am closing this issue as off-topic (since it isn't related to the ROS driver)

closed 2017-11-06 by JWhitleyAStuff

jlblancoc, 2017-11-06:
Sure @JWhitleyAStuff close it!
But it was @hcv1027 not me who was having troubles ;-)

Cheers

JWhitleyAStuff, 2017-11-06:
Thanks for the correction, @jlblancoc.

wawaafan, 2018-01-01:
Do you have resolve this problem?  I also meet the same question, could you tell me how to resolve this?

jack-oquin, 2018-01-02:
The destination IP of 255.255.255.255 looks wrong. Have you tried setting the actual IP address?

saikishor, 2018-10-18:
@jack-oquin is having destination IP as 255.255.255.255 wrong?. I have a velodyne sensor. I connected it to my network, it works very well, but I am not able to access its webpage to configure stuff. Any idea how to recover to the original settings?
[image]

duchengyao, 2019-03-01:
make sure  TTL in PPS and RS232 in GPRMC

yasmeenel3sh, 2019-05-25:
@duchengyao I am having the same problem now ,but I am sure the correct signals are connected in the correct places , though what happens is in the web interface it shows that the pps is locked and it shows the pps position, however on wire shark the nmea gprmc packet isnot being send by veloyne ,what could be the reason for that?

JWhitleyAStuff, 2019-05-28:
@yasmeenel3sh - This is a *very old* ticket. Could you please create a new one if you are still having problems?

OxMarco, 2023-09-24:
I am getting the same issue, if you solved the problem it would be good to have some info on how to do it