ros-drivers/velodyne #125: How does

LiShuaixin, 2017-09-26:


LiShuaixin, 2017-09-26:
Hello there, I have a question about output data. As I asked, how does the coordination system of output data define. Is that the origin at the geometrical centre of sensor, the x-axis pointing to the right, the y-axis pointing upward and z-axis pointing forward? It's a fundamental and essential information for multi-sensors combined system, like lidar/IMU.

Kaiser1401, 2017-09-26:
From my experience the ROS Data is as Follows:
Z-Axis: pointing up (Axis the sensor turns around)
X-Axis: axis that internally has angle 0
Y-Axis: remaining for right handed coordinate system

So:
for Velodyne16: X_ROS is opposite of the cable
for Velodyne32: Y_ROS is opposite of the cable (I think, It is different to a 16 though)
for Velodyne64: haven't put my hands on

Unfortunately the systems are not consistent, neither is the internal 0-angle of the velodynes ...

JWhitleyAStuff, 2017-11-06:
@LiShuaixin / @Kaiser1401: According to the documentation:

The ROS_X axis (forward on a vehicle) is as follows:
VLP-16: The 0-degree point of the sensor lies along ROS_X when the cable is facing -ROS_Y (toward the passenger side of a vehicle).
HDL-32E: The 0-degree point of the sensor lies along ROS_X when the cable is facing -ROS_Y (toward the passenger side of a vehicle).
HDL-64E: The 0-degree point of the sensor lies along ROS_X when the cable is facing -ROS_X (toward the back of a vehicle).

Closing this ticket and leaving for historical reasons (off-topic but useful).

closed 2017-11-06 by JWhitleyAStuff