ros-drivers/velodyne #128: "double vision" effect in PointCloud2 msg

facontidavide, 2017-10-18:
Hi,

playing around with the data published by this driver a noticed a potential issue. My velodyne is mounted on a robot that is moving relatively rapidly.

I noticed that a single message of PointCloud2 contains points from more than a single 360 degrees rotation. This result in detecting twice the same object in the same message, as you can notice in the attached image; since the robot is moving, the detection is not overlapping.

[image]

The solution would be, I guess, to "split" the message when a 360º rotation is detected either here
https://github.com/ros-drivers/velodyne/blob/34d1dad3ea2fd6a1f30bf2a80441c792e0e7b199/velodyne_pointcloud/src/conversions/convert.cc#L70
or here
https://github.com/ros-drivers/velodyne/blob/34d1dad3ea2fd6a1f30bf2a80441c792e0e7b199/velodyne_pointcloud/src/conversions/transform.cc#L82

Does this make sense in your opinion?

Cheers 

Davide

lucasveronese, 2017-10-25:
The problem that you are having is a big issue. I do not know which robot you are using... but, you do have to take into account the motion model of your robot. At the manual of the velodyne has time spent for doing a vertical scan. For each of those, the movement should be consider. I.e., using the odomotries measurement, the robot would be posed and the vertical velodyne points of one shot should  be translated  and rotated to right place. I had the same problem in 2015 mapping large-scale  areas. This will fix this problem fore sure.

facontidavide, 2017-10-26:
To do what you said I guess that I would need at least the timestamp at the beginning of the robot scan and at the end, to transform the XYZ points according to my odometry...

What I found confusing or suboptimal is that apparently the amplitude of the scan differs in the starting and final angle of the scan itself.

If ww have consistently a single message for a single LIDAR rotation of 360 degrees, it would be easier to build a motion model.

lucasveronese, 2017-10-26:
I am not saying that. The robot could be in frame (x=0,y=0,theta=0). The velodyne points must be related to coordinate center of the robot. Then, for each vertical scan you move the robot using the odometry and the time (which is into the manual) to record it. After you put the points in world coordinetes (0,0,0), you apply the transformation to the position of your SLAM.

volkandre, 2017-12-29:
If you are looking for consistent 360 degree scans you might want to look into the "cut_at_specific" angle feature, which currently has its own branch: https://github.com/ros-drivers/velodyne/tree/cut_at_specified_angle_feature

Please also see the latest pull-request for this feature:
https://github.com/ros-drivers/velodyne/pull/138

A better description of the underlying problem with additional screenshots can be found here:
https://github.com/ros-drivers/velodyne/issues/33

JWhitleyAStuff, 2018-05-10:
PR #138 was merged. Should resolve this issue.

closed 2018-05-10 by JWhitleyAStuff