ros-drivers/velodyne #131: Compilation error

mpkuse, 2017-11-15:
I am trying to catkin_make this code. I have ros-kinetic
Most of the things seem to compile accept for linking with velodyne_rawdata.so.

```
[ 94%] Linking CXX shared library /home/mpkuse/catkin_ws/devel/lib/libvelodyne_rawdata.so
/usr/bin/ld: /usr/local/lib/libyaml-cpp.a(emitter.cpp.o): relocation R_X86_64_32 against `.rodata.str1.8' can not be used when making a shared object; recompile with -fPIC
/usr/local/lib/libyaml-cpp.a: error adding symbols: Bad value
collect2: error: ld returned 1 exit status
velodyne/velodyne_pointcloud/src/lib/CMakeFiles/velodyne_rawdata.dir/build.make:341: recipe for target '/home/mpkuse/catkin_ws/devel/lib/libvelodyne_rawdata.so' failed
make[2]: *** [/home/mpkuse/catkin_ws/devel/lib/libvelodyne_rawdata.so] Error 1
CMakeFiles/Makefile2:16023: recipe for target 'velodyne/velodyne_pointcloud/src/lib/CMakeFiles/velodyne_rawdata.dir/all' failed
make[1]: *** [velodyne/velodyne_pointcloud/src/lib/CMakeFiles/velodyne_rawdata.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j12 -l12" failed
```

JWhitleyAStuff, 2018-02-25:
@mpkuse - What OS/hardware platform are you trying to build this in?

JWhitleyAStuff, 2018-05-10:
Closing due to lack of response.

SubMishMar, 2020-02-16:
I am having the same issue on my Ubuntu 16.04 PC running ROS Kinetic. Any idea why?

JWhitleyWork, 2020-02-16:
@SubMishMar Is the error trace that you're getting *exactly* the same as the above one?

SubMishMar, 2020-02-16:
Yes exactly!

```
Errors     << velodyne_pointcloud:make /home/subodh/catkin_ws/logs/velodyne_pointcloud/build.make.001.log                                                                                                                             
/usr/bin/ld: /usr/local/lib/libyaml-cpp.a(convert.cpp.o): relocation R_X86_64_32 against `.text' can not be used when making a shared object; recompile with -fPIC
/usr/local/lib/libyaml-cpp.a: error adding symbols: Bad value
collect2: error: ld returned 1 exit status
make[2]: *** [/home/subodh/catkin_ws/devel/.private/velodyne_pointcloud/lib/libvelodyne_rawdata.so] Error 1
make[1]: *** [src/lib/CMakeFiles/velodyne_rawdata.dir/all] Error 2
make: *** [all] Error 2
```

SubMishMar, 2020-02-16:
Although, I just fixed it using: https://github.com/jbeder/yaml-cpp/issues/682#issuecomment-495180374

JWhitleyWork, 2020-02-16:
I was going to suggest that it was likely a yaml-cpp issue. Thanks for the follow-up!

closed 2020-02-16 by JWhitleyWork