ros-drivers/velodyne #133: Turning the Velodyne VLP16 Laser Scan Off

matheusns, 2017-11-23:
Hello guys,  is there a way to turn off the Velodyne VLP16 laser scan off without removing from the power source? I'm developing an application that is crashing when some sensors are collecting data. I'd like to turn off the scanning during other sensor collecting.

jlblancoc, 2017-11-27:
In case someone wants to integrate these features in this repo, in these range of lines:

https://github.com/MRPT/mrpt/blob/master/libs/hwdrivers/src/CVelodyneScanner.cpp#L1030-L1125

we implemented Return type (dual/strongest), Laser On/Off and Change RPM, in our MRPT implementation of Velodyne sensor.

JWhitleyAStuff, 2018-01-26:
Would you mind forking this repo, making the required changes, and submitting a pull request?

closed 2018-02-15 by matheusns