ros-drivers/velodyne #134: VLP16 layer displacement

janblumenkamp, 2017-11-29:
Hello everyone,
I installed the velodyne ROS drivers without any problems and I do get a pointcloud from the lidar. But this pointcloud seems to be displaced, as in the layers all have a slightly different rotation.
This picture shows a wall and a corner that is supposed to be straight:
[image]
[image]
[image]
This is not only on this wall, but around the whole room.
I start the driver with `roslaunch velodyne_driver nodelet_manager.launch model:=VLP16` and the PCL converter with `rosrun nodelet nodelet standalone velyne_pointcloud/CloudNodelet`
What am I doing wrong?
Best,
Jan

trainman419, 2017-11-30:
Are you using the high-res version of the VLP-16 or a standard VLP-16? (The high-res version has different beam angles that might explain the displacement)

janblumenkamp, 2017-12-02:
I am pretty sure it is a standard one, but I can't really tell. I guess the high-res one explicitly states that it is the high resolution one on the housing?)

janblumenkamp, 2017-12-05:
It seems like I actually have a high-res lidar and I do have a calibration file as well, now. I will try to convert it and try again then. But shouldn't be there a custom high res calibration file? Or is the high-res VLP16 simply not supported out of the box?

janblumenkamp, 2017-12-06:
I have converted the file with
```
rosrun velodyne_pointcloud gen_calibration.py VeloView-VLP-16-HiRes.xml 
```
and got the output

```
converting "VeloView-VLP-16-HiRes.xml" to "VeloView-VLP-16-HiRes.yaml"
16 lasers
```
Then I started the node with
```
roslaunch velodyne_driver nodelet_manager.launch model:=VLP16 calibration:=VeloView-VLP-16-HiRes.yaml
```
but the result is the same... I am not sure if the second command actually accepts the calibration file as there is no message related to it. I can also type any name, even one that does not exist, and the node starts anyway as always...
So, the issue persists. Looking forward to some guidance!

JWhitleyAStuff, 2018-01-26:
The issue may be related to the location of the calibration file. Have you placed it in the same folder as the others (velodyne_pointcloud/params)?

closed 2018-04-27 by JWhitleyAStuff