ros-drivers/velodyne #140: Inconsistent message timings

ryanmeasel, 2018-01-09:
Hardware: VLP 16
Driver version: Latest
OS Version: Ubuntu 16.04
ROS Distro: Kinetic
ROS version: 1.12.12

Using the latest driver and standard `VLP16_points.launch` (with `rpm=1200` and `npackets=1`), the messages are not sent on regular intervals. See an [example bag here](https://drive.google.com/file/d/1q7D4WfMiTG1YMzKukGt8ki8gZ7AiY5br/view?usp=sharing). 

Upon investigating the [example bag](https://storage.googleapis.com/cartographer-public-data/bags/backpack_3d/b3-2016-04-05-14-14-00.bag) for the Google Cartographer Project, the messages intervals are constant. Furthermore, they publish at a rate of ~1500 Hz whereas my install does ~750 Hz.

Is this behavior incorrect? If so, what is causing it?

ryanmeasel, 2018-01-09:
Figured out the messaging rate. From manual:
“In dual return mode, the data rate of the sensor doubles.”

Still curious about the messaging interval.

jack-oquin, 2018-01-09:
I have not used ``npackets=1``, but I'd expect the normal Linux scheduler algorithm to cause activities at that Hertz rate to be clustered. 

If you really need such small message rates, you might try running the driver with [RT_PREEMPT scheduling](https://wiki.linuxfoundation.org/realtime/documentation/howto/applications/preemptrt_setup). Note that UbuntuStudio provides a realtime kernel with those patches. 

A simpler solution would be specifying a larger ``npackets`` value.

JWhitleyAStuff, 2018-12-06:
Closing due to lack of activity and provided explanation by @jack-oquin.

closed 2018-12-06 by JWhitleyAStuff