ros-drivers/velodyne #142: Error in packet rate for the VLP-32C

mpitropov, 2018-01-29:
With the motor set to 32 RPM and a packet rate of 1808, I get around 8.3 hz on the /velodyne_packets topic.

With a packet rate of 1508:
rostopic hz /velodyne_packets 
subscribed to [/velodyne_packets]
average rate: 9.987
	min: 0.100s max: 0.100s std dev: 0.00011s window: 10
average rate: 9.984
	min: 0.100s max: 0.100s std dev: 0.00009s window: 19
average rate: 9.982
	min: 0.100s max: 0.100s std dev: 0.00008s window: 29
average rate: 9.982
	min: 0.100s max: 0.100s std dev: 0.00007s window: 39

Has anyone else tested the packet rate set for this model? I think that the packet rate should be at around 1508.

jarvis-huang, 2018-01-29:
I experienced the problem as @mpitropov . Any fix would be highly appreciated.

asimay, 2018-02-09:
@mpitropov , hi , 32C has 60,0000 points per second, it is not same with 32E. so you should change the packet rate, for 32C the packet rate is not 1808.

asimay, 2018-02-09:
@jarvis-huang , hi, are you udacity's employee? I'm udacity's SDC students and mentor. :)

JWhitleyAStuff, 2018-04-26:
Fixed by #154.

closed 2018-04-26 by JWhitleyAStuff