ros-drivers/velodyne #143: layer 8 and above in VLP-32C are distorted

jarvis-huang, 2018-01-29:
Hi, I am testing a VLP-32C. I believe I'm using the right calibration file `velodyne_pointcloud/params/VeloView-VLP-32C.yaml`. 

However, when visualizing the pointcloud, I'm in front of a flat ground, I only see clear ring pattern from layer 1~7 and layer 8 and above just seem like random noise. I believe there is some bug in the driver mapping raw packets into pointcloud. Can anybody help with this?

[image]

jarvis-huang, 2018-02-08:
Not reproducible any more. Closing.

closed 2018-02-08 by jarvis-huang

bloom256, 2021-09-07:
I'm facing a very similar problem. I've recorded several hundred records. About 3% of them have corrupted LiDAR data. LiDAR settings are always the same. LiDAR is a VLP32-C. A sample bag is here https://drive.google.com/file/d/1AvHe-CCgg3Gasz4E8XFpLWA787LAHMnS/view?usp=sharing.

[image]

jarvis-huang, 2021-09-07:
[rawdata.zip](https://github.com/ros-drivers/velodyne/files/7123729/rawdata.zip)
@bloom256 Please unzip and copy the two files over the old files and rebuild. Let me know if they fix your issue. They did for me.

bloom256, 2021-09-08:
> [rawdata.zip](https://github.com/ros-drivers/velodyne/files/7123729/rawdata.zip)
> @bloom256 Please unzip and copy the two files over the old files and rebuild. Let me know if they fix your issue. They did for me.

Thanks. I replaced files but the result is the same. It looks like the distance stored in velodyne packets is wrong for some points.

jarvis-huang, 2021-09-09:
@bloom256 can you upload your entire velodyne package, your raw data (bag containing velodyne packets) as well as how you run your code (e.g. launch file)?

bloom256, 2021-09-10:
@jarvis-huang 
The bag is here https://drive.google.com/file/d/1AvHe-CCgg3Gasz4E8XFpLWA787LAHMnS/view?usp=sharing
Launch file is like this: 
```
<launch>
    <arg name="project" default="velodyne_pointcloud_test"/>
    <node pkg="rviz" type="rviz" name="$(arg project)_rviz" required="true" args="-d $(find velodyne_pointcloud)/launch/include/test.rviz" />
    <include file="$(find velodyne_pointcloud)/launch/VLP-32C_points.launch"/>
    <node pkg="rosbag" type="play" name="player" output="screen" 
        args="--clock -l /Corrupted_bags/bag_2021-08-12-16-43-42_0_vlp32c.bag" />
</launch>
```

shurikrulik, 2022-09-01:
I faced with this issue too. I have VLP-32C and VLP-16, and there is issue only with 32C. Sometimes it gives me "broken" pointcloud. Moreover it can breaks on its own at any random moment and than get repaired in several seconds/minutes. Here i have bag with "lucky" moment when pointcloud brokes during bag recording, it means that there is no mistake in calibration file, but some  trouble in velodyne driver or maybe some undiscovered issue on hardware side. There are /velodyne_points and /velodyne_packets topics in this bag.  [https://drive.google.com/file/d/1ylVH5iultaEEBXERJ0W0mi5KLPVjiO_w/view?usp=sharing](url)

@JWhitleyWork Can i ask you for advice ?  Have you ever faced with this issue, or is there some direction where i have to find the reason of this trouble ?

https://user-images.githubusercontent.com/7719550/187938751-15b0ab51-ca2a-4f76-94bc-aa528befdc65.mp4

[image]
[image]

jarvis-huang, 2022-09-02:
Here is my copy of `velodyne_pointcloud` package: [velodyne_pointcloud.zip](https://github.com/ros-drivers/velodyne/files/9478345/velodyne_pointcloud.zip). It includes the launch files as well. Try replace and build and see if it fixes your problem. If not, then I'm out of ideas here.