ros-drivers/velodyne #148: ROS Driver Build Error

choisangwook, 2018-02-13:
Hello.
I'm trying to build a velodyne driver for https://github.com/ros-drivers/velodyne.git and the following error occurs :
I would like to ask if there is a way to resolve this part.
Thank you

[55 %] Building CXX object velodyne_driversr srcslib_CMakeFelesvelodyne_input.direction.cc.
[57 %] Built-target multI- iSca nRegistration
[59 %] Build target velodyne_laserscan
In file inserted from /home/sangwook/catkin_ws/src/velodyne/velodyne_driver/src/lib/input.cc:34:0:
/ homesangwookcatkin_wssr sr cbelodyne_driverincludeevelotch_drivervelodyne_drivership:37:driversangwooke / input h :
Compilation-terminated
Velodynevelodyne_driversparrylib_CMakeFilesverning_direction.director.composite:62:recipe for target
Make[2]****[velodyne_driversr srikeFliesvelodyne_inpuction.direction.ccc.] Error 1
CMakeFilesMakefile2:5244:recipe for target ' velodynefarodyne_driversriverallib_CMakeFiles_velody'.
Make[1] : *** [velodynevelodyne_driversr srakeFelesvelodyne_inpuction.direction2]
Make[1] : *** Waiting for unfinisheds ....
[62 %] Build target velodyne_puck_decoder
[64 %] Build target velodyne_puck_driver
[64 %] Build target velodyne_msg_generate_messages
Makefile:138 : recise for target ' all ' failed
Make : *** [all] Error 2
Invoking " make - j- l " failed

trainman419, 2018-02-13:
Your compile output seems garbled or corrupt somehow; it's very difficult to read. Are you sure you copied the output correctly? (if your computer uses a language that isn't english, it's probably better to use the native output instead of trying to translate it to english).

I don't see anything in your output that looks like a useful compile error. Perhaps there's an error message earlier in the output that explains the failure?

choisangwook, 2018-02-13:
Dear trainman419

The environment is ubuntu 16.04.3 and kenetic ROS and the total compile result is shown below.

sangwook@sangwook-VirtualBox:~/catkin_ws$ pwd
/home/sangwook/catkin_ws
sangwook@sangwook-VirtualBox:~/catkin_ws$ cd src
sangwook@sangwook-VirtualBox:~/catkin_ws/src$ ls -al
total 28
drwxrwxr-x  7 sangwook sangwook 4096  2월 13 09:09 .
drwxrwxr-x  5 sangwook sangwook 4096  2월  8 16:46 ..
lrwxrwxrwx  1 sangwook sangwook   50  2월  8 16:46 CMakeLists.txt -> /opt/ros/kinetic/share/catkin/cmake/toplevel.cmake
drwxrwxr-x  8 sangwook sangwook 4096  2월  8 16:57 loam_velodyne
drwxrwxr-x 34 sangwook sangwook 4096  2월  9 15:19 pcl
drwxrwxr-x  8 sangwook sangwook 4096  2월 13 09:08 velodyne
drwxrwxr-x  7 sangwook sangwook 4096  2월  9 16:47 velodyne_pointcloud_to_laserscan
drwxrwxr-x  7 sangwook sangwook 4096  2월  9 16:33 velodyne_puck
sangwook@sangwook-VirtualBox:~/catkin_ws/src$ cd ..
sangwook@sangwook-VirtualBox:~/catkin_ws$ ls -al
total 24
drwxrwxr-x  5 sangwook sangwook 4096  2월  8 16:46 .
drwxr-xr-x 20 sangwook sangwook 4096  2월 13 08:47 ..
drwxrwxr-x 12 sangwook sangwook 4096  2월 13 09:37 build
-rw-rw-r--  1 sangwook sangwook   98  2월  8 16:46 .catkin_workspace
drwxrwxr-x  5 sangwook sangwook 4096  2월 13 09:37 devel
drwxrwxr-x  7 sangwook sangwook 4096  2월 13 09:09 src
sangwook@sangwook-VirtualBox:~/catkin_ws$ catkin_make
Base path: /home/sangwook/catkin_ws
Source space: /home/sangwook/catkin_ws/src
Build space: /home/sangwook/catkin_ws/build
Devel space: /home/sangwook/catkin_ws/devel
Install space: /home/sangwook/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/sangwook/catkin_ws/build"
####
####
#### Running command: "make -j4 -l4" in "/home/sangwook/catkin_ws/build"
####
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target std_msgs_generate_messages_eus
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] Built target _velodyne_msgs_generate_messages_check_deps_VelodyneScan
[  0%] Built target std_msgs_generate_messages_nodejs
[  0%] Built target _velodyne_msgs_generate_messages_check_deps_VelodynePacket
[  0%] Built target _velodyne_puck_msgs_generate_messages_check_deps_VelodynePuckPoint
[  0%] Built target _velodyne_puck_msgs_generate_messages_check_deps_VelodynePuckPacket
[  0%] Built target _velodyne_puck_msgs_generate_messages_check_deps_VelodynePuckSweep
[  0%] Built target _velodyne_puck_msgs_generate_messages_check_deps_VelodynePuckScan
[  0%] Built target dynamic_reconfigure_generate_messages_nodejs
[  1%] Built target velodyne_driver_gencfg
[  8%] Built target loam
[  8%] Built target sensor_msgs_generate_messages_nodejs
[  8%] Built target geometry_msgs_generate_messages_cpp
[  8%] Built target sensor_msgs_generate_messages_lisp
[  8%] Built target tf2_msgs_generate_messages_lisp
[  8%] Built target sensor_msgs_generate_messages_cpp
[  8%] Built target actionlib_msgs_generate_messages_lisp
[  8%] Built target rosgraph_msgs_generate_messages_nodejs
[  8%] Built target geometry_msgs_generate_messages_lisp
[  8%] Built target geometry_msgs_generate_messages_nodejs
[  8%] Built target roscpp_generate_messages_cpp
[  8%] Built target geometry_msgs_generate_messages_eus
[  8%] Built target actionlib_generate_messages_py
[  8%] Built target actionlib_msgs_generate_messages_py
[  8%] Built target sensor_msgs_generate_messages_py
[  8%] Built target geometry_msgs_generate_messages_py
[  8%] Built target roscpp_generate_messages_eus
[  8%] Built target actionlib_generate_messages_cpp
[  8%] Built target tf2_msgs_generate_messages_eus
[... 76 lines cut ...]
In file included from /home/sangwook/catkin_ws/src/velodyne/velodyne_driver/src/lib/input.cc:34:0:
/home/sangwook/catkin_ws/src/velodyne/velodyne_driver/include/velodyne_driver/input.h:37:18: fatal error: pcap.h: No such file or directory
compilation terminated.*_*
[ 56%] Built target multiScanRegistration
_[ 57%] Building CXX object velodyne/velodyne_driver/src/lib/CMakeFiles/velodyne_input.dir/input.cc.o
**[ 59%] Built target velodyne_laserscan
In file included from /home/sangwook/catkin_ws/src/velodyne/velodyne_driver/src/lib/input.cc:34:0:
/home/sangwook/catkin_ws/src/velodyne/velodyne_driver/include/velodyne_driver/input.h:37:18: fatal error: pcap.h: No such file or directory
compilation terminated.*_*
velodyne/velodyne_driver/src/lib/CMakeFiles/velodyne_input.dir/build.make:62: recipe for target 'velodyne/velodyne_driver/src/lib/CMakeFiles/velodyne_input.dir/input.cc.o' failed
make[2]: *** [velodyne/velodyne_driver/src/lib/CMakeFiles/velodyne_input.dir/input.cc.o] Error 1
CMakeFiles/Makefile2:5244: recipe for target 'velodyne/velodyne_driver/src/lib/CMakeFiles/velodyne_input.dir/all' failed
make[1]: *** [velodyne/velodyne_driver/src/lib/CMakeFiles/velodyne_input.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 62%] Built target velodyne_puck_driver
[ 64%] Built target velodyne_puck_decoder
[ 64%] Built target velodyne_msgs_generate_messages
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
sangwook@sangwook-VirtualBox:~/catkin_ws$ 

It display an error at 57% ~59% (fatal error: pcap.h: No such file or directory)
I need your help please
thank you

choisangwook, 2018-02-13:
Dear trainman419
The environment is ubuntu 16.04.3 and kenetic ROS and the total compile result is shown below.
sangwook@sangwook-VirtualBox:~/catkin_ws$ pwd/home/sangwook/catkin_wssangwook@sangwook-VirtualBox:~/catkin_ws$ cd srcsangwook@sangwook-VirtualBox:~/catkin_ws/src$ ls -altotal 28drwxrwxr-x  7 sangwook sangwook 4096  2월 13 09:09 .drwxrwxr-x  5 sangwook sangwook 4096  2월  8 16:46 ..lrwxrwxrwx  1 sangwook sangwook   50  2월  8 16:46 CMakeLists.txt -&gt; /opt/ros/kinetic/share/catkin/cmake/toplevel.cmakedrwxrwxr-x  8 sangwook sangwook 4096  2월  8 16:57 loam_velodynedrwxrwxr-x 34 sangwook sangwook 4096  2월  9 15:19 pcldrwxrwxr-x  8 sangwook sangwook 4096  2월 13 09:08 velodynedrwxrwxr-x  7 sangwook sangwook 4096  2월  9 16:47 velodyne_pointcloud_to_laserscandrwxrwxr-x  7 sangwook sangwook 4096  2월  9 16:33 velodyne_pucksangwook@sangwook-VirtualBox:~/catkin_ws/src$ cd ..sangwook@sangwook-VirtualBox:~/catkin_ws$ ls -altotal 24drwxrwxr-x  5 sangwook sangwook 4096  2월  8 16:46 .drwxr-xr-x 20 sangwook sangwook 4096  2월 13 08:47 ..drwxrwxr-x 12 sangwook sangwook 4096  2월 13 09:37 build-rw-rw-r--  1 sangwook sangwook   98  2월  8 16:46 .catkin_workspacedrwxrwxr-x  5 sangwook sangwook 4096  2월 13 09:37 develdrwxrwxr-x  7 sangwook sangwook 4096  2월 13 09:09 srcsangwook@sangwook-VirtualBox:~/catkin_ws$ catkin_makeBase path: /home/sangwook/catkin_wsSource space: /home/sangwook/catkin_ws/srcBuild space: /home/sangwook/catkin_ws/buildDevel space: /home/sangwook/catkin_ws/develInstall space: /home/sangwook/catkin_ws/install######## Running command: "make cmake_check_build_system" in "/home/sangwook/catkin_ws/build"############ Running command: "make -j4 -l4" in "/home/sangwook/catkin_ws/build"####[  0%] Built target std_msgs_generate_messages_cpp[  0%] Built target std_msgs_generate_messages_py[  0%] Built target std_msgs_generate_messages_eus[  0%] Built target std_msgs_generate_messages_lisp[  0%] Built target _velodyne_msgs_generate_messages_check_deps_VelodyneScan[  0%] Built target std_msgs_generate_messages_nodejs[  0%] Built target _velodyne_msgs_generate_messages_check_deps_VelodynePacket[  0%] Built target _velodyne_puck_msgs_generate_messages_check_deps_VelodynePuckPoint[  0%] Built target _velodyne_puck_msgs_generate_messages_check_deps_VelodynePuckPacket[  0%] Built target _velodyne_puck_msgs_generate_messages_check_deps_VelodynePuckSweep[  0%] Built target _velodyne_puck_msgs_generate_messages_check_deps_VelodynePuckScan[  0%] Built target dynamic_reconfigure_generate_messages_nodejs[  1%] Built target velodyne_driver_gencfg[  8%] Built target loam[  8%] Built target sensor_msgs_generate_messages_nodejs[  8%] Built target geometry_msgs_generate_messages_cpp[  8%] Built target sensor_msgs_generate_messages_lisp[  8%] Built target tf2_msgs_generate_messages_lisp[  8%] Built target sensor_msgs_generate_messages_cpp[  8%] Built target actionlib_msgs_generate_messages_lisp[  8%] Built target rosgraph_msgs_generate_messages_nodejs[  8%] Built target geometry_msgs_generate_messages_lisp[  8%] Built target geometry_msgs_generate_messages_nodejs[  8%] Built target roscpp_generate_messages_cpp[  8%] Built target geometry_msgs_generate_messages_eus[  8%] Built target actionlib_generate_messages_py[  8%] Built target actionlib_msgs_generate_messages_py[  8%] Built target sensor_msgs_generate_messages_py[  8%] Built target geometry_msgs_generate_messages_py[  8%] Built target roscpp_generate_messages_eus[  8%] Built target actionlib_generate_messages_cpp[  8%] Built target tf2_msgs_generate_messages_eus[  8%] Built target actionlib_msgs_generate_messages_nodejs[  8%] Built target tf2_msgs_generate_messages_py[  8%] Built target roscpp_generate_messages_nodejs[  8%] Built target rosgraph_msgs_generate_messages_lisp[  8%] Built target tf_generate_messages_eus[  8%] Built target rosgraph_msgs_generate_messages_cpp[  8%] Built target actionlib_generate_messages_eus[  8%] Built target tf_generate_messages_cpp[  8%] Built target tf2_msgs_generate_messages_cpp[  8%] Built target rosgraph_msgs_generate_messages_eus[  8%] Built target tf_generate_messages_lisp[  8%] Built target rosgraph_msgs_generate_messages_py[  8%] Built target tf_generate_messages_nodejs[  8%] Built target tf2_msgs_generate_messages_nodejs[  8%] Built target roscpp_generate_messages_py[  8%] Built target tf_generate_messages_py[  8%] Built target actionlib_generate_messages_lisp[  8%] Built target sensor_msgs_generate_messages_eus[  8%] Built target actionlib_msgs_generate_messages_eus[  8%] Built target actionlib_msgs_generate_messages_cpp[  8%] Built target actionlib_generate_messages_nodejs[  8%] Built target bond_generate_messages_py[  8%] Built target roscpp_generate_messages_lisp[  8%] Built target dynamic_reconfigure_generate_messages_cpp[  8%] Built target nodelet_generate_messages_py[  8%] Built target nodelet_generate_messages_cpp[  8%] Built target dynamic_reconfigure_gencfg[  8%] Built target diagnostic_msgs_generate_messages_nodejs[  8%] Built target bond_generate_messages_lisp[  8%] Built target diagnostic_msgs_generate_messages_eus[  8%] Built target diagnostic_msgs_generate_messages_cpp[  8%] Built target bond_generate_messages_eus[  8%] Built target diagnostic_msgs_generate_messages_lisp[  8%] Built target nodelet_generate_messages_nodejs[  8%] Built target dynamic_reconfigure_generate_messages_lisp[  8%] Built target dynamic_reconfigure_generate_messages_eus[  8%] Built target nodelet_generate_messages_eus[  8%] Built target dynam [... cut ...] b/CMakeFiles/velodyne_input.dir/all] Error 2make[1]: *** Waiting for unfinished jobs....[ 62%] Built target velodyne_puck_driver[ 64%] Built target velodyne_puck_decoder[ 64%] Built target velodyne_msgs_generate_messagesMakefile:138: recipe for target 'all' failedmake: *** [all] Error 2Invoking "make -j4 -l4" failedsangwook@sangwook-VirtualBox:~/catkin_ws$ ​It display an error at 57% ~59% (fatal error: pcap.h: No such file or directory)I need your help pleasethank you
​-----Original Message-----

trainman419, 2018-02-13:
It looks like you're missing pcap.h, which should be one of the dependencies for this package. Did you install the dependencies after you checked this package out into your workspace?

If you didn't, you can install all of the required dependencies with rosdep

```
cd ~/catkin_ws
rosdep install --from-paths src -i -y
```

(I recommend that you do this every time you add a new package to your workspace, to make sure you have all of the required dependencies installed).

choisangwook, 2018-02-13:
Dear Austin​I have resolve thatI don't Install libpcap0.8-dev​sudo apt-get install libpcap0.8-dev​After this command it running All of that succesfully​thank you출처: http://blog.whitelife.co.kr/159 [White Life Story]-----Original Message-----

jack-oquin, 2018-02-14:
It looks like this question has been answered. Thanks for the help, @trainman419!

closed 2018-02-14 by jack-oquin

choisangwook, 2018-02-18:
Dear jack-oquin

apt-get install libpcap0.8-dev
After command run, it disappear error

Thank you very much for helping me