ros-drivers/velodyne #156: Unable to open calibration file /deprecated_file_path/vlp16.yaml

AnkilP, 2018-03-13:
The velodyne driver outputs an error message saying that it cannot open the yaml file in a location that has been deleted. We are passing the right location when we initialize the node but when we open rviz, this error comes up. Where could this reference be?

jack-oquin, 2018-03-14:
How are you launching the node, and what is the exact file path of the calibration file?

JWhitleyAStuff, 2018-04-27:
@AnkilP - Please let us know if you are still having this problem. If not, we'll close the issue. Thanks.

AnkilP, 2018-05-08:
hey sorry, i had no idea i got email notifications - the problem was that
the file was no longer placed there... it's a pretty amateur mistake

JWhitleyAStuff, 2018-05-08:
No problem! Thanks for letting us know. Closing issue.

closed 2018-05-08 by JWhitleyAStuff