ros-drivers/velodyne #157: VLP16 publishing point cloud with timestamp from GPRMC

zheng-rong, 2018-03-20:
Hi all,
I am using VLP-16 (pulk lite) with provided PPS/GPRMC. Does the current velodyne driver (I am using master branch) support the feature that publish point cloud messages with time stamps from GPRMC? I did a quick search throughout the code but didn't find the operation that parses the GPRMC time to the message timestamp.

Thanks in advance,
Zheng

jack-oquin, 2018-03-21:
That is not supported by the current driver. 

The problem is more complex than it first appears. The data need to be published using ROS time and not GPS time, so the results can be synchronized with messages from other ROS devices. 

GPS time *could* be parsed and published by the driver as a separate [sensor_msgs/TimeReference](http://docs.ros.org/jade/api/sensor_msgs/html/msg/TimeReference.html) message on some appropriate topic. I suppose that could even be used by some other node to regulate the system-wide ROS time.

JWhitleyAStuff, 2019-03-28:
Solved by #222.

closed 2019-03-28 by JWhitleyAStuff