ros-drivers/velodyne #162: Enable more than 2 returns

ebennb, 2018-05-10:
Would it be possible to have 5 returns or more per laser fire? Right now it is dual return with strongest and last, but it seems easy enough to have it write out a few of the not strongest returns as well. That would be great for our UAV LIDAR work. Thanks!

JWhitleyAStuff, 2018-05-10:
This isn't something that the driver can control - it's based on the limitations of the hardware. Which sensor are you using?

JWhitleyAStuff, 2018-06-06:
After reviewing the documentation, Velodyne lidars are only capable of 3 modes: strongest return, last return, or dual return. The sensor only returns up to two results per laser fire and is not capable of returning more than two.

closed 2018-06-06 by JWhitleyAStuff